kellerza/sunsynk #133: How to add to HA when running with Docker?

t0mtaylor, 2023-04-22:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

How to add to HA when running with Docker?

Can't add the HASS Addon from the repo, as this errors appears:

`This redirect is not supported by your Home Assistant installation. It needs either the Home Assistant Operating System or Home Assistant Supervised installation method. For more information, see the [documentation](https://www.home-assistant.io/installation).`

Would be good to have a docker compose file to run separately along side HA running in docker?

**Expected behavior**
Should be able to install when running HA with Docker, or instructions of how to run in Docker separately with a docker compose file?

### Your environment

- Home Assistant version: 2022.2.x
- Addon version:
- Inverter make: Sunsynk/Deye/Sol-Ark/...
- Inverter Model: \_\_
- Inverter Firmware: \_\_

### Logs

```log
paste logs in 3x backticks...
```

kellerza, 2023-04-22:
Try the suggested solution in #31

t0mtaylor, 2023-04-22:
I'll give that a try thanks, just found it when searching through the closed tickets

Would be great if theres something in the Readme documentation however, to run this either with the supervised docker container, or without supervised 👍

kellerza, 2023-04-22:
If you manage to test the solution you are welcome to submit a PR on how to do this!

For the non-supervised you will have to download and build and I believe the current base is only for arm architectures,- although a normal python-3.9 docker based should work fine

kellerza, 2023-08-09:
Try this - https://kellerza.github.io/sunsynk/guide/standalone-deployment#run-standalone-using-docker-compose

closed 2023-08-09 by kellerza