kellerza/sunsynk #134: Essential Power MathSensor isn't working properly

duanemck, 2023-04-24:
### Issue related to the Python sunsynk library

### Describe the issue/bug

The Essential Power sensor seems to be calculating incorrectly. According to #75 it is a combination of 
* Inverter Power
* Grid LD Power
* Aux Power

It looks to me like it's not subtracting the "Grid LD power" on my inverter. If you have a look at the below HA sensors:
[image]
(I have "Aux" labeled as "Generator" in HA as I have a generator connected. But at the moment it's not being used).

Essential Power = 1416 + (-640) + 0 = 776. Instead I'm getting 1428W

You'll also notice that I have a CBI nanoview measuring my usage. It never matches the inverter exactly but it's normally within range, in the screenshot it's showing 760W (pretty close to the expected value above) and not 1416W that the add-on is reporting.

### Expected behavior

Covered most of this above, I'm expecting the correct Essential Power value

### Logs (if applicable)

I can't see anything related in the logs.

```log
paste logs in 3x backticks...
```

kellerza, 2023-05-05:
Can you try this as mysensors.py? It prints the calculation in the logs every couple of seconds

add these sensors - old1/old2 should be the December one, hopefully they are the same
```yaml
SENSORS:
  - essential_power
  - essential_power_old1
  - essential_power_old2
```

```python
import attrs
import re

from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import RWSensor, ResolveType
from sunsynk.sensors import RegType, ValType, SensorDefinitions, MathSensor, _LOGGER

SENSORS = SensorDefinitions()

@attrs.define(slots=True, eq=False)
class DebugMathSensor(MathSensor):
    """Math sensor, add multiple registers."""
    cycle = attrs.field(default=1, init=False, repr=False)

    def reg_to_value(self, regs: RegType) -> ValType:
        """Calculate the math value."""
        res = super().reg_to_value(regs)
        self.cycle -= 1
        if self.cycle < 1:
            self.cycle = 20
            _LOGGER.info("%s=%s, %s", dict(zip(self.address, regs)), res, self)
        return res

SENSORS += (
    # https://github.com/kellerza/sunsynk/issues/75
    #  https://powerforum.co.za/topic/8646-my-sunsynk-8kw-data-collection-setup/?do=findComment&comment=147591
    DebugMathSensor(
        (175, 169, 166), "Essential power", WATT, factors=(1, 1, -1), absolute=True
    ),
    DebugMathSensor(
        (175, 169, 166), "Essential power old1", WATT, factors=(1, 1, -1)
    ),
    DebugMathSensor(
        (175, 167, 166), "Essential power old2", WATT, factors=(1, 1, -1)
    ),
)
```

duanemck, 2023-05-11:
Looks like old2 lines up closest with the actual

[image]

kloon, 2023-05-12:
I was just about to open an issue on this as well. My panels have only been installed this week and prior to that I was running in UPS mode only.

I switch between zero export and limit to load in the evenings and back to zero export in the mornings to feed non essentials.

I am noticing the calculation of both `ss_essential_power` and `ss_load_power` reflect the usage of essential + non essentials when in `Zero Export` mode. I understand that it might be right for `ss_load_power`, but for `ss_essential_power` I would expect it would subtract the non-essential power from the total and present that.

[](https://d.pr/i/FzTTrm+)

My screenshot above is while in `Zero Export` mode and SS Essential Power is the `ss_essential_power` sensor and Essential Power is the `ss_load_power` sensor.

As you can see my non-essential and essential readings are pretty much the same and comparing it with my solar production it is clear that my non-essential load is actually included in both the `ss_essential_power` and `ss_load_power` sensors.

kellerza, 2023-05-18:
In the latest version there are now 3 essential power sensors.

- The one used in the dev&normal addon is called `essential_1_power` (I'm using this one and renamed it to **sensor.ss_essential_power** in HA) - @duanemck probably the one you need as well

- The "absolute" one discussed in #75 is called `essential_2_power` - @lispwarez 

- The one use on the powerforum and @slipx06  power flow card is called `essential_power`

@kloon - let us know what works for you

I'm expecting this has something to do with the use of Aux / register 235 - sensor `generator_input` but have not investigated it much further. For now, you can decide which one of the 3 fits your installation...

kloon, 2023-05-18:
> @kloon - let us know what works for you

Thanks, @kellerza, using `essential_1_power` does the job for me.

closed 2023-05-18 by kellerza

duanemck, 2023-05-18:
> In the latest version there are now 3 essential power sensors.
> 
> * The one used in the dev&normal addon is called `essential_1_power` (I'm using this one and renamed it to **sensor.ss_essential_power** in HA) - @duanemck probably the one you need as well
> * The "absolute" one discussed in [Essential Power reading issue #75](https://github.com/kellerza/sunsynk/issues/75) is called `essential_2_power` - @lispwarez
> * The one use on the powerforum and @slipx06  power flow card is called `essential_power`
> 
> @kloon - let us know what works for you
> 
> I'm expecting this has something to do with the use of Aux / register 235 - sensor `generator_input` but have not investigated it much further. For now, you can decide which one of the 3 fits your installation...

Thanks for the fix, using `essential_1_power` and it seems to line up for the most part (accounting for inverter usage and inefficiencies)

RenierBoshoff, 2024-10-28:
> essential_1_power

I am assuming I need to chenge this in the MQTT sensor setting for https://kellerza.github.io/sunsynk/ addon?
[image]