kellerza/sunsynk #136: Number: Set SS Prog1 Capacity - Not allowed via Call Service

cakuhn, 2023-05-07:
Is there a way to call a service and set the value of Prog_x_capacity?  Seems like this is not doable as this entity is a slider / input number.  I have created a sensor that calculates the minium battery percentage level to get through load shedding and it would be great if one could use the variable to set the Program capacity.

Below is the log produced why starting the input of the variable "{{" imediately warn of not being able to do:

thanks

Visual editor is not supported for this configuration
Templates not supported in visual editor
You can still edit your config in YAML.

service: number.set_value
data:
  **value: "{{"**
target:
  entity_id: number.prog1_capacity

kellerza, 2023-05-07:
[image]

or in visual mode

[image]

kellerza, 2023-05-07:
{{ is still an incomplete template. And not a number either.

test your templates in the developer tools first

kellerza, 2023-05-07:
[image]

cakuhn, 2023-05-07:
Thanks, I tested it now and did not realise it actually executes correctly.  

It still shows the below, however it updates the inveter perfectly:-)

Thanks

Visual editor is not supported for this configuration
Templates not supported in visual editor
You can still edit your config in YAML.

kellerza, 2023-05-07:
Indeed, and you probably need data_template