kellerza/sunsynk #138: Timeouts / addon stops working

RowanZee18, 2023-05-09:
### Issue related to

Modbus TCP to Modbus RTU

### Describe the issue/bug

It seems after some time of running home assistant stops receiving sensor information and they all become unavailable.  When I login I find the Sunsynk/Deye Inverter Add-on (multi) has stopped even though watchdog is enabled. The logs show the following; 
2023-05-09 19:35:10,404 ERROR   Read Error: <class 'Exception'>: (1,3,5) 
2023-05-09 19:35:10,426 INFO    ############################################################
2023-05-09 19:35:10,427 INFO    No response on the Modbus interface tcp://192.168.1.105:502, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2023-05-09 19:35:10,428 INFO    ############################################################
2023-05-09 19:35:10,429 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
[cmd] ./run.py exited 0
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

If I start the addon at this point it logs the same error and stops. The modbus rtu plugin is still running with multiple entries shown below and seems to start logging this at the time everything stops working. Looks like a connection from a local ip in the 172 range which is strange because my network is not in that range.  

**Expected behavior**
The modbus addon shouldn't stop sending info to the mqtt server

### Your environment

- Home Assistant version: 2022.5.1
- Addon version: 2021.10.1
- Inverter make: Sunsynk
- Inverter Model: 8kw
- Inverter Firmware: M6.0.1 4 / S 1.7.2.4 / CE.4.2.4

### Logs

09 May 2023 19:34:35 conn_close(): closing connection from 172.30.33.0
09 May 2023 19:35:08 conn_open(): accepting connection from 172.30.33.0
09 May 2023 19:35:15 conn[172.30.33.0]: timeout, closing connection
09 May 2023 19:35:15 conn_close(): closing connection from 172.30.33.0

kellerza, 2023-05-10:
172.30.x.x is a private IP address used internally by Docker, so it simply shows that the Multi addon connects to the mbusd addon on the same Docker host. It maps to your external 192.168.1.0/24 range

The watchdog might stop restarting the addon if it exits repeatedly. What you see in the log is only the last time you/the watchdog restarted the addon.

Could anything have changed wrt the USB-RS485 adaptor/your wiring etc?

Have you tried restarting the mbusd addon? (Mine has probably been running for a year on a Pi 1b without being restarted, so unlikely its the mbusd application,- but could still be your HW/wiring etc)

RowanZee18, 2023-05-10:
Thanks for clearing up the internal docker IP. My setup is still new. I only just got it up and running recently and I'm unable to keep it running for more than 24 hours at a time. It seems to stop working each day. Nothing gets touched or moved in that period and works perfectly before failing randomly.

"target": "https://github.com/kellerza/sunsynk/issues/138#issuecomment-1541326770",
"url": "https://github.com/kellerza/sunsynk/issues/138#issuecomment-1541326770",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

kellerza, 2023-05-10:
This does not seem right. You can try the pymodbus driver to connect to the serial interface directly if you suspect mbusd - see #131 

Although for me mbusd seems bulletproof. I do not run the mbusd addon directly, but have not had any complaints from others

RowanZee18, 2023-06-02:
I'm still experiencing this issue even after moving to pymodbus. The last update definitely improved the stability but it still drops every second day. I can confirm its not the mbusb issue as I do not need to restart this.

kellerza, 2023-08-26:
Try the latest version of the addon and if this still persists, please open a new issue.

closed 2023-08-26 by kellerza