kellerza/sunsynk #140: Total Grid Import: Does not match with actual grid import

kloon, 2023-05-22:
### You Inverter

Details about your Inverter: Sunsynk 8KW Single Phase - Firmware COMM:E426 MCU:6022-1724

### Describe the issue/bug and what you expect

The total grid import sensor does not seem accurate. As per #134 I think this again comes down to the mode the inverter is set in.

Things I am noticing:
- When in limit to load or export modes it does not count grid charging towards grid import
- When in limit to load only the grid import for essentials are counted, non-essentials are not counted towards grid import.

I can see according to the sensor this data is retrieved directly from modbus registers 78 and 80 on the inverter and can confirm that the sunsynk app also does not align correctly with actual grid import so I guess this is one of those issues around should it be fixed in the addon or not. I have also read on social media about people complaining about the numbers on the synsynk app does not match that of actual readings on accounts and it varies quite a bit.

I have not had time to go through a more detailed look at the modbus document to see if there are registers one can combine to get to the right numbers so would like to hear anyone else's findings on this.

kloon, 2023-05-23:
To test this theory, I have gone ahead and set up a Riennman Sum Integral sensor on the CT grid sensor value, and then setup a Utility meter sensor on the Riennman Sum Integral sensor so it resets every day.

Will see how it goes over the next couple of days and report back with my findings.

kellerza, 2023-05-24:
I've never thought of that, but now that you mention it, I am probably seeing something similar.

During the evenings I switch to "limit to load" so that I do not drain my batteries with the geyser. But I have also been seeing a mismatch between my reported energy use and my prepaid meter. This would explain it.

It would not be easy to fix. You could:
1. try an integral on the grid_ct_sensor as you suggested. 
2. sum the total_grid_import with the total_non_ess (external meter), but you will have to account for the inverter program, since you should not sum these during zero export/export to grid

kellerza, 2023-06-09:
How is your sensors working out? Do you mind posting the config here, or even adding it to the examples in the docs?

kloon, 2023-06-09:
According to my calculations the custom sensors are working as they should and reporting correctly. I am just waiting for my next monthly bill to confirm 100% things align. I'd be happy to share the sensor config, however you prefer.

I have emails some people in Sunsynk about this and also opened a ticket, but yet to receive any response.

ga-it, 2023-06-13:
Same issue here with exporting and the "Total Grid Import" value. When mode was "Zero Export" then sum value was correct and included non-essentials and reflected correctly on sunsynk.net. After enabling export, non-essential KWH were not included in the grid import nor consumption figures. The instantaneous read sensor "Non-Essential power" (KW) does pick up the geyser, etc but cannot be used to feed the energy dashboard which wants a cumulative measure. If this sensor was available as a sum, then you could correct the grid import value.

kellerza, 2023-06-14:
It’s unfortunate that these sensors change when you change the export mode.

The addon does not keep any state. And this is not something I plan on adding soon. To calculate the energy, you will have to keep state

@kloon solution of doing this in HASS is probably your best option today

kloon, 2023-07-12:
I finally got a reply from Sunsynk on this issue after 3 weeks, no surprise it was a generic one saying to update to the latest firmware, which I did, yet the issue is still present.

kellerza, 2023-08-09:
Adding a wontfix label as this is the Inverter's bahaviour, but will keep the issue open for reference