kellerza/sunsynk #153: Not really an issue, but a feature request

zs3y, 2023-06-19:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

NOT a bug, but a feature request.

Since the Sunsynks built in clock is a bit on the drifty end, would it be possible to enable a once a week or once a day time syncronisation to the machine running the add-on. I am pretty certain that just about every HA instance on bare metal is, with ntp, pretty much correct, time wise. Even HA in a VM is likely to run on a host that is ntp time syncronised and has more accurate time than the inverter. Maybe a on/off toggle to enable/disable this feature if there's someone out there that does not want it.

**Expected behavior**

Assume the toggle to enable this feature, then at, let's say 00:00:00 on a daily basis the add-on should update the date and time of the inverter to the correct/current date and time. Most likely once a week on a Monday or Sunday morning would probably also be plenty.

### Your environment

- Home Assistant 2023.6.2
- Addon version: 2023.05.19-0.3.6
- Inverter make: Sunsynk
- Inverter Model: 5kW Inverter
- Inverter Firmware: V0.2.0.1 / V3.3.7.4 / VE.4.2.6

### Logs

```log
No log to add
```

### Additionally

I must admit, I looked at the example, but cannot figure this lot out, otherwise I would have done this myself, I must thank you, though, even if this feature request gets ignored or deemed not worthwhile, this integration for the Sunsynk is working pretty great and allows me to put automations in place that weren't possibly without it, always trekking to the garage to make changes was getting a bit old.... :-)

kellerza, 2023-06-19:
If you have a working automation (with the correct template to write to the sensor), it might be something worth adding to the addon docs?

closed 2023-06-19 by kellerza

zs3y, 2023-07-26:
HELP!
Not a working automation but...in the log...
`2023-07-26 14:11:28,186 CRITICAL Writing sensor date_time=2023-07-26 14:11:20 [(22, 23, 24)=(5895, 6670, 2836)]`

If I try and set the time, the above (with the time and date changed to suit the actual event date and time) appears in the Sunsynk log... no actual date or time is written to the inverter, is there something I'm overlooking? I am just using the entry on lovelace and am changing the time component to suit and hitting enter at the appropriate moment.
[image]
and in the logbook the entries are for this instance...
[image]

which means to me, it wants to update the date and time to the specified values, but this actually does no happen...

Hoe nou? wat maak ek verkeerd? Help!

Nurgus, 2025-03-10:
Same problem here, I set the date_time text to the correct string for right now ( eg 2025-03-10 13:23:29 ) and I see this in the log:

```
13:23:29 INFO    Queue update date_time=2025-03-10 13:23:29
13:23:30 INFO    Writing sensor date_time=2025-03-10 13:23:29 Registers:{62=0x1903 63=0x0a0d 64=0x171d}
```

But the time doesn't really change. It remains wrong by a few minutes.