kellerza/sunsynk #156: Crash when changing battery charge settings

ASomerN, 2023-07-08:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

When requesting a change to the battery settings, the add-on crashes (using Dev add-on).

**Expected behavior**
Setting changes are made

### Your environment

- Home Assistant version: 2023.7.1
- Addon version: 2023.01.19b-0.2.9
- Inverter make: Sunsynk
- Inverter Model: 16k
- Inverter Firmware: COMM:e424 - MCU:9013

### Logs

```
2023-07-08 17:52:54,813 INFO    MQTT: Connecting to homeassistant@core-mosquitto:1883
2023-07-08 17:52:54,859 INFO    MQTT: Connection successful
2023-07-08 17:53:08,654 INFO    Writing sensor Prog1 Capacity: prog1_capacity=(40,) [old (15,)]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 479, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 461, in main
    await write_sensors()
  File "/usr/src/app/./run.py", line 440, in write_sensors
    await SUNSYNK.write_sensor(  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 50, in write_sensor
    await self.write_register(address=sensor.reg_address[0], value=val1)
  File "/usr/local/lib/python3.9/site-packages/sunsynk/pysunsynk.py", line 88, in write_register
    w_r = await self.client.write_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

kellerza, 2023-07-09:
Can you try with the multi addon?

ASomerN, 2023-07-11:
Hi, Sorry this took a little while it's not my install.

I've switched over to multi, I can read everything fine, and unlike the dev add-on it doesn't crash when I try and make a setting change but at the same time it doesn't make the change,  this is in debug level 1 logs when trying to change battery SOC in one of the timeslots after start-up:

```
2023-07-11 08:54:43,251 INFO    options Loading HASS OS configuration
2023-07-11 08:54:43,318 INFO    Using Single phase sensor definitions.
2023-07-11 08:54:43,322 INFO    Connecting to /dev/ttyUSB0
2023-07-11 08:54:43,323 INFO    PyModbus 3.2.2 Serial: /dev/ttyUSB0
2023-07-11 08:54:43,341 INFO    Serial connected.
2023-07-11 08:54:43,344 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:54:43,344 INFO    Reading startup sensors rated_power, serial, prog6_time, prog2_time, prog1_time, prog3_time, prog4_time, prog5_time, battery_low_capacity
2023-07-11 08:54:43,518 INFO    ############################################################
2023-07-11 08:54:43,519 INFO                Inverter serial number 'removed'
2023-07-11 08:54:43,519 INFO    ############################################################
2023-07-11 08:54:43,534 INFO    MQTT: Connecting to homeassistant@core-mosquitto:1883
2023-07-11 08:54:43,548 INFO    MQTT: Connection successful
2023-07-11 08:54:46,790 INFO    Battery power:step:mean: -3992->-4109, 2 samples
2023-07-11 08:54:48,813 INFO    PV2 power:step:mean: 1846->1961, 4 samples
2023-07-11 08:54:48,822 INFO    Battery power:step:mean: -4109->-4244, 2 samples
2023-07-11 08:54:52,837 INFO    Battery power:step:mean: -4244->-4150, 4 samples
2023-07-11 08:54:54,838 INFO    PV2 power:step:mean: 1961->1876, 6 samples
2023-07-11 08:54:54,844 INFO    Battery power:step:mean: -4150->-4052, 2 samples
2023-07-11 08:55:04,889 INFO    Battery power:step:mean: -4052->-3966, 10 samples
2023-07-11 08:55:10,914 INFO    PV2 power:step:mean: 1876->1785, 16 samples
2023-07-11 08:55:10,920 INFO    Battery power:step:mean: -3966->-3844, 6 samples
2023-07-11 08:55:12,917 INFO    PV2 power:step:mean: 1785->1697, 2 samples
2023-07-11 08:55:12,923 INFO    Battery power:step:mean: -3844->-3643, 2 samples
2023-07-11 08:55:14,921 INFO    PV2 power:step:mean: 1697->1579, 2 samples
2023-07-11 08:55:14,926 INFO    Battery power:step:mean: -3643->-3379, 2 samples
2023-07-11 08:55:18,708 INFO    Serial lost connection.
2023-07-11 08:55:18,714 INFO    Serial lost connection.
2023-07-11 08:55:18,714 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 08:55:18,722 ERROR   timeout reading register 169 (1)
2023-07-11 08:55:18,734 INFO    Serial connected.
2023-07-11 08:55:18,736 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:18,829 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:18,830 INFO    Serial connected.
2023-07-11 08:55:18,830 ERROR   Factory protocol connect callback called while connected.
2023-07-11 08:55:19,059 INFO    PV1 power:step:mean: 813->672, 31 samples
2023-07-11 08:55:19,065 INFO    PV2 power:step:mean: 1579->1322, 1 samples
2023-07-11 08:55:19,068 INFO    Battery power:step:mean: -3379->-2809, 1 samples
2023-07-11 08:55:21,061 INFO    Battery power:step:mean: -2809->-2645, 2 samples
2023-07-11 08:55:21,736 CRITICAL Writing sensor prog1_capacity=32 [(268,)=(32,)] 
2023-07-11 08:55:24,747 INFO    Serial lost connection.
2023-07-11 08:55:24,750 ERROR   timeout writing register 268=32
2023-07-11 08:55:24,754 INFO    Serial lost connection.
2023-07-11 08:55:24,754 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 08:55:24,762 INFO    Serial connected.
2023-07-11 08:55:24,764 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:24,962 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:24,963 INFO    Serial connected.
2023-07-11 08:55:24,964 ERROR   Factory protocol connect callback called while connected.
2023-07-11 08:55:27,939 INFO    Serial lost connection.
2023-07-11 08:55:27,944 INFO    Serial lost connection.
2023-07-11 08:55:27,945 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 08:55:27,949 ERROR   timeout reading register 178 (1)
2023-07-11 08:55:27,959 INFO    Serial connected.
2023-07-11 08:55:27,960 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:28,354 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:28,355 INFO    Serial connected.
2023-07-11 08:55:28,355 ERROR   Factory protocol connect callback called while connected.
2023-07-11 08:55:30,967 INFO    Serial lost connection.
2023-07-11 08:55:30,972 INFO    Serial lost connection.
2023-07-11 08:55:30,973 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 08:55:30,975 ERROR   timeout reading register 16 (2)
2023-07-11 08:55:30,985 INFO    Serial connected.
2023-07-11 08:55:30,986 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:31,783 INFO    Connected to /dev/ttyUSB0
2023-07-11 08:55:31,784 INFO    Serial connected.
[... 39 lines cut ...]
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_SENSORS_BATCH_SIZE: 60
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: removed
MQTT_PASSWORD: removed
DEBUG: 1
DEBUG_DEVICE: /dev/ttyUSB0

```

ASomerN, 2023-07-11:
I've done a complete uninstall and re-install. I can see the critical log now when attempting to write:

```
2023-07-11 13:57:53,007 INFO    MQTT: Connecting to homeassistant@core-mosquitto:1883
2023-07-11 13:57:53,033 INFO    MQTT: Connection successful
2023-07-11 13:58:17,367 INFO    Serial lost connection.
2023-07-11 13:58:17,371 INFO    Serial lost connection.
2023-07-11 13:58:17,372 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 13:58:17,379 ERROR   timeout reading register 274 (1)
2023-07-11 13:58:17,381 CRITICAL Writing sensor prog1_capacity=53 [(268,)=(53,)] 
2023-07-11 13:58:17,392 INFO    Serial connected.
2023-07-11 13:58:17,394 INFO    Connected to /dev/ttyUSB0
2023-07-11 13:58:17,490 INFO    Connected to /dev/ttyUSB0
2023-07-11 13:58:17,491 INFO    Serial connected.
2023-07-11 13:58:17,491 ERROR   Factory protocol connect callback called while connected.
2023-07-11 13:58:20,405 INFO    Serial lost connection.
2023-07-11 13:58:20,407 ERROR   timeout writing register 268=53
2023-07-11 13:58:20,411 INFO    Serial lost connection.
2023-07-11 13:58:20,412 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 13:58:20,420 INFO    Serial connected.
2023-07-11 13:58:20,421 INFO    Connected to /dev/ttyUSB0
2023-07-11 13:58:20,622 INFO    Connected to /dev/ttyUSB0
2023-07-11 13:58:20,623 INFO    Serial connected.
2023-07-11 13:58:20,624 ERROR   Factory protocol connect callback called while connected.
2023-07-11 13:58:23,594 INFO    Serial lost connection.
2023-07-11 13:58:23,599 INFO    Serial lost connection.
2023-07-11 13:58:23,600 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-07-11 13:58:23,603 ERROR   timeout reading register 178 (1)
2023-07-11 13:58:23,612 INFO    Serial connected.
2023-07-11 13:58:23,614 INFO    Connected to /dev/ttyUSB0
2023-07-11 13:58:24,010 INFO    Connected to /dev/ttyUSB0
2023-07-11 13:58:24,011 INFO    Serial connected.
2023-07-11 13:58:24,011 ERROR   Factory protocol connect callback called while connected.
```

kellerza, 2023-07-19:
And reading is ok?

Can you modify the inverter settings on the inverter's control panel? (I believe there was some lockout mode where an inverter could be placed in read-only mode)

If this is still not working, try the fault finding again, as the problem does present itself like a bad connection (timeouts)

For a direct serial port, I would still use mbusd as a TCP proxy...

ASomerN, 2023-07-20:
Reading is working from this add-on. We can make changes on the inverter directly and also via the sunsynk app.
Do you know where on the inverter/app you would enable disable lock-out (want to check this just in case).

I've copied the set-up I've used but I'm on 8kw inverter and this issue is on a 16kw.
I'll swap out all the hardware and reconfigure and let you know if it persists on the multi add-on.

ASomerN, 2023-07-24:
I'm stumped. I can read data OK.
I've got this writing fine on my 8kw Sunsynk inverter with an adapter direct to HA. The issue is on a 16kw Sunsynk inverter.

I've swapped out the USB adapter and ethernet: https://www.amazon.co.uk/gp/product/B01N3LM0PU/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
Done a rebuild, reinstall, and complete reboot of HA and everything is up to date. I can read everything in real time but can't write these system settings.

I've tried the different settings you outlined here: https://github.com/kellerza/sunsynk/issues/131 
I reverted back to Direct Serial + pymodbus as the others had no impact on this.

Is this an issue with the 16kw Inverter? I am happy to try anything else you can suggest to try and diagnose this.

kellerza, 2023-07-24:
You are probably the first user with a 16kW inverter,- is this a 3-phase inverter?

ASomerN, 2023-07-25:
Yes. It's not using 3 phase but I believe it would be capable:
[image]
[image]

ASomerN, 2023-07-25:
I've just compared this to mine and there's no difference. My 8kw has options for phase a/b/c too but I'm not on 3 phase from the grid either. So the difference seems to be mine is 8kw and the one not working is 16kw

kellerza, 2023-07-25:
The register I had in mind for the lockout was. But this probably affect changes via the UI and Modbus

020 | Remote Lock | R/W

ASomerN, 2023-07-30:
I've moved the PORT to use: tcp://homeassistant.local:502 as advised, this has removed all timeout errors.
I've set debug level to 2. When I try and make a change to Battery SOC this is the error:

```
2023-07-30 10:34:31,232 CRITICAL sunsynk.sunsynk Writing sensor prog1_capacity=30 [(268,)=(30,)] 
2023-07-30 10:34:31,233 DEBUG   pymodbus.logging send: 0x0 0x9e 0x0 0x0 0x0 0x9 0x1 0x10 0x1 0xc 0x0 0x1 0x2 0x0 0x1e
2023-07-30 10:34:31,235 DEBUG   pymodbus.logging Adding transaction 158
2023-07-30 10:34:32,895 DEBUG   asyncio <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>> received EOF
2023-07-30 10:34:32,897 INFO    pymodbus.logging Protocol lost connection.
2023-07-30 10:34:32,898 DEBUG   pymodbus.logging Client disconnected from modbus server: None
2023-07-30 10:34:32,899 DEBUG   pymodbus.logging Getting transaction 158
2023-07-30 10:34:32,899 DEBUG   pymodbus.logging Waiting 100 ms before next connection attempt.
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 320, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 298, in main
    await write_sensors()
  File "/usr/src/app/./run.py", line 274, in write_sensors
    await STATE[ssi].inv.write_sensor(
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 65, in write_sensor
    await self.write_register(address=addr, value=regs[idx])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/pysunsynk.py", line 71, in write_register
    res = await self.client.write_registers(  # type:ignore
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/base.py", line 231, in async_execute
    resp = await asyncio.wait_for(req, timeout=self.params.timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 478, in wait_for
    return fut.result()
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

I've also asked via supplier for inverter firmware to be checked.

Trying 3 phase option in the add-on had no effect as it failed to read any of the configured sensors.

ASomerN, 2023-08-13:
Is there any additional logging I can generate to try and assist diagnostics?

kellerza, 2023-08-15:
Can you try to write with an external tool? You should be able to connect from your laptop to the mbusd server

Maybe https://github.com/favalex/modbus-cli

trackmafia, 2023-08-16:
Good day, I have the sunsynk 16k and this is working perfect for me using kellerza instructions and rs485 adapter he said was most reliable. Are you sure you have everything wired correctly? It's possible this could be a wiring/crimping problem? Or also try lowering the number of registers read in one go? When I set up mine, increasing the timeout between reads/writes in mbusd conf file made read errors less likely, so you may have to play with mbusd conf settings and test with modpoll also. Just some ideas.

ASomerN, 2023-08-16:
I'll try again but at this point I've replaced everything including etherent and usb adapter. I've also gone direct to the can port the wifi dongle uses.

Just to confirm, are you able to write settings OK? I can read fine just not write battery soc and turn grid charge on.

trackmafia, 2023-08-17:
Good day,

Yes I can write everything ok, I'm using this version of addon, plan to update today:  2023.08.06-0.4.0

I did not have grid charge function, but I enabled since you were having issues. Previously I had grid charge permanently enabled and used grid charge current. I was abled to adjust the current via slider without issue.

I've done some testing and here is what I found:

1. Grid charge enabled is setup as a read only sensor in the addon, showing 0 if off, 1 if on. So you really cannot write to this.
2. I created a swtichrw custom sensor and this worked, I'll post the code below. Although it does work in changing grid charge off and on, it always throws this error, not sure why:

2023-08-17 07:03:28,275 CRITICAL Writing sensor grid_charge=OFF [(232,)=(0,)] 
2023-08-17 07:03:29,709 ERROR   failed to write register 232=0

Here's the code:

from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor, SwitchRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()

SENSORS += SwitchRWSensor(232, "Grid Charge")

ASomerN, 2023-08-20:
Hi.

This is great you replicated the issue. 

I think this confirm an issue specifically on 16kw inverter and this add-on as I'm using the exact same set up, add on, configuration etc on an 8kw sunsynk converter without issues and can toggle grid/gen/grid+gen/no grid or gen without any issues for 8kw inverter.

I'll also try and enable grid charge for one of the battery switch points via app then adjust the target as previously I think this also caused a crash.

kellerza, 2023-08-20:
@ASomerN, what sensor are you trying to write to? `prog1_charge` (274) or `grid_charge` (232)?

Can you write these with an external tool?

From what @trackmafia posted, it seems like you cannot write to 232 registers, but the rest seems ok.

ASomerN, 2023-08-20:
Hi, I'm trying to write the setting supplied on this:

[image]

When trying to change any of these settings it crashes:

[image]

I believe these are:

  - prog1_time
  - prog2_time
  - prog3_time
  - prog4_time
  - prog5_time
  - prog6_time
  - prog1_power
  - prog2_power
  - prog3_power
  - prog4_power
  - prog5_power
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge

I can change any of the settings using the official app and directly in the inverter.

I won't be able to connect anything directly other than what's there right now as it's at my parent's house (I'm gladly doing the dutiful son tasks). Also, I'm not 100% sure what I'm required to do should I connect a different device.

I hope that helps clarify what it is that's not working. Everything is the same as with my set-up apart from the inverter and it works for me at my property using the "multi" add-on.

trackmafia, 2023-08-22:
@kellerza All of those I tried to write to work. It shows an error in the logs when writing every time, but still proceeds with the write. It seems like the first try fails but then 2nd try works. To confirm, I have written to grid charge enabled, grid current, and battery charge current. All have this behavior. 

However, what I said was that the grid charge enabled code is set to sensor, which is read only. I made a custom sensor using switchRWsensor for the same modbus register, and then it worked. So you may want to consider changing sensor to switchrwsensor for this register.

kellerza, 2023-08-22:
Thanks for the update @trackmafia - will add a small delay (10ms) before attempting to write anything in the next version, maybe that improves things

ASomerN, 2023-08-22:
Hi thanks fpr the info both and I'll aim to be at the property soonish to
try new things.

I'll give the custom code a go, but to confirm, this doesn't happen with my
8kw converter when writing which confuses me.

I've asked the installers to get firmware updated but due to sunsynk
mondodb migration it's some way off happening.

ASomerN, 2023-08-30:
I've determined there's nothing wrong software-wise, but it's worth leaving this open until I can document everything and suggest updates to the wiki.

Long story short, having replaced everything again, and tested with a new USB to RS232, the 16kw Sunsynk inverter only allows writing of settings via the rs232 port the wifi dongle uses.

I've had a dig around and on this link: https://solar-assistant.io/shop/products/sunsynk_rs485
It says:

[image]

Following through the help section it seems to suggest a method of connecting both battery and solar-assistant monitoring: "Note with the latest 16kW models/firmware the BMS port is used for SolarAssistant monitoring and battery CAN communication at the same time. See [the 2 in 1 BMS port](https://solar-assistant.io/help/deye/2_in_1_bms_port)"
It doesn't suggest it can write though.

I'm exploring options to move the default Sunsynk monitoring app to the read-only port and connect the home assistant system into the RS232.

kellerza, 2023-08-30:
I'm not sure it will work with the 2-in-1 port, since that is still RS485 but just a different port.

According to the links you shared, you have to move it to the RS232 port. So no more RS485

[image]

trackmafia, 2023-08-30:
I'm using the rs485 port and writing is happening for me just fine. I have the WiFi dongle on the serial port, and used to have battery communication on canbus, but I removed that. Maybe you have a newer model, I don't have the 2 in 1 port, everything is separate on my model.

ASomerN, 2023-08-31:
What's the method for connecting via the solarman WiFi adapter? The one
supplied in the UK doesn't have an Ethernet port inside.

kellerza, 2023-08-31:
The solarman dongle connects to your wifi

The addon then connects to the local IP of the solarman dongle (over wifi)

ASomerN, 2023-08-31:
Thanks.

Is there an example of the yml configuration you would use for this? I've
tested so far with umodbus and pymodbus. Would assume I need to reference
an ip address?

kellerza, 2023-09-01:
The dongle_serial_number is required & documented [here](https://kellerza.github.io/sunsynk/reference/multi-options#inverters)

You need to identify the local IP,- either from your router, I use an old freeware version of softperfect network scanner [see here](https://www.portablefreeware.com/?id=730)

Also see https://github.com/kellerza/sunsynk/discussions/174

ASomerN, 2023-09-01:
Unfortunately the wifi logger that was supplied is the sunsynk version not solarman, the serial number isn't an integer its a mix of letters and numbers. 

I'll try and find where to get a solarman dongle from supplier. I've been asked to try and keep both the sunsynk app working and also the home assistant connection.

closed 2026-08-15 by kellerza