kellerza/sunsynk #160: add-on installation trouble

eduardkieser, 2023-07-18:
Sunsynk/Deye Inverter Add-on (multi) (2023.05.19-0.3.6)

I am unable to install the add on. When pressing the "install" button, it does a bit of thinking and a popup appears with the following error:
```
Failed to install add-on
The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt==1.6.1 mqtt-entity==0.0.3 sunsynk[pymodbus,umodbus]==0.3.6 pyyaml==5.4.1' returned a non-zero code: 1
```

It seems like this might actually be more related to the python package, but it appeard on the HA add-on ui, so i'll leave it here for now.

- Home Assistant Core: 2023.4.6
- OS Version: Home Assistant OS 10.0  [running on rpi 3b]
- Inverter make: Sunsynk 8kw
- Inverter Model: Unknown to me
- Inverter Firmware: ditto

kellerza, 2023-07-19:
This is most likely related to your Internet connectivity. I would check, in order: DNS resolution, do you need/use any proxies, can you ping 8.8.8.8 / pypi.org from your HASS terminal.

Sometimes just trying it again later also works

alanmors, 2023-07-19:
I am getting the same error upon trying to install. Have checked what you suggested

Failed to install add-on
The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt==1.6.1 mqtt-entity==0.0.3 sunsynk[pymodbus,umodbus]==0.3.6 pyyaml==5.4.1' returned a non-zero code: 1

Home Assistant 2023.7.2
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest

Racerider101, 2023-07-19:
I am getting the same error also.  HA fully updated to latest versions, restarted, rebooted and can ping 8.8.8.8 / pypi.org with no packet loss.  Tried to run install about 6 or 7 times, each time the same error as others posted above.  Other add-ons able to install with no issues.

Also tried the deprecated one, same error.  Out of ideas now.

eduardkieser, 2023-07-19:
It looks like it's related to the pyyaml installation. If I run the full pip line in a venv on my ubuntu machine it fails at the pyyaml step.
Also if I run:
```
pip3 install --no-cache-dir --disable-pip-version-check pyyaml==5.4.1
```
I get a long error that ends with:
```
File "/tmp/pip-build-env-ssvx1l8g/overlay/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
          raise AttributeError(attr)
      AttributeError: cython_sources
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```
If I change the version to say, `6.0.1` it seems to work. Perhaps the hosted package for `5.4.1` is broken somehow?

kellerza, 2023-07-19:
Likely the HA docker images has been updated with cython3 - https://github.com/yaml/pyyaml/issues/601

Will change to pyyaml 6.0.1, but do not have time to test this today

kellerza, 2023-07-19:
> If I run the full pip line in a venv on my ubuntu machine it fails at the pyyaml step.

You can also look at the Supervisor logs in HA and you see something similar. 

If you rebuild the addon (might have to uninstall it to download the new dockerfile since I did not change the version) it should pull the new Dockerfile with pyyaml 6.0.1.

Please post here if there are any issues with yaml, but in normal conditions it only uses json and not yaml

kellerza, 2023-07-19:
You can try to update. Please confirm if all ok. There were a couple of other changes that also got included with the upgrade.

Racerider101, 2023-07-19:
Bobs your uncle as they say!  Install smooth now.  Thank you for the super speedy work! Not got it connected to inverters yet, next job later today.

alanmors, 2023-07-19:
Thanks for your speedy response. All installed.

eduardkieser, 2023-07-19:
Success!! Thanks. Closing.

closed 2023-07-19 by eduardkieser