kellerza/sunsynk #163: Slow update of RW sensors

cods4, 2023-07-20:
Thanks for your hard work @Ivan-L and @kellerza to get the writable sensors working. I am using them regularly.

I have an issue where sensors take a very long time to update after changing them from within homeassistant.

I.e I can change select.inv1_prog2_charge from 'No Grid or Gen' to "Allow Grid' in the HA UI and when I click out of the entity and then open it up again, it is showing 'No Grid or Gen' again. It will eventually update, but it often takes 10minutes or more.

The change in the inverters happens almost immediately, but the value takes a long time to be read back in to the HA entity.
I have also confirmed that it's not updating in MQTT either.

I'm running the latest version of the add-on (0.3.7), but this issue didn't just start today, it has been going on for a while.

I am connected to the inverters via RS232 ports and mbusd on a raspberry pi 4.

I don't have any connection issues in the logs (and haven't done since moving from RS485 to RS232).

kellerza, 2023-08-06:
Really strange @cods4, the behavior should be to read it again immediately after write - see [here](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-multi/run.py#L285-L291)

Did you change the sensor filter/modifier by any chance?

cods4, 2023-08-14:
Sorry about the delay getting back to this.
I haven't used any modifiers except for trying it on one of my custom sensors.

Could the issue be that I am querying the inverter too much (i.e I'm monitoring a lot of sensors? 
I am using mbusd running on a rpi4 connected to the inverters' RS232 ports. I don't get any communication errors in the logs, just some related to my custom sensors which I think can be ignored.

Can you notice anything that looks wrong below?

Here are some logs:
```
2023-08-15 09:48:14,065 CRITICAL Writing sensor export_control=Zero Export to CT [(244,)=(2,)] 
2023-08-15 09:48:24,823 CRITICAL Writing sensor grid_peak_shaving=ON [(280,)=(256,)] [Register 0-->256]
2023-08-15 09:48:25,115 WARNING unknown1: Unknown register value 256
2023-08-15 09:48:46,701 WARNING unknown1: Unknown register value 256
2023-08-15 09:48:53,879 WARNING unknown1: Unknown register value 256
2023-08-15 09:49:01,075 WARNING unknown1: Unknown register value 256
2023-08-15 09:49:01,126 CRITICAL Writing sensor grid_peak_shaving=OFF [(280,)=(0,)] [Register 256-->0]
2023-08-15 09:49:08,653 WARNING unknown1: Unknown register value 256
2023-08-15 09:49:08,657 CRITICAL Writing sensor export_control=Export First [(244,)=(0,)] 
```

Here is my config:
Note: I understand 'SENSORS_FIRST_INVERTER:' isn't used yet. This is just sitting there waiting until you enable it in a future update ;)
```
DRIVER: umodbus
INVERTERS:
  - SERIAL_NR: "0123456789"
    HA_PREFIX: INV1
    MODBUS_ID: 1
    PORT: tcp://192.168.23.17:50001
  - SERIAL_NR: "0123456789"
    HA_PREFIX: INV2
    MODBUS_ID: 1
    PORT: tcp://192.168.23.17:50002
  - SERIAL_NR: "0123456789"
    HA_PREFIX: INV3
    MODBUS_ID: 1
    PORT: tcp://192.168.23.17:50003
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - battery_voltage
  - battery_power
  - battery_current
  - inverter_power
  - inverter_voltage
  - inverter_frequency
  - grid_frequency
  - grid_power
  - grid_ld_power
  - grid_voltage
  - grid_current
  - grid_ct_power
  - load_power
  - pv1_power
  - pv1_voltage
  - pv1_current
  - pv2_power
  - pv2_voltage
  - pv2_current
  - essential_power
  - non_essential_power
  - total_active_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - total_load_energy
  - total_pv_energy
  - fault
  - dc_transformer_temperature
[... 159 lines cut ...]
SENSORS += Sensor(0, "Device Type")
SENSORS += Sensor(1, "Modbus Address")
SENSORS += Sensor(2, "Communication Protocol Version")
SENSORS += Sensor(9, "Processor Type")
SENSORS += Sensor(20, "Remote Lock")
SENSORS += Sensor(52, "PV Curve Enable")
SENSORS += Sensor(53, "Hardware Matching")
SENSORS += Sensor(56, "Limiter Function Enable")
SENSORS += Sensor(185, "Battery Status")
SENSORS += RWSensor(210, "Battery Max Charge Curent", AMPS, -1)
SENSORS += RWSensor(211, "Battery Max Discharge Curent", AMPS, -1)
SENSORS += RWSensor(243, "Battery vs Load Priority")
SENSORS += RWSensor(244, "Limit Control Function")
SENSORS += RWSensor(245, "Limit Grid Output Power")
SENSORS += RWSensor(246, "External Current Sensor Clamp Phase")
SENSORS += SwitchRWSensor(247, "Solar Sell")
SENSORS += SelectRWSensor(244, "Export Control", options={0: "Export First", 1: "Zero Export to Load", 2: "Zero Export to CT"})
SENSORS += SwitchRWSensor(280, "Unknown1", on=0x01, bitmask=0x000)
SENSORS += SwitchRWSensor(280, "Gen Peak Shaving", on=0x10, bitmask=0x010)
SENSORS += SwitchRWSensor(280, "Grid Peak Shaving", on=0x100, bitmask=0x100)
SENSORS += SwitchRWSensor(280, "Micro Grid Export", on=0x1000, bitmask=0x1000)
SENSORS += SwitchRWSensor(280, "Unknown3", on=0x2000, bitmask=0x2000)
SENSORS += SwitchRWSensor(280, "Unknown4", on=0x4000, bitmask=0x4000)
SENSORS += SwitchRWSensor(280, "DRM", on=0x8000, bitmask=0x8000)
```

kellerza, 2023-08-15:
Interesting config 😄 I don't see anything wrong and suspect this might be related to reading a lot.

I'm busy with a fairly big update to make reading more predictable. Replacing filters with schedules. It will also allow reading multiple inverters at the same time (assuming you have different bus+HW reading these, which I suspect you might have).

In your case sensors for the first inverter will also improve things.

For now my recommendation is to wait a bit until I release a new version, I already have it running here, so maybe later this week

cods4, 2023-08-17:
You fixed it! Thanks very much @kellerza!

closed 2023-08-17 by cods4