kellerza/sunsynk #165: Id_entity

cc120689, 2023-08-09:
Only a few question:
Is there a way for the addon not to add the device name and serial number to the entity_id?

kellerza, 2023-08-09:
Do you mean to the Device's description?

Device name you can specify in the configuration (Sunsynk/Deye/My inverter etc). Today it always adds the serial nubmber to that description

Why would you want to hide it?

cc120689, 2023-08-09:
mainly because I already had them configured like this without the name of the device in front, anyway it doesn't matter anymore I have modified the name showed  so I will not suffer any more changes in future updates.
The thing about adding the name and the serial number in the entity_id of the sensors only started to happen in the last update

kellerza, 2023-08-11:
Only noticed this today for the first time

It seems to be a change in the way HA assigns friendly_name: https://www.home-assistant.io/integrations/mqtt/#naming-of-mqtt-entities

> The entity is a member of a device and entity.name is not None: friendly_name = f"{device.name} {entity.name}"

The device.name today is `f"{MANUFACTURER} Inverter {SERIAL}"`. That should probably change to either just `MANUFACTURER`, or introduce a new `NAME` in settings. 

Related to this is the entity_name, the entity.name should NOT include the `SENSOR_PREFIX` (as is the case today). Sensor prefix was meant for the entity_id, which seem to be the entity_object_id - so need to introduce entity.object_id = `f"SENSOR_PREFIX {sensor.name}"`

Comments welcome

cc120689, 2023-08-11:
Now I have everything configured again with the new entity ID, when I have a while I'll try adding a prefix to see if it replaces all that or simply adds it, I'll comment when I try it, thanks for your answer, you have a prefix ?  With what id do your sensors appear?

cods4, 2023-08-15:
> he device.name today is `f"{MANUFACTURER} Inverter {SERIAL}"`. That should probably change to either just `MANUFACTURER`, or introduce a new `NAME` in settings.

Sounds good. It should be an option in settings for each inverter. I.e I used to have INV1 PV1 Power, INV2 PV1 Power etc. Now it is 'Deye Inverter 734243290 PV1 Power'
I would like to be able to enter NAME: INV1, NAME: INV2... settings for each inverter

kellerza, 2023-08-15:
Today we have `ha_prefix` per inverter and `manufacturer` as global setting. Can we work with these 2 only, or would you want a different name&ha_prefix per inverter. One for the entity object IDs and one for the pretty display

my preference is to have only the ha_prefix

cc120689, 2023-08-15:
It would be great if it could be customized to that level, manufacturer + num.  serial is too long.  I would prefer that adding the prefix overrides the global configuration and does not add a manufacturer. 
Thanks for your attention!

cods4, 2023-08-15:
> Today we have `ha_prefix` per inverter and `manufacturer` as global setting. Can we work with these 2 only, or would you want a different name&ha_prefix per inverter. One for the entity object IDs and one for the pretty display
> 
> my preference is to have only the ha_prefix

You're right I forgot about the ha_prefix
For me just having ha_prefix & sensor_definition would work. The friendly name is important to some because voice assistants etc use it. It would be a pain if Home Assistant kept setting it back to some long string like it is now.

cc120689, 2023-08-15:
In any case, from your comment I understand that the id of the entity does not depend on your complement, but on the global configuration of Home assistant.  I attached a screenshot so that it is completely clear what id I mean.  If so, we can consider this petition closed and, as always, thanks for your work, you have more and more followers in Spain!.
[image]

kellerza, 2023-08-15:
Using just `ha_prefix` (per inverter) and `manufacturer` (global), I get the following

[image]

here you see the `entity_id = sensor.ss_battery_current` and the `name = SS Battery power` (without an override)

[image]

The only downside I see here is the `SS` as the name of the device (first picture). But since that forms the prefix to all the sensor's friendly_names there is no way to really improve this

PR #168 (2023-08-17): Read sensors based on Schedules
- Support Schedules for more accurate reading & reporting
  - System timer tracks read events and if they complete before the next schedule
  - Hopefully less missed events #163 
  - Deprecated filter & sensor modifiers
  - This allows slow updates for the solarman driver added in #144
- Supports additional sensors for the first inverter (typically settings)
- Support sensor groups in configuration to reduce long lists of sensors (see [here](https://github.com/kellerza/sunsynk/compare/nofilter?expand=1#diff-6ac49386a1ac7d2f06455c607732d082c55730250ed3d106663ae72 [... cut]
closed 2023-08-17 by kellerza