kellerza/sunsynk #167: max limit for max charge and discharge current

trackmafia, 2023-08-14:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py)

### You Inverter

Sunsynk 16kw

### Describe the issue/bug and what you expect

using this addon and it is working perfectly with 16kw inverter using your instructions, well done, much appreciated. 

For sensor battery max charge current and max discharge current, max is set to 185. However on 16k inverter max can go up to 275 for both. this is for modbus register 210 and 211.  I created a custom sensor to get around this. However, it shows up as a number and not a slider and i'm not sure how to change this in home assistant? Can you advise if the max can be changed or how to make the custom sensor show up with a slider as in the original?

Also, please note that 16kw has 3 MPPTs, i got around this easily by creating custom sensors for pv3_power, pv3_current and pv3_volts . Just posting here for anybody in the future.

### Logs (if applicable)

```

```

kellerza, 2023-08-15:
Thanks @trackmafia - try `NUMBER_ENTITY_MODE` for the slider - see [here](https://kellerza.github.io/sunsynk/reference/multi-options). The default `auto` mode might have decided that 285 is too much for a slider.

Do you mind doing a PR to add the 3rd MPPT?
It should be ok to change Max current as well.

trackmafia, 2023-08-16:
Kellerza,

This worked perferctly. Thanks very much. I'm new to github but i'll learn how to do pull request and submit within a week.

Thanks again for this awesome software.

deondti, 2025-01-14:
I'm running into the same issue with my 16KW so my max discharge keeps defaulting to 185 which trips the inverter when I exceed about 9Kw
Current version: 0.8.2
NUMBER_ENTITY_MODE: box

Manually entering a value above 185 it tells me it is  out of range ( 0-185)

Can you please advise how to modify the range?

deondti, 2025-01-16:
@kellerza 

src/sunsynk/definitions/single_phase.py

line 314
SENSORS += NumberRWSensor(230, "Grid Charge Battery current", AMPS, min=0, max=290)
SENSORS += NumberRWSensor(210, "Battery Max Charge current", AMPS, min=0, max=290)
SENSORS += NumberRWSensor(211, "Battery Max Discharge current", AMPS, min=0, max=290)

I CANNOT FIGURE OUT HOW TO DO A PR

kellerza, 2025-01-16:
What are your values for the following sensors?

```
    Sensor(314, "Battery Charge Limit current", AMPS, -1),
    Sensor(315, "Battery Discharge Limit current", AMPS, -1),
```

deondti, 2025-01-16:
I Don't get anything on those even after adding them as custom sensors

[11:00:42] INFO: [32mService restart after closing[0m
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
11:01:11 INFO    Loading config: /data/options.json
11:01:11 INFO    sunsynk library version: 0.8.2
11:01:11 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  30  |   90   |    80.0   |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
11:01:11 INFO    Using Single phase sensor definitions.
11:01:11 INFO    Importing /share/hass-addon-sunsynk/mysensors.py...
11:01:11 INFO      custom sensors: pv3_power, pv3_voltage, pv3_current, l1_load_current, battery_charge_limit_current, battery_discharge_limit_current
11:01:11 ERROR   Unknown sensor specified: ct_grid_10
11:01:11 INFO    Added hidden sensors as other sensors depend on it: Device type, Protocol, Serial, Rated power
11:01:11 INFO    Connecting to tcp://192.168.20.172:8899
11:01:11 INFO    Reading startup sensors Device type, Protocol, Serial
11:01:11 INFO    ############################################################
11:01:11 INFO                 Inverter serial number '****0592'
11:01:11 INFO    ############################################################
11:01:11 INFO    Device type: Single phase hybrid, using the single-phase sensor definitions
11:01:11 INFO    Protocol version: 2.1
11:01:11 INFO    Reading all sensors Device type, Protocol, Serial, PV2 power, PV3 power, Battery SOC, PV3 voltage, Radiator temperature, PV3 current, Inverter current, AUX power, Battery current, Battery power, Battery voltage, Day Battery Charge, Day Battery Discharge, Day Grid Export, Day Grid Import, Day Load Energy, Day PV Energy, Essential power, Grid Connected, Grid CT power, Grid frequency, Grid L2 power, Grid power, Grid voltage, Grid current, Inverter Power, Inverter voltage, Load frequency, Load power, Load L1 power, Load L2 power, Non-Essential power, Overall state, Priority Load, PV1 current, PV1 power, PV1 voltage, PV2 current, PV2 voltage, Use Timer, Total Battery Charge, Total Battery Discharge, Total Grid Export, Total Grid Import, Total PV Energy, Load Limit, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Date Time, Grid Charge Battery current, Grid Charge enabled, Solar Export, Battery Max Charge current, Battery Max Discharge current, Battery Shutdown Capacity, Battery Restart Capacity, Battery Low Capacity, Rated power
11:01:13 INFO    MQTT: Connecting to ICM@192.168.20.22:1884
11:01:13 INFO    MQTT: Connection successful
11:01:14 INFO    Read every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  5  |       prog3_time, prog4_capacity, prog2_capacity, prog6_time, prog4_time,        |
|     |      prog3_capacity, prog5_charge, prog4_charge, prog3_power, prog6_charge,      |
|     |   grid_charge_enabled, prog1_time, grid_charge_battery_current, prog2_charge,    |
|     |      prog1_power, load_limit, prog6_capacity, prog3_charge, priority_load,       |
|     | battery_max_charge_current, prog1_charge, battery_low_capacity, prog1_capacity,  |
|     |      prog2_power, prog4_power, prog6_power, battery_max_discharge_current,       |
|     |  prog2_time, device_type, use_timer, battery_restart_capacity, inverter_power,   |
|     | prog5_capacity, battery_shutdown_capacity, prog5_time, solar_export, prog5_power |
|  15 | load_frequency, protocol, overall_state, pv2_current, pv2_voltage, battery_soc,  |
[... cut ...]
|     |   grid_frequency, use_timer, inverter_power, prog5_capacity, battery_current,    |
|     |  battery_shutdown_capacity, prog5_time, inverter_current, radiator_temperature,  |
|     | total_battery_charge, pv3_voltage, day_grid_export, grid_current, solar_export,  |
|     |                             prog5_power, pv3_current                             |
+-----+----------------------------------------------------------------------------------+

210 and 211are reading correclty as set by the deye app
314 and 315 don't give me anything

kellerza, 2025-01-17:
> 314 = 665.0 A 315 = 665.0 A

Maybe these sensors should indicate the max?

Mine shows 200, which is safer than the fixed 290 you propose

deondti, 2025-01-17:
btw @kellerza 
I have managed to solve the problem @ mysensors.py and then adding them to Configuration definitions( I forgot that part)

#configuration part
""SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - pv3_power
  - battery_soc
  - pv3_voltage
  - radiator_temperature
  - pv3_current
  - inverter_current
  - grid_ct_power_10
  - grid_power_10
  - Battery_Charge_Limit_current
  - Battery_Discharge_Limit_current""

#mysensors.py
""from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()
#############

#############
SENSORS += (
    Sensor(188, "PV3 power", WATT, -1),
    Sensor(113, "PV3 voltage", VOLT, 0.1),
    Sensor(114, "PV3 current", AMPS, 0.1),
    Sensor(175, "L1_load_current", AMPS, 0.1),
    Sensor(314, "Battery Charge Limit current", AMPS, -1),
    Sensor(315, "Battery Discharge Limit current", AMPS, -1),
    Sensor(172, "Grid_Ct_Power_10", WATT, -10),
    Sensor(169, "Grid_power_10", WATT, -10),

    
)

SENSORS += NumberRWSensor(230, "Grid Charge Battery current", AMPS, min=0, max=290)
SENSORS += NumberRWSensor(210, "Battery Max Charge current", AMPS, min=0, max=290)
SENSORS += NumberRWSensor(211, "Battery Max Discharge current", AMPS, min=0, max=290)
""

I was hoping to include this in future updates with the PR

deondti, 2025-01-17:
> > 314 = 665.0 A 315 = 665.0 A
> 
> Maybe these sensors should indicate the max?
> 
> Mine shows 200, which is safer than the fixed 290 you propose

Max=290 is what's possibly needed and is the maximum allowed under the DEYE android APP
200 will only allow for +/-10KW, if I would be drawing 12Kw and eskom fails it would try and draw 235A from the batterys and you get a DC over current fault while the inverter shuts down.

deondti, 2025-01-17:
@kellerza 
Perhaps it's time for inverter rating selections

Sensor Definitions
Single Phase 5KW
Single Phase 8KW
Single Phase 16KW
Three phase.....
Etc.....
having different sensor options.py

See my custom sensors for 169 and 172 else it only reads eg. 500W instead of 5000W
It seems to be a issue with the 16KW and I can only assume maybe the 12KW
I'm sure the 5Kw works perfectly without the custom sensors

kellerza, 2025-03-18:
We now have single-phase-16kw

closed 2025-03-18 by kellerza