kellerza/sunsynk #169: Crash after updating HA to 2022.8.2

Yakkity, 2023-08-15:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Updated HA core and add on terminates with python error

**Expected behavior**
Normal startup

### Your environment

- Home Assistant version: 2023.8.2
- Addon version: 2023.08.06-0.4.0
- Inverter make: Sunsynk
- Inverter Model: Ecco 3.y
- Inverter Firmware: M 2.3.8.4  / S 1.5.1.5 / C E.4.2.C

### Logs

```
[Info](http://192.168.68.109:8123/hassio/addon/4bb530c7_hass-addon-sunsynk-multi/info)
[Documentation](http://192.168.68.109:8123/hassio/addon/4bb530c7_hass-addon-sunsynk-multi/documentation)
[Configuration](http://192.168.68.109:8123/hassio/addon/4bb530c7_hass-addon-sunsynk-multi/config)
[Log](http://192.168.68.109:8123/hassio/addon/4bb530c7_hass-addon-sunsynk-multi/logs)
Sunsynk/Deye Inverter Add-on (multi)
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2023-08-15 22:54:31,944 INFO    Loading HASS OS configuration
2023-08-15 22:54:31,999 INFO    Using Single phase sensor definitions.
2023-08-15 22:54:32,001 INFO    Connecting to /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB7WRZ4C-if00-port0
2023-08-15 22:54:32,001 INFO    PyModbus 3.2.2 Serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB7WRZ4C-if00-port0
2023-08-15 22:54:32,003 INFO    Serial connected.
2023-08-15 22:54:32,004 INFO    Connected to /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB7WRZ4C-if00-port0
2023-08-15 22:54:32,004 INFO    Reading startup sensors rated_power, serial, prog6_time, prog2_time, battery_low_capacity
2023-08-15 22:54:32,262 INFO    ############################################################
2023-08-15 22:54:32,262 INFO                Inverter serial number '2210185238'
2023-08-15 22:54:32,263 INFO    ############################################################
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 331, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 267, in main
    await hass_discover_sensors(ast=ast, rated_power=powr)
  File "/usr/src/app/./run.py", line 72, in hass_discover_sensors
    ents = [s.create_entity(dev) for s in ast.state.values() if not s.hidden]
  File "/usr/src/app/./run.py", line 72, in <listcomp>
    ents = [s.create_entity(dev) for s in ast.state.values() if not s.hidden]
  File "/usr/src/app/state.py", line 168, in create_entity
    options=sensor.available_values(15, self.get_state),
  File "/usr/local/lib/python3.9/site-packages/sunsynk/rwsensors.py", line 189, in available_values
    val = SSTime(string=str(resolve(self, 0))).minutes
  File "/usr/local/lib/python3.9/site-packages/sunsynk/helpers.py", line 80, in __init__
    self.str_value = string
  File "/usr/local/lib/python3.9/site-packages/sunsynk/helpers.py", line 104, in str_value
    self.minutes = int(hours) * 60 + int(minutes)
ValueError: invalid literal for int() with base 10: 'None'
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal
```

kellerza, 2023-08-16:
Will add a guard against this & log an error, but it seems that we receive an invalid value/set topic on MQTT, which is probably coming from HA.
 
Do you have any automation that changes a `progX_time` sensors? Or any `progX_` sensor?

kellerza, 2023-08-17:
Please try v0.5.0

kellerza, 2023-08-17:
The new packaging did not go as smooth as planned. Should be fixed now

closed 2023-08-20 by kellerza