kellerza/sunsynk #171: grid_connected sensor showing problem

Aimtjie, 2023-08-17:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### grid_connected sensor showing problem

the sensor grid_connected is changing between problem, ok, and unavailable

**Expected behavior**
I expected to show ok?

### Your environment

- Home Assistant version: 2022.2.8
- Addon version:2023.08.17-0.5.3
- Inverter make: Sunsynk.
- Inverter Model: 5.5kw
- Inverter Firmware: 3.3.8.4

### Logs

```txt
18:58:06 INFO    MQTT: Connection successful
19:01:35 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,243,19) 'Serial' object has no attribute 'host'
19:01:55 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,200,1) 'Serial' object has no attribute 'host'
19:08:40 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,230,3) 'Serial' object has no attribute 'host'
19:09:00 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,268,12) 'Serial' object has no attribute 'host'
19:09:50 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,230,1) 'Serial' object has no attribute 'host'
19:16:45 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,230,1) 'Serial' object has no attribute 'host'
19:17:15 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,217,3) 'Serial' object has no attribute 'host'
19:25:20 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,210,2) 'Serial' object has no attribute 'host'
19:27:26 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,611,1) 'Serial' object has no attribute 'host'
19:30:55 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,230,3) 'Serial' object has no attribute 'host'
19:31:55 ERROR   Read Error (poll_need_to_read): <class 'Exception'>: (1,200,1) 'Serial' object has no attribute 'host'
```

kellerza, 2023-08-17:
Can you paste your config?

These error seems related to the umodbus driver

louispires, 2023-08-17:
Same issue here...

`DRIVER: umodbus
INVERTERS:
  - SERIAL_NR: "123"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: serial:///dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - total_battery_charge
  - total_battery_discharge
  - total_grid_import
  - total_grid_export
  - battery_soc
  - battery_temperature
  - battery_voltage
  - battery_power
  - battery_current
  - non_essential_power
  - inverter_power
  - inverter_voltage
  - inverter_frequency
  - inverter_current
  - essential_power
  - grid_voltage
  - grid_power
  - grid_current
  - grid_frequency
  - grid_ct_power
  - overall_state
  - fault
  - environment_temperature
  - dc_transformer_temperature
  - radiator_temperature
  - day_active_energy
  - day_battery_charge
  - day_battery_discharge
  - day_grid_export
  - day_grid_import
  - day_load_energy
  - day_pv_energy
  - day_reactive_energy
  - month_grid_energy
  - month_load_energy
  - month_pv_energy
  - total_active_energy
  - total_load_energy
  - total_pv_energy
  - year_grid_export
  - year_load_energy
  - year_pv_energy
  - load_power
  - PV1_power
  - PV2_power
  - PV1_voltage
  - PV2_voltage
  - PV1_current
  - PV2_current
  - Prog1_Time
  - Prog2_Time
  - Prog3_Time
  - Prog4_Time
  - Prog5_Time
  - Prog6_Time
  - Prog1_power
  - Prog2_power
  - Prog3_power
  - Prog4_power
  - Prog5_power
  - Prog6_power
  - Prog1_Capacity
  - Prog2_Capacity
  - Prog3_Capacity
  - Prog4_Capacity
  - Prog5_Capacity
  - Prog6_Capacity
  - Prog1_charge
  - Prog2_charge
  - Prog3_charge
  - Prog4_charge
  - Prog5_charge
  - Prog6_charge
  - Prog1_mode
  - Prog2_mode
  - Prog3_mode
  - Prog4_mode
  - Prog5_mode
  - Prog6_mode
  - Use_Timer
  - Priority_Load
  - AUX_power
  - grid_connected
SENSORS_FIRST_INVERTER: []
MANUFACTURER: SS
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 60
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mosquittomqtt
MQTT_PASSWORD: 12345
DEBUG: 0
DEBUG_DEVICE: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
`

w1tw0lf, 2023-08-18:
Same issue here.

Might be because it is set as "device_class: problem". So when it is on the status shows problem and when it is off, status is ok.

kellerza, 2023-08-18:
> Same issue here.
> 
> Might be because it is set as "device_class: problem". So when it is on the status shows "problem" and when it is off, status is "ok"

🤦‍♂️ its inverted. Thx.

will remove “problem” class as the sensors work well with the power flow card

Aimtjie, 2023-08-18:
Phew at least it's nothing serious haha :-) Thank you!

kellerza, 2023-08-18:
Will be in the next release

closed 2023-08-18 by kellerza