kellerza/sunsynk #172: Sensors became unavailable

ItsRobbo, 2023-08-19:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

After update to version 2023.08.17-0.5.2 all sensors became unavailable. After deleting the :now on all sensors that had been designated to immediately update, the sensors reporting Watts started working again.
I wondered if it was therefore the "schedules" that had broken the sensors and added schedules for kwh, no_unit, a and any_unit (see log and config below). 
Subsequent to this change more sensors started working. 
Those that are still not working and showing "unavalable" are battery_soc , battery_voltage , battery_temperature ,  grid_voltage  , inverter_frequency , inverter_voltage , radiator_temperature and rated_power 

**Expected behavior**
Same sensors that worked before update, still work

### Your environment

- Home Assistant version: 2023.8.2
- Addon version: 2023.08.17-0.5.3
- Inverter make: Sunsynk
- Inverter Model: 8KW
- Inverter Firmware: COMM:e419 - MCU:3880

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "21XXXXXXX"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - total_pv_power
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - total_active_power
  - total_load_power
  - total_pv_energy
  - year_grid_export
  - year_load_power
  - year_pv_power
  - radiator_temperature
  - battery_soc
  - battery_temperature
  - battery_voltage
  - battery_current
  - total_load_energy
  - total_active_energy
  - pv1_power
  - pv1_voltage
  - pv1_current
  - pv2_power
  - pv2_voltage
  - pv2_current
  - battery_power
  - inverter_power
  - inverter_frequency
  - inverter_voltage
  - inverter_current
  - overall_state
  - grid_connected_status
  - date_time
  - essential_power
  - grid_ct_power
  - grid_ld_power
  - grid_frequency
  - grid_power
  - grid_voltage
  - grid_current
  - load_power
  - load_l1_power
  - load_l2_power
  - day_active_energy
  - day_battery_charge
  - day_battery_discharge
  - day_grid_export
  - day_grid_import
  - day_load_power
  - day_pv_energy
  - day_active_power
  - month_grid_power
  - month_pv_power
  - month_load_power
  - prog1_time
  - prog2_time
  - prog3_time
  - prog4_time
  - prog5_time
  - prog6_time
  - prog1_power
  - prog2_power
  - prog3_power
  - prog4_power
  - prog5_power
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
  - priority_load
  - load_limit
  - use_timer
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: kwh
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
  - KEY: w
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
  - KEY: no_unit
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
  - KEY: a
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 1
    CHANGE_PERCENT: 0
  - KEY: any_unit
    READ_EVERY: 15
    REPORT_EVERY: 120
    CHANGE_ANY: false
    CHANGE_BY: 0
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: XXXXXXX
MQTT_PASSWORD: XXXXXXX
DEBUG: 0
DEBUG_DEVICE: /dev/ttyS0

```

### Logs

```s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
12:32:46 INFO    Loading HASS OS configuration
12:32:46 INFO    Replaced Schedule(key='kwh', read_every=5, report_every=60, change_by=80, change_percent=0, change_any=False)
12:32:46 INFO    Replaced Schedule(key='w', read_every=5, report_every=60, change_by=80, change_percent=0, change_any=False)
12:32:46 INFO    Replaced Schedule(key='no_unit', read_every=5, report_every=60, change_by=80, change_percent=0, change_any=False)
12:32:46 INFO    Added    Schedule(key='a', read_every=5, report_every=60, change_by=1, change_percent=0, change_any=False)
12:32:46 INFO    Replaced Schedule(key='any_unit', read_every=15, report_every=120, change_by=0, change_percent=0, change_any=False)
12:32:46 INFO    Default  Schedule(key='date_time', read_every=60, report_every=60, change_by=0, change_percent=0, change_any=True)
12:32:46 INFO    Default  Schedule(key='rated_power', read_every=0, report_every=0, change_by=0, change_percent=0, change_any=False)
12:32:46 INFO    Default  Schedule(key='serial', read_every=0, report_every=0, change_by=0, change_percent=0, change_any=False)
12:32:46 INFO    Default  Schedule(key='rw', read_every=5, report_every=300, change_by=0, change_percent=0, change_any=True)
12:32:46 INFO    Using Single phase sensor definitions.
12:32:46 ERROR   Your config includes deprecated sensors. Replace total_pv_power with total_pv_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace total_active_power with total_active_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace total_load_power with total_load_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace year_load_power with year_load_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace year_pv_power with year_pv_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace grid_connected_status with grid_connected
12:32:46 ERROR   Your config includes deprecated sensors. Replace day_load_power with day_load_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace day_active_power with day_active_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace month_grid_power with month_grid_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace month_pv_power with month_pv_energy
12:32:46 ERROR   Your config includes deprecated sensors. Replace month_load_power with month_load_energy
12:32:46 INFO    Connecting to /dev/ttyUSB0
12:32:46 INFO    PyModbus 3.4.1 Serial: /dev/ttyUSB0
12:32:46 INFO    Reading startup sensors Prog3 Time, Prog2 Time, Battery Low Capacity, Rated power, Serial, Prog4 Time, Prog6 Time, Prog1 Time, Prog5 Time
12:32:46 INFO    ############################################################
12:32:46 INFO                Inverter serial number '21XXXXXXXX'
12:32:46 INFO    ############################################################
12:32:46 INFO    MQTT: Connecting to mqttXXXX@core-mosquitto:1883
12:32:46 INFO    MQTT: Connection successful

```

kellerza, 2023-08-19:
Can you address the errors in the log? 

Are you sure you don’t have duplicate entries for these not working? I.e sensor.ss_battery_temperature2

ItsRobbo, 2023-08-20:
Thanks so much for looking into this.
Cleaned up the errors and unfortunately just the Watts working, all other sensors are "Unknown" in the MQTT. The log below is from that version.

I then tried adding further schedules. Each time I stopped the addon, changed the YAML, rebuilt the addon, started it and then restarted home assistant. Adding kWh , a and no_unit helped getting some of the other sensors back but then adding any_unit reversed all that progress and went to no sensors reporting at all.

I read through all the sensors again and dont see any duplicates, I have posted the config again as it stands with just the Watts working.

Logs
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
13:06:51 INFO    Loading HASS OS configuration
13:06:51 INFO    Replaced Schedule(key='w', read_every=5, report_every=60, change_by=80, change_percent=0, change_any=False)
13:06:51 INFO    Default  Schedule(key='date_time', read_every=60, report_every=60, change_by=0, change_percent=0, change_any=True)
13:06:51 INFO    Default  Schedule(key='rated_power', read_every=0, report_every=0, change_by=0, change_percent=0, change_any=False)
13:06:51 INFO    Default  Schedule(key='serial', read_every=0, report_every=0, change_by=0, change_percent=0, change_any=False)
13:06:51 INFO    Default  Schedule(key='rw', read_every=5, report_every=300, change_by=0, change_percent=0, change_any=True)
13:06:51 INFO    Default  Schedule(key='kwh', read_every=300, report_every=300, change_by=0, change_percent=0, change_any=False)
13:06:51 INFO    Default  Schedule(key='any_unit', read_every=15, report_every=300, change_by=0, change_percent=0, change_any=False)
13:06:51 INFO    Default  Schedule(key='no_unit', read_every=15, report_every=300, change_by=0, change_percent=0, change_any=True)
13:06:51 INFO    Using Single phase sensor definitions.
13:06:51 INFO    Connecting to /dev/ttyUSB0
13:06:51 INFO    PyModbus 3.4.1 Serial: /dev/ttyUSB0
13:06:51 INFO    Reading startup sensors Serial, Battery Low Capacity, Prog3 Time, Prog2 Time, Prog4 Time, Prog6 Time, Prog1 Time, Rated power, Prog5 Time
13:06:51 INFO    ############################################################
13:06:51 INFO                Inverter serial number '21XXXXXXXX'
13:06:51 INFO    ############################################################
13:06:52 INFO    MQTT: Connecting to XXXX@core-mosquitto:1883
13:06:52 INFO    MQTT: Connection successful
```
Config
```
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "21XXXXXXX"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - total_pv_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - total_active_energy
  - total_load_energy
  - total_pv_energy
  - year_grid_export
  - year_load_energy
  - year_pv_energy
  - radiator_temperature
  - battery_soc
  - battery_temperature
  - battery_voltage
  - battery_current
  - total_load_energy
  - total_active_energy
  - pv1_power
  - pv1_voltage
  - pv1_current
  - pv2_power
  - pv2_voltage
  - pv2_current
  - battery_power
  - inverter_power
  - inverter_frequency
  - inverter_voltage
  - inverter_current
  - overall_state
  - grid_connected
  - date_time
  - essential_power
  - grid_ct_power
  - grid_ld_power
  - grid_frequency
  - grid_power
  - grid_voltage
  - grid_current
  - load_power
  - load_l1_power
  - load_l2_power
  - day_active_energy
  - day_battery_charge
  - day_battery_discharge
  - day_grid_export
  - day_grid_import
  - day_load_energy
  - day_pv_energy
  - day_active_energy
  - month_grid_energy
  - month_pv_energy
  - month_load_energy
  - prog1_time
  - prog2_time
  - prog3_time
  - prog4_time
  - prog5_time
  - prog6_time
  - prog1_power
  - prog2_power
  - prog3_power
  - prog4_power
  - prog5_power
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
  - priority_load
  - load_limit
  - use_timer
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: w
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: 
MQTT_PASSWORD: 
DEBUG: 0
DEBUG_DEVICE: /dev/ttyS0
 ```

kellerza, 2023-08-20:
No need to restart Home Assistant. It’s not related to schedules as these are clearly printed in the log.

try adding debug and posting the log.

also post a screenshot of the SS device+sensors.

another option would be to look at the MQTT discovery options with mqtt-explorer. Delete every /homeassistant/x/serial branch. What other serials do you have

ItsRobbo, 2023-08-20:
I'm going to apologise upfront, I'm learning as Im going on this. What do you mean in the final part by deleting every /homeassistant/x/serial branch ? The only devices under MQTT are for the alarm through Paradox Alarm Interface (PAI) and the Sunsynk.

I added debug and got the below. I do see reference to SOC and temperatures so hopefully that helps.

```
2023-08-20 19:54:13.341 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on SUNSYNK/status/2105307122/availability (qos=0): b'online'
2023-08-20 19:54:13.595 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/sensor/2105307122/total_pv_energy/config (qos=0): b'{"unique_id": "2105307122_total_pv_energy", "device": {"identifiers": ["2105307122"], "manufacturer": "Sunsynk", "model": "8kW Inverter (**7122)", "name": "SS"}, "state_topic": "SUNSYNK/status/2105307122/total_pv_energy", "name": "Total PV Energy", "availability": [{"topic": "SUNSYNK/status/2105307122/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing", "object_id": "ss_total_pv_energy"}'
2023-08-20 19:54:13.595 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'unique_id': '2105307122_total_pv_energy', 'device': {'identifiers': ['2105307122'], 'manufacturer': 'Sunsynk', 'model': '8kW Inverter (**7122)', 'name': 'SS'}, 'state_topic': 'SUNSYNK/status/2105307122/total_pv_energy', 'name': 'Total PV Energy', 'availability': [{'topic': 'SUNSYNK/status/2105307122/availability'}], 'device_class': 'energy', 'unit_of_measurement': 'kWh', 'state_class': 'total_increasing', 'object_id': 'ss_total_pv_energy', 'platform': 'mqtt'}
2023-08-20 19:54:13.595 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 2105307122 total_pv_energy, sending update
2023-08-20 19:54:13.596 DEBUG (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('sensor', '2105307122 total_pv_energy') '{'unique_id': '2105307122_total_pv_energy', 'device': {'identifiers': ['2105307122'], 'manufacturer': 'Sunsynk', 'model': '8kW Inverter (**7122)', 'name': 'SS'}, 'state_topic': 'SUNSYNK/status/2105307122/total_pv_energy', 'name': 'Total PV Energy', 'availability': [{'topic': 'SUNSYNK/status/2105307122/availability'}], 'device_class': 'energy', 'unit_of_measurement': 'kWh', 'state_class': 'total_increasing', 'object_id': 'ss_total_pv_energy', 'platform': 'mqtt'}'
2023-08-20 19:54:13.596 DEBUG (MainThread) [homeassistant.components.mqtt.mixins] Ignoring unchanged update for: sensor.ss_total_pv_energy
2023-08-20 19:54:13.596 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('sensor', '2105307122 total_pv_energy'): deque([])
2023-08-20 19:54:13.619 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/sensor/2105307122/total_battery_charge/config (qos=0): b'{"unique_id": "2105307122_total_battery_charge", "device": {"identifiers": ["2105307122"], "manufacturer": "Sunsynk", "model": "8kW Inverter (**7122)", "name": "SS"}, "state_topic": "SUNSYNK/status/2105307122/total_battery_charge", "name": "Total Battery Charge", "availability": [{"topic": "SUNSYNK/status/2105307122/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing", "object_id": "ss_total_battery_charge"}'
2023-08-20 19:54:13.623 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'unique_id': '2105307122_total_battery_charge', 'device': {'identifiers': ['2105307122'], 'manufacturer': 'Sunsynk', 'model': '8kW Inverter (**7122)', 'name': 'SS'}, 'state_topic': 'SUNSYNK/status/2105307122/total_battery_charge', 'name': 'Total Battery Charge', 'availability': [{'topic': 'SUNSYNK/status/2105307122/availability'}], 'device_class': 'energy', 'unit_of_measurement': 'kWh', 'state_class': 'total_increasing', 'object_id': 'ss_total_battery_charge', 'platform': 'mqtt'}
2023-08-20 19:54:13.623 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 2105307122 total_battery_charge, sending update
2023-08-20 19:54:13.623 DEBUG (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('sensor', '2105307122 total_battery_charge') '{'unique_id': '2105307122_total_battery_charge', 'device': {'identifiers': ['2105307122'], 'manufacturer': 'Sunsynk', 'model': '8kW Inverter (**7122)', 'name': 'SS'}, 'state_topic': 'SUNSYNK/status/2105307122/total_battery_charge', 'name': 'Total Battery Charge', 'availability': [{'topic': 'SUNSYNK/status/2105307122/availability'}], 'device_class': 'energy', 'unit_of_measurement': 'kWh', 'state_class': 'total_increasing', 'object_id': 'ss_total_battery_charge', 'platform': 'mqtt'}'
2023-08-20 19:54:13.623 DEBUG (MainThread) [homeassistant.components.mqtt.mixins] Ignoring unchanged update for: sensor.ss_total_battery_charge
2023-08-20 19:54:13.623 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('sensor', '2105307122 total_battery_charge'): deque([])
2023-08-20 19:54:13.624 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/sensor/2105307122/total_battery_discharge/config (qos=0): b'{"unique_id": "2105307122_total_battery_discharge", "device": {"identifiers": ["2105307122"], "manufacturer": "Sunsynk", "model": "8kW Inverter ( [... cut ...]  'max': 8000, 'object_id': 'ss_prog3_power', 'platform': 'mqtt'}
2023-08-20 19:54:15.133 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: number 2105307122 prog3_power, sending update
2023-08-20 19:54:15.133 DEBUG (MainThread) [homeassistant.components.mqtt.mixins] Got update for entity with hash: ('number', '2105307122 prog3_power') '{'unique_id': '2105307122_prog3_power', 'device': {'identifiers': ['2105307122'], 'manufacturer': 'Sunsynk', 'mod

ItsRobbo, 2023-08-20:
And here are screenshots:

[image]

[image]

[image]

kellerza, 2023-08-20:
Unavailable is not quite the same as Unknown…

are you still restarting?

what happens if you simply restart the addon?

or wait for 5min (worst case schedule)

kellerza, 2023-08-20:
When you restart HA sensors are Unknown until the next update

ItsRobbo, 2023-08-20:
Apologies for the unavailable in subject, yes "Unknown" . (tried to correct myself in the post a few hours ago).

I reset the addon to the default schedule settings before sending you the logs, but no HA restart.
But now after 30 mins of HA and the addon running without being touched, and the house using battery, the SOC% does not appear. It has also been left overnight last night and no % change being reported. I can see the % running down on the solarman app. Also it was only post the autoupdate on Thursday that they became unknown.

I clicked restart, have waited 10 minutes, no change to the above sensor states. WIll not touch until morning again.

Thanks again for the help!

ItsRobbo, 2023-08-21:
So my HA restarts every morning at 2am. When I woke up this morning the kwh had now become unknown as well. I think they had probably reported during the day yesterday when I was fiddling with schedules and then frozen at that amount? So this morning I have just the Watts reporting again. Here is the log from the 2am restart. Not sure how mcuh data might help

```
2023-08-21 02:01:02.145 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received retained message on homeassistant/sensor/SunSynk/voltage_inverter/config (qos=0): b'{"unit_of_measurement":"V","state_topic":"homeassistant/sensor/SunSynk/voltage_inverter/value","name":"SunSynk_voltage_inverter","unique_id":"voltage_inverter","last_reset_topic":"SunSynk/lastreset","last_reset_value_template":"1970-01-01T00:00:00+00:00","device":{"identifiers":["SunSynk"],"name":"SunSynk","model":"SynSynk","manufacturer":"SunSynk"},"device_class":"voltage"}'
2023-08-21 02:01:02.146 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'unit_of_measurement': 'V', 'state_topic': 'homeassistant/sensor/SunSynk/voltage_inverter/value', 'name': 'SunSynk_voltage_inverter', 'unique_id': 'voltage_inverter', 'last_reset_topic': 'SunSynk/lastreset', 'last_reset_value_template': '1970-01-01T00:00:00+00:00', 'device': {'identifiers': ['SunSynk'], 'name': 'SunSynk', 'model': 'SynSynk', 'manufacturer': 'SunSynk'}, 'device_class': 'voltage', 'platform': 'mqtt'}
2023-08-21 02:01:02.146 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor SunSynk voltage_inverter
2023-08-21 02:01:02.147 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received retained message on homeassistant/sensor/SunSynk/voltage_grid/config (qos=0): b'{"unit_of_measurement":"V","state_topic":"homeassistant/sensor/SunSynk/voltage_grid/value","name":"SunSynk_voltage_grid","unique_id":"voltage_grid","last_reset_topic":"SunSynk/lastreset","last_reset_value_template":"1970-01-01T00:00:00+00:00","device":{"identifiers":["SunSynk"],"name":"SunSynk","model":"SynSynk","manufacturer":"SunSynk"},"device_class":"voltage"}'
2023-08-21 02:01:02.147 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'unit_of_measurement': 'V', 'state_topic': 'homeassistant/sensor/SunSynk/voltage_grid/value', 'name': 'SunSynk_voltage_grid', 'unique_id': 'voltage_grid', 'last_reset_topic': 'SunSynk/lastreset', 'last_reset_value_template': '1970-01-01T00:00:00+00:00', 'device': {'identifiers': ['SunSynk'], 'name': 'SunSynk', 'model': 'SynSynk', 'manufacturer': 'SunSynk'}, 'device_class': 'voltage', 'platform': 'mqtt'}
2023-08-21 02:01:02.147 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor SunSynk voltage_grid
2023-08-21 02:01:02.147 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received retained message on homeassistant/sensor/SunSynk/battery_volt/config (qos=0): b'{"unit_of_measurement":"V","state_topic":"homeassistant/sensor/SunSynk/battery_volt/value","name":"SunSynk_battery_volt","unique_id":"battery_volt","last_reset_topic":"SunSynk/lastreset","last_reset_value_template":"1970-01-01T00:00:00+00:00","device":{"identifiers":["SunSynk"],"name":"SunSynk","model":"SynSynk","manufacturer":"SunSynk"},"device_class":"voltage"}'
2023-08-21 02:01:02.148 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'unit_of_measurement': 'V', 'state_topic': 'homeassistant/sensor/SunSynk/battery_volt/value', 'name': 'SunSynk_battery_volt', 'unique_id': 'battery_volt', 'last_reset_topic': 'SunSynk/lastreset', 'last_reset_value_template': '1970-01-01T00:00:00+00:00', 'device': {'identifiers': ['SunSynk'], 'name': 'SunSynk', 'model': 'SynSynk', 'manufacturer': 'SunSynk'}, 'device_class': 'voltage', 'platform': 'mqtt'}
2023-08-21 02:01:02.148 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor SunSynk battery_volt
2023-08-21 02:01:02.148 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received retained message on homeassistant/sensor/SunSynk/battery_charge_volts/config (qos=0): b'{"unit_of_measurement":"V","state_topic":"homeassistant/sensor/SunSynk/battery_charge_volts/value","name":"SunSynk_battery_charge_volts","unique_id":"battery_charge_volts","last_reset_topic":"SunSynk/lastreset","last_reset_value_template":"1970-01-01T00:00:00+00:00","device":{"identifiers":["SunSynk"],"name":"SunSynk","model":"SynSynk","manufacturer":"SunSynk"},"device_class":"voltage"}'
2023-08-21 02:01:02.148 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'unit_of_measurement': 'V', 'state_topic': 'homeassistant/sensor/SunSynk/battery_charge_volts/value', 'name': 'SunSynk_battery_charge_volts', 'unique_id': 'battery_charge_volts', 'last_reset_topic': 'SunSynk/lastreset', 'last_reset_value_template': '1970-01-01T00:00:00+00:00', 'device': {'identifiers': ['SunSynk'], 'name': 'SunSynk', 'model': 'SynSynk', 'manufacturer': 'SunSynk'}, 'device_class': 'voltage', 'platform': 'mqtt'}
2023-08-21 02:01:02.148 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor SunSynk battery_charge_volts
2023-08-21 02:01:02.149 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received retained message on homeassistant/sensor/SunSynk/Solar_PV1_volts/config (qos=0): b'{"unit_of_measurement":"V","state_topic":"homeassistant/sensor/SunSynk/Solar_PV1_volts/value","name":"S [... cut ...]
2023-08-21 02:01:03.889 DEBUG (MainThread) [homeassistant.components.mqtt.client] Subscribing to SUNSYNK/status/2105307122/priority_load, mid: 2, qos: 0
2023-08-21 02:01:03.889 DEBUG (MainThread) [homeassistant.components.mqtt.client] Subscribing to SUNSYNK/status/2105307122/use_timer, mid: 2, qos: 0
2023-08-21 02:01:03.889 DEBUG (MainThread) [homeassistant.components.mqtt.client] Subscribing to SUNSYNK/status/2105307122/prog1_time, mid: 2, qos: 0
2023-08-21 02:01:03.889 DEBUG (MainThread) [homeassistant.component

kellerza, 2023-08-22:
@ItsRobbo -please update to the latest version - the logs you are posting is the Home Assistant Logs. The Addon's logs wil be more useful (last tab ion the addon screen)

Can you also post the output of the "Callback stats" sensor - could be `sensor.ss_cb_stats` **with the attributes**

[image]

N1c084, 2023-08-23:
Hi 
I have loose the grid connected status from last update
Entity become not available 
My config line that working before update was: grid_connected_status:last

Thanks

kellerza, 2023-08-23:
Read the logs:

```
WARNING Modifiers was replaced by schedules: grid_connected_status:now
ERROR   Your config includes deprecated sensors. Replace grid_connected_status with grid_connected
```
The new one is a binary_sensor, not sensor

N1c084, 2023-08-23:
Thanks 
Sorry not see that message on my phone app.

ItsRobbo, 2023-08-24:
Thanks again for the assistance and apologies for the delay. I had network issues and a whole bunch of things were giving errors so wanted that to be sorted out before moving on.
I installed the latest addon, 2023.08.23-0.5.5

Call back stats as follows, no long calls?

[image]

ItsRobbo, 2023-08-24:
The log within the addon is below. Same issue continues that only the Watt sensors reporting.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
17:35:20 INFO    Loading HASS OS configuration
17:35:21 INFO    Schedules:
+-------------+-----+------+--------+-----------+----------+------------+
|     Key     | src | Read | Report | Change by | Change % | Change any |
+-------------+-----+------+--------+-----------+----------+------------+
|  date_time  |     |  60  |   60   |           |          |    True    |
| rated_power |     | once |        |           |          |            |
|    serial   |     | once |        |           |          |            |
|      rw     |     |  5   |  300   |           |          |    True    |
|      w      |  *  |  5   |   60   |     80    |          |            |
|     kwh     |     | 300  |  300   |           |          |            |
|   any_unit  |     |  15  |  300   |           |          |            |
|   no_unit   |     |  15  |  300   |           |          |    True    |
+-------------+-----+------+--------+-----------+----------+------------+
17:35:21 INFO    Using Single phase sensor definitions.
17:35:21 INFO    Connecting to /dev/ttyUSB0
17:35:21 INFO    PyModbus 3.4.1 Serial: /dev/ttyUSB0
17:35:21 INFO    Reading startup sensors Rated power, Prog6 Time, Prog2 Time, Prog4 Time, Serial, Battery Low Capacity, Prog3 Time, Prog1 Time, Prog5 Time
17:35:21 INFO    ############################################################
17:35:21 INFO                 Inverter serial number '****7122'
17:35:21 INFO    ############################################################
17:35:21 INFO    MQTT: Connecting to mqttuser@core-mosquitto:1883
17:35:21 INFO    MQTT: Connection successful
17:35:22 INFO    Read every (first)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  5  | priority_load, use_timer, prog6_time, prog5_capacity, prog5_charge, prog4_time, |
|     |  prog4_power, battery_power, load_power, prog1_capacity, battery_low_capacity,  |
|     |     prog6_capacity, prog5_time, inverter_power, prog1_power, prog3_capacity,    |
|     |      prog2_power, pv2_power, prog2_capacity, prog4_capacity, load_l2_power,     |
|     |  grid_ct_power, prog3_time, prog4_charge, prog2_charge, prog3_power, pv1_power, |
|     |       load_l1_power, prog2_time, prog3_charge, prog5_power, grid_ld_power,      |
|     |       prog6_power, prog1_time, essential_power, load_limit, prog1_charge,       |
|     |                             prog6_charge, grid_power                            |
|  15 |     pv2_voltage, pv1_voltage, battery_current, pv2_current, grid_connected,     |
|     |   battery_temperature, battery_soc, pv1_current, grid_voltage, overall_state,   |
|     |    inverter_frequency, radiator_temperature, inverter_voltage, grid_current,    |
|     |                battery_voltage, inverter_current, grid_frequency                |
|  60 |                                    date_time                                    |
| 300 |  year_load_energy, day_battery_discharge, day_battery_charge, month_pv_energy,  |
|     |     total_grid_import, year_grid_export, month_grid_energy, day_grid_import,    |
|     | total_pv_energy, total_load_energy, month_load_energy, total_battery_discharge, |
|     |     total_active_energy, day_active_energy, day_load_energy, year_pv_energy,    |
|     |     day_grid_export, total_grid_export, total_battery_charge, day_pv_energy     |
+-----+---------------------------------------------------------------------------------+
17:35:22 INFO    Report every (first)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  60 |       essential_power, pv1_power, load_l1_power, pv2_power, battery_power,       |
|     | grid_ld_power, load_power, load_l2_power, date_time, grid_ct_power, grid_power,  |
|     |                                  inverter_power                                  |
| 300 |  priority_load, use_timer, prog6_time, prog5_capacity, pv1_current, prog4_time,  |
|     |           prog5_charge, overall_state, prog4_power, year_load_energy,            |
|     |   day_battery_discharge, prog1_capacity, battery_low_capacity, prog6_capacity,   |
|     |    day_battery_charge, prog5_time, prog1_power, pv1_voltage, prog3_capacity,     |
|     | month_pv_energy, total_grid_import, year_grid_export, prog2_power, battery_soc,  |
|     |       month_grid_energy, prog2_capacity, prog4_capacity, day_grid_import,        |
|     |        total_pv_energy, total_load_energy, month_load_energy, prog3_time,        |
[... cut ...]
|     |    day_load_energy, year_pv_energy, pv2_voltage, prog1_time, grid_frequency,     |
|     |       battery_current, day_grid_export, grid_connected, total_grid_export,       |
|     |      total_battery_charge, battery_temperature, day_pv_energy, load_limit,       |
|     | prog1_charge, prog6_charge, inverter_frequency, inverter_voltage, grid_current,  |
|     |                                 inverter_current                                 |
+-----+----------------------------------------------------------------------------------+
```

kellerza, 2023-08-25:
Do you get any read errors after this nice clean log you posted above?

ItsRobbo, 2023-08-25:
Maybe this is where I am misunderstanding, the addon logs dont change after initial load. Thats why I posted the HA logs where I could see the DEBUG messages posted.

kellerza, 2023-08-26:
The addon logs should not change if reads are successful.

can you try adjusting read_allow_gap down to 1?

ItsRobbo, 2023-08-26:
FIXED!
that read_allow_gap down to 1 fixed it immediately.
Thanks so much for the help!

closed 2023-08-26 by ItsRobbo