kellerza/sunsynk #173: Modbus Errors [Connection] Connection lost during request

dandare100, 2023-08-24:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Since upgrading to 2023.08.23-0.5.5 from 2023.08.17-0.5.3 I get these messages in the logs now

```
20:07:40 ERROR   Read Error (poll_need_to_read): (1,166,13) Modbus Error: [Connection] Connection lost during request
20:07:45 ERROR   Read Error (poll_need_to_read): (1,90,2) Modbus Error: [Connection] Connection lost during request
20:07:55 ERROR   Exception in read_ss: Multiple Modbus read errors: (1,166,13) Modbus Error: [Connection] Connection lost during request
20:08:00 ERROR   Exception in read_ss: Multiple Modbus read errors: (1,90,2) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (18 in 5 min) Read Error (poll_need_to_read): (1,166,13) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (16 in 5 min) Exception in read_ss: Multiple Modbus read errors: (1,166,13) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (8 in 5 min) Read Error (poll_need_to_read): (1,90,2) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (7 in 5 min) Exception in read_ss: Multiple Modbus read errors: (1,90,2) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   Read Error (poll_need_to_read): (1,72,11) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   Exception in read_ss: Multiple Modbus read errors: (1,72,11) Modbus Error: [Connection] Connection lost during request
```

**Expected behavior**
I didn't used to get these errors

### Your environment

- Home Assistant version: 2023.7.1
- Addon:
  - Name: sunsynk-multi
  - Version:  2023.08.23-0.5.5
- Inverter:
  - Make: Sunsynk 8kW
  - Model: 8kW
  - Firmware: \_\_
- Adaptor details:
  - RS485/RS232/solarman
  

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2200000000"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: tcp://homeassistant.local:502
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - total_pv_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - battery_soc
  - grid_voltage
  - load_power
  - grid_power
  - pv1_power
  - pv2_power
  - radiator_temperature
  - dc_transformer_temperature
  - battery_power
  - battery_current
  - essential_1_power
  - non-essential_power
  - pv1_voltage
  - pv2_voltage
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: sunsynk
MQTT_PASSWORD: redacted
DEBUG: 0
DEBUG_DEVICE: /dev/ttyAMA0

```

### Logs

```logs

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
20:07:30 INFO    Loading HASS OS configuration
20:07:30 INFO    Schedules:
+-------------+-----+------+--------+-----------+----------+------------+
|     Key     | src | Read | Report | Change by | Change % | Change any |
+-------------+-----+------+--------+-----------+----------+------------+
|  date_time  |     |  60  |   60   |           |          |    True    |
| rated_power |     | once |        |           |          |            |
|    serial   |     | once |        |           |          |            |
|      rw     |     |  5   |  300   |           |          |    True    |
|      w      |  *  |  5   |   60   |     80    |          |            |
|     kwh     |     | 300  |  300   |           |          |            |
|   any_unit  |     |  15  |  300   |           |          |            |
|   no_unit   |     |  15  |  300   |           |          |    True    |
+-------------+-----+------+--------+-----------+----------+------------+
20:07:30 INFO    Using Single phase sensor definitions.
20:07:30 INFO    Connecting to tcp://homeassistant.local:502
20:07:30 INFO    PyModbus 3.4.1 tcp: homeassistant.local:502
20:07:30 INFO    Reading startup sensors Rated power, Serial
20:07:30 INFO    ############################################################
20:07:30 INFO                 Inverter serial number '****2100'
20:07:30 INFO    ############################################################
20:07:30 INFO    MQTT: Connecting to sunsynk@core-mosquitto:1883
20:07:30 INFO    MQTT: Connection successful
20:07:31 INFO    Read every (first)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  5  |  pv1_power, battery_power, grid_power, non_essential_power, essential_1_power,  |
|     |                              pv2_power, load_power                              |
|  15 |   pv1_voltage, battery_soc, dc_transformer_temperature, radiator_temperature,   |
|     |                    pv2_voltage, battery_current, grid_voltage                   |
| 300 | total_grid_export, total_pv_energy, total_grid_import, total_battery_discharge, |
|     |                               total_battery_charge                              |
+-----+---------------------------------------------------------------------------------+
20:07:31 INFO    Report every (first)
+-----+-------------------------------------------------------------------------------+
|  s  |                                    Sensors                                    |
+-----+-------------------------------------------------------------------------------+
|  60 | pv1_power, battery_power, grid_power, non_essential_power, essential_1_power, |
|     |                             pv2_power, load_power                             |
| 300 |    total_grid_export, battery_soc, pv1_voltage, dc_transformer_temperature,   |
|     |          total_pv_energy, total_grid_import, total_battery_discharge,         |
|     |   total_battery_charge, radiator_temperature, pv2_voltage, battery_current,   |
|     |                                  grid_voltage                                 |
+-----+-------------------------------------------------------------------------------+
20:07:40 ERROR   Read Error (poll_need_to_read): (1,166,13) Modbus Error: [Connection] Connection lost during request
20:07:45 ERROR   Read Error (poll_need_to_read): (1,90,2) Modbus Error: [Connection] Connection lost during request
20:07:55 ERROR   Exception in read_ss: Multiple Modbus read errors: (1,166,13) Modbus Error: [Connection] Connection lost during request
20:08:00 ERROR   Exception in read_ss: Multiple Modbus read errors: (1,90,2) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (18 in 5 min) Read Error (poll_need_to_read): (1,166,13) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (16 in 5 min) Exception in read_ss: Multiple Modbus read errors: (1,166,13) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (8 in 5 min) Read Error (poll_need_to_read): (1,90,2) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   (7 in 5 min) Exception in read_ss: Multiple Modbus read errors: (1,90,2) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   Read Error (poll_need_to_read): (1,72,11) Modbus Error: [Connection] Connection lost during request
20:10:00 ERROR   Exception in read_ss: Multiple Modbus read errors: (1,72,11) Modbus Error: [Connection] Connection lost during request
20:15:00 ERROR   (40 in 5 min) Read Error (poll_need_to_read): (1,166,13) Modbus Error: [Connection] Connection lost during request
20:15:00 ERROR   (40 in 5 min) Exception in read_ss: Multiple Modbus read errors: (1,166,13) Modbus Error: [Connection] Connection lost during request
20:15:00 ERROR   (19 in 5 min) Read Error (poll_need_to_read): (1,90,2) Modbus Error: [Connection] Connection lost during request
20:15:00 ERROR   (19 in 5 min) Exception in read_ss: Multiple Modbus read errors: (1,90,2) Modbus Error: [Connection] Connection lost during request

```

kellerza, 2023-08-26:
Can you try adjusting READ_ALLOW_GAP down to 1?

kellerza, 2023-08-26:
You can also try updating to 0.5.6

dandare100, 2023-08-27:
Howzit. Thank you for the prompt response and for a lekker plugin.

Adjusting the READ_ALLOW_GAP to 1 did the trick. It is sorted now and running smoothly

closed 2023-08-27 by dandare100

kellerza, 2023-08-27:
🚀 Glad it's working!

With 0.5.7 calculating the batch size should work even with a gap, but you only need to adjust this if you are having read issues/specific modbus gateways