kellerza/sunsynk #176: How to enable essential load 1 & 2 on HA Add-On?

k2dp2k, 2023-08-30:
Hey hoo,

how can i enable the two sensors on HA-Add-On?
My inverter is running off-grid, without pv connected, it is fed by an mi on the gen port, so the standard essential load delivers wrong values...

My best :-)

kellerza, 2023-08-30:
Hi, maybe load_1_power?

https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py#L63

k2dp2k, 2023-08-30:
Nope, thats not working... non of the load 1,2 ore essential 1,2 are coming up when i define them in den add-on...

kellerza, 2023-08-30:
Load_l1_power?

k2dp2k, 2023-08-31:
Thats working ;-) Thank you!
Btw... all the essential loads are displaying the same value, essential 2 is displaying an positive value instead of a negative one...

kellerza, 2023-08-31:
Great!

You probably want to look at `aux_power` as well. Since you have a fairly unique setup it might take some experimenting to see which sensors works for you (the sensors simply translate to registers on the inverter)

closed 2023-08-31 by kellerza