kellerza/sunsynk #177: Constant Attribute Error's in Log & only 5 sensors showing/available

duncthorp, 2023-08-31:
Sunsynk / mbusd Home Assistant Add-On

Firstly, great work on this... 

The add-on seems to work fine and is receiving data from the inverter correctly but I'm getting constant errors within the log:

03:55:00 ERROR   (1 in 5 min) OSError in read_ss: AttributeError reading 2 registers from 96: 'Serial' object has no attribute 'host' (poll_need_to_read)
04:00:00 ERROR   (5 in 5 min) OSError in read_ss: AttributeError reading 1 registers from 184: 'Serial' object has no attribute 'host' (poll_need_to_read)
04:05:00 ERROR   (4 in 5 min) OSError in read_ss: AttributeError reading 1 registers from 184: 'Serial' object has no attribute 'host' (poll_need_to_read)
04:05:00 ERROR   (1 in 5 min) OSError in read_ss: AttributeError reading 4 registers from 72: 'Serial' object has no attribute 'host' (poll_need_to_read)
04:10:00 ERROR   (2 in 5 min) OSError in read_ss: AttributeError reading 1 registers from 184: 'Serial' object has no attribute 

These seem to repeat every 5 mins or so. 
Maybe a related issue but I've also only got a few sensors available. So total pv & battery power but cannot 'see' any of the others??

### config
DRIVER: umodbus
INVERTERS:
  - SERIAL_NR: "220*********"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: serial:///dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - total_pv_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - battery_soc
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 50
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: homeassistant
MQTT_PASSWORD: *******************
DEBUG: 0
DEBUG_DEVICE: /dev/ttyAMA0

### Your environment
- Home Assistant version: 2023.8.x
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter:
  - Make: Sunsynk
  - Model: 5.5kw ecco
- Adaptor details:
  - RS485 - Wave USB-to-RS485

kellerza, 2023-08-31:
The pymodbus driver works better with direct serial

duncthorp, 2023-08-31:
Thanks for the reply. 
Any idea what settings I need to use for the pymodbus driver? 
Also, how can I get all the sensors showing?
Really appreciate your help
Ta

kellerza, 2023-08-31:
1. Click on pymodbus - and change the port https://kellerza.github.io/sunsynk/reference/multi-options#port
2. Use sensor groups - https://kellerza.github.io/sunsynk/reference/definitions#groups-of-sensors
3. If you still want more sensors, read the rest of that page + get the names from the definition file

closed 2023-08-31 by kellerza