kellerza/sunsynk #178: Failure to install

fbasson, 2023-09-04:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

I managed to install a previous version earlier this evening.  Later, an update was detected by the supervisor.  When trying to update, I received the below error.  

Logger: homeassistant.components.websocket_api.http.connection
Source: components/hassio/update.py:169
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: September 4, 2023 at 10:52:53 PM (4 occurrences)
Last logged: September 4, 2023 at 11:43:00 PM

[547420111168] Error updating Sunsynk/Deye Inverter Add-on (multi): Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:v0.5.8: 500 Server Error for http+docker://localhost/v1.42/images/create?tag=v0.5.8&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 167, in async_install
    await async_update_addon(self.hass, slug=self._addon_slug, backup=backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 52, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:v0.5.8: 500 Server Error for http+docker://localhost/v1.42/images/create?tag=v0.5.8&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1974, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2011, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 421, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 169, in async_install
    raise HomeAssistantError(f"Error updating {self.title}: {err}") from err
homeassistant.exceptions.HomeAssistantError: Error updating Sunsynk/Deye Inverter Add-on (multi): Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:v0.5.8: 500 Server Error for http+docker://localhost/v1.42/images/create?tag=v0.5.8&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")

I then uninstalled the pervious version and try to install the latest version, which resulted in the error message below.  I get the impression the version of update in the update instruction is somehow wrong.

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:130
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: September 4, 2023 at 11:07:54 PM (11 occurrences)
Last logged: September 4, 2023 at 11:58:25 PM

Failed to to call /addons/4bb530c7_hass-addon-sunsynk-multi/install - Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:v0.5.8: 500 Server Error for http+docker://localhost/v1.42/images/create?tag=v0.5.8&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
Failed to to call /store/repositories/4bb530c7 - Unknown error, see supervisor
Failed to to call /store/addons/4bb530c7_hass-addon-sunsynk-multi - Addon 4bb530c7_hass-addon-sunsynk-multi with version latest does not exist in the store
Failed to to call /store/addons/4bb530c7_hass-addon-sunsynk - Addon 4bb530c7_hass-addon-sunsynk with version latest does not exist in the store
Failed to to call /addons/4bb530c7_hass-addon-sunsynk/install - The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt~=1.5.0 pyyaml~=5.4.1 sunsynk[pymodbus,umodbus]==0.2.5' returned a non-zero code: 1

**Expected behavior**
Installation of the new version.

### Your environment

- Home Assistant version: 2023.08.1
- Addon:
  - Name: sunsynk-multi
  - Version: v0.5.8
- Inverter:
  - Make: Sunsynk
  - Model: 8kw
  - Firmware: \_\_
- Adaptor details:
  - RS485/RS232/solarman
  - RS485/RS232 brand

### You configuration

```yaml
No configuration yet.  Initial configuration lost due to uninstall
paste config in the block surrounded by 3x backticks

```

### Logs

```logs
logs already provided in problem description
paste logs in 3x backticks...

```

chickenbeef, 2023-09-04:
Seems the build is failing?

https://github.com/kellerza/sunsynk/actions/runs/6077074296/job/16486144744

kellerza, 2023-09-05:
> Edit I see the changelog is `Test ghrc.io builds` so probably a known issue 😅

Indeed a known issue. The aim is to build the addon on github so that you only need to pull an updated container instead of building them on your own. 

New installs will be an issue, but if you had 0.5.7 running it should still be ok?

w1tw0lf, 2023-09-05:
Same issue here and on 2023.08.27-0.5.7.

kellerza, 2023-09-05:
If you reload it should be possible to install a new version. No more local building @N1c084 

Almost there. Will release another one tonight!

w1tw0lf, 2023-09-05:
All good, did restart of home assistant and updated without issues.

chickenbeef, 2023-09-05:
> Indeed a known issue. The aim is to build the addon on github so that you only need to pull an updated container instead of building them on your own.
>
> New installs will be an issue, but if you had [...]

0.5.7 works totally fine. I saw the notification to update to 0.5.8 in Home Assistant so I actioned it and then saw it was failing to pull the manifest of the new image.

If 0.5.8 only has build changes, I'll skip it for now :)

fbasson, 2023-09-05:
> If you reload it should be possible to install a new version. No more local building @N1c084
> 
> Almost there. Will release another one tonight!

Will attempt install again tonight.  Thanks for the update.

N1c084, 2023-09-05:
Hi
I try this without success , recover from older full backup not working too.
So i do a new installation of  all HA on new hardware and now the 0.5.7
installation is ok , I do not try 0.5.8 for now.
Thanks

cc120689, 2023-09-05:
Same error, I can finally update it correctly.
The new plugin is working and delivering data but on the cover of the plugin it appears as stopped

cc120689, 2023-09-05:
now yes!  restarting the entire machine.  thank you!

kellerza, 2023-09-11:
Since the new build mechanism has been working for a couple of days now I'll close this. Please re-open if you see an issue with the new build mechanism.

closed 2023-09-11 by kellerza