kellerza/sunsynk #180: Cant start the Add-on on my Deye 12kW 3phases

lordvalium, 2023-09-06:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

My issue is that after a proper configuration my Addon stops (HALT).

**Expected behavior**
Should provide information via Entities

### Your environment

- Home Assistant version: 2023.8.4
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter:
  - Make: Deye
  - Model: 12kW 3phases EU 230V
  - Firmware: \_\_
- Adaptor details:
  - RS485 to USB 
  
  (set and found via Debug)

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "12345678"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: ""
SENSOR_DEFINITIONS: three-phase
SENSORS:
  - energy_management
  - power_flow_card
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Deye
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt-user
MQTT_PASSWORD: xyz
DEBUG: 1
DEBUG_DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ043VSX-if00-port0

```

### Logs

```logs

2023-09-06 16:28:48,022 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Grid Import (reason: Unit)
2023-09-06 16:28:48,022 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Load Energy (reason: Unit)
2023-09-06 16:28:48,023 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day PV Energy (reason: Unit)
[... 74 lines cut ...]
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/sunsynk/src/sunsynk/pysunsynk.py", line 83, in read_holding_registers
    res = await self.client.read_holding_registers(  # type:ignore
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 201, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
[16:28:51] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

```

kellerza, 2023-09-06:
Both debugs clearly show a timeout. 

```
 pymodbus.logging send: 0x1 0x3 0x0 0x3 0x0 0x5 0x75 0xc9
```

I can fix the retry mechanism in the first log, but it won't solve your issue. Have you looked at the fault-finding page on the docs?

kellerza, 2023-09-07:
The traceback in your first log should be solved now. But you will still have to look at the fault finding guide to get this working

Niruservices, 2023-09-26:
Morning - I need some help please. I have an  issue with opening the connection to the solarman dongle;  "pysolarmanv5.pysolarmanv5.NoSocketAvailableError: Cannot open connection to 192.168.1.186" The connect call fails and I read that it could be something to do with the encryption on the dongle - I am trying via a RPi. If I use a the RS485 - which port to use (CAN is occupied with BMS)? Also, where can I get the applicable cable?

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting

On MQTT:
023-09-26 07:48:10: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:49:10: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:49:10: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:49:52: New connection from 172.30.32.2:48572 on port 1883.
2023-09-26 07:49:52: Client <unknown> closed its connection.
2023-09-26 07:50:10: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:50:10: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:51:10: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:51:10: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:51:52: New connection from 172.30.32.2:53956 on port 1883.
2023-09-26 07:51:52: Client <unknown> closed its connection.
2023-09-26 07:52:10: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:52:10: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:53:11: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:53:11: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:53:52: New connection from 172.30.32.2:57820 on port 1883.
2023-09-26 07:53:52: Client <unknown> closed its connection.
2023-09-26 07:54:11: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:54:11: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:55:11: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:55:11: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:55:52: New connection from 172.30.32.2:37094 on port 1883.
2023-09-26 07:55:52: Client <unknown> closed its connection.
2023-09-26 07:56:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:56:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:57:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:57:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:57:52: New connection from 172.30.32.2:51896 on port 1883.
2023-09-26 07:57:52: Client <unknown> closed its connection.
2023-09-26 07:58:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:58:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:59:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:59:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:59:52: New connection from 172.30.32.2:45136 on port 1883.
2023-09-26 07:59:52: Client <unknown> closed its connection.
2023-09-26 08:00:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:00:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:01:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:01:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:01:52: New connection from 172.30.32.2:60454 on port 1883.
2023-09-26 08:01:52: Client <unknown> closed its connection.
2023-09-26 08:02:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:02:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:03:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:03:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:03:52: New connection from 172.30.32.2:55940 on port 1883.
2023-09-26 08:03:52: Client <unknown> closed its connection.
2023-09-26 08:04:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:04:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:05:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:05:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:05:52: New connection from 172.30.32.2:44346 on port 1883.
2023-09-26 08:05:52: Client <unknown> closed its connection.
2023-09-26 08:06:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:06:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:07:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:07:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:07:52: New connection from 172.30.32.2:57240 on port 1883.
2023-09-26 08:07:52: Client <unknown> closed its connection.
2023-09-26 08:08:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:08:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
[... 90 lines cut ...]
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 78, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 73, in main
    loop.run_until_complete(main_loop(loop))
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 38, in main_loop
    await ist.connect()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 113, in connect
    await self.inv.connect()
[... cut ...]
pysolarmanv5.pysolarmanv5.NoSocketAvailableError: Cannot open connection to 192.168.1.186
[20:21:36] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Niruservices, 2023-09-26:
Sorry, some copy error.... MQTT in the wrong place;

Niruservices, 2023-09-26:
Add-on (multi) Deye 12Kw 3 Phase
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
20:21:31 INFO    Loading configuration: /data/options.json
20:21:31 INFO    sunsynk library version: 0.6.4
20:21:31 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  20  |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
20:21:31 INFO    Using three phase sensor definitions.
20:21:31 ERROR   Unknown sensor specified: aux_power
20:21:31 ERROR   Unknown sensor specified: essential_power
20:21:31 ERROR   Unknown sensor specified: grid_connected
20:21:31 ERROR   Unknown sensor specified: inverter_current
20:21:31 ERROR   Unknown sensor specified: load_frequency
20:21:31 ERROR   Unknown sensor specified: non_essential_power
20:21:31 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
20:21:31 INFO    Connecting to tcp://192.168.1.186:8899
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 68, in connect
    self.reader, self.writer = await asyncio.open_connection(
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1085, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1069, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 634, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 674, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.1.186', 8899)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 78, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 73, in main
    loop.run_until_complete(main_loop(loop))
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 38, in main_loop
    await ist.connect()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 113, in connect
    await self.inv.connect()
  File "/usr/local/lib/python3.11/site-packages/sunsynk/solarmansunsynk.py", line 51, in connect
    await self.client.connect()
  File "/usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 73, in connect
    raise NoSocketAvailableError(f"Cannot open connection to {self.address}")
pysolarmanv5.pysolarmanv5.NoSocketAvailableError: Cannot open connection to 192.168.1.186
[20:21:36] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Niruservices, 2023-09-26:
MQTT:

2023-09-26 07:52:10: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:53:11: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:53:11: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:53:52: New connection from 172.30.32.2:57820 on port 1883.
2023-09-26 07:53:52: Client <unknown> closed its connection.
2023-09-26 07:54:11: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:54:11: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:55:11: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:55:11: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:55:52: New connection from 172.30.32.2:37094 on port 1883.
2023-09-26 07:55:52: Client <unknown> closed its connection.
2023-09-26 07:56:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:56:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:57:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:57:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:57:52: New connection from 172.30.32.2:51896 on port 1883.
2023-09-26 07:57:52: Client <unknown> closed its connection.
2023-09-26 07:58:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:58:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:59:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:59:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 07:59:52: New connection from 172.30.32.2:45136 on port 1883.
2023-09-26 07:59:52: Client <unknown> closed its connection.
2023-09-26 08:00:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:00:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:01:12: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:01:12: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:01:52: New connection from 172.30.32.2:60454 on port 1883.
2023-09-26 08:01:52: Client <unknown> closed its connection.
2023-09-26 08:02:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:02:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:03:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:03:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:03:52: New connection from 172.30.32.2:55940 on port 1883.
2023-09-26 08:03:52: Client <unknown> closed its connection.
2023-09-26 08:04:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:04:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:05:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:05:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:05:52: New connection from 172.30.32.2:44346 on port 1883.
2023-09-26 08:05:52: Client <unknown> closed its connection.
2023-09-26 08:06:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:06:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:07:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:07:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:07:52: New connection from 172.30.32.2:57240 on port 1883.
2023-09-26 08:07:52: Client <unknown> closed its connection.
2023-09-26 08:08:13: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:08:13: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:08:28: Saving in-memory database to /data//mosquitto.db.
2023-09-26 08:09:14: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:09:14: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:09:52: New connection from 172.30.32.2:40638 on port 1883.
2023-09-26 08:09:52: Client <unknown> closed its connection.
2023-09-26 08:10:14: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:10:14: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:11:14: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:11:14: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:11:52: New connection from 172.30.32.2:60108 on port 1883.
2023-09-26 08:11:52: Client <unknown> closed its connection.
2023-09-26 08:12:14: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:12:14: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:13:15: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:13:15: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:13:53: New connection from 172.30.32.2:51858 on port 1883.
2023-09-26 08:13:53: Client <unknown> closed its connection.
2023-09-26 08:14:15: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:14:15: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
[... 7 lines cut ...]
2023-09-26 08:17:15: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:17:53: New connection from 172.30.32.2:57754 on port 1883.
2023-09-26 08:17:53: Client <unknown> closed its connection.
2023-09-26 08:18:15: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:18:15: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:19:16: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:19:16: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:19:53: New connection from 172.30.32.2:60002 on port 1883.
2023-09-26 08:19:53: Client <unknown> closed its connection.
2023-09-26 08:20:16: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:20:16: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:21:16: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:21:16: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:21:53: New connection from 172.30.32.2:40760 on port 1883.
2023-09-26 08:21:53: Client <unknown> closed its connection.
[... cut ...]
2023-09-26 08:23:16: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:23:16: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:23:53: New connection from 172.30.32.2:45478 on port 1883.
2023-09-26 08:23:53: Client <unknown> closed its connection.
2023-09-26 08:24:17: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:24:17: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:25:17: Received PINGREQ from 0Loy1diDJ1jYSdF5mD9cav
2023-09-26 08:25:17: Sending PINGRESP to 0Loy1diDJ1jYSdF5mD9cav

kellerza, 2023-09-26:
@Niruservices - are you sure you have the corretc IP for your dongle? 

`OSError: [Errno 113] Connect call failed ('192.168.1.186', 8899)`

Niruservices, 2023-09-26:
Evening,

 

It is really odd – as it shows the correct MAC & shows connection speeds on the router – yet – not able to ping it either.

 

 

 

BR;

Nico

Niruservices, 2023-09-27:
Morning,

 

BTW, if I use RS485 – which bus to select py or U?

 

BR;

Nico

kellerza, 2023-09-28:
You can use either, whichever is the most reliable for you. 

I have found that the recent pymodbus is fairly reliable for direct serial.

umodbus has always worked well (via some gateway like mbusd, but not direct serial)

kellerza, 2023-09-28:
@lordvalium did you manage to get it running?

lordvalium, 2023-10-20:
no unfortunately not @kellerza 
Have bought that addon https://www.amazon.de/dp/B08DXRKFZR but its not recognized in Homeassistant

closed 2025-03-18 by kellerza