kellerza/sunsynk #181: Max Power output Sunsynk 5kw

e1ace, 2023-09-06:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Prog 1 through 6 is limited to between 0 and 5000w.

**Expected behavior**
The data sheet for the 5kw Sunsynk says a max of 5500w output. If I adjust the settings on the inverter itself I can set between 0 and 5500w, but on the Add-On it is limited.

### Your environment

- Home Assistant version: 2023.8.x
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-09-04
- Inverter:
  - Make: Sunsynk
  - Model: 5kW
  - Firmware: \_\_
- Adaptor details:
  - RS485

kellerza, 2023-09-06:
What do you see in Settings->Devices?
[image]

The max is read from the inverter (`RATED_POWER`). The only way to force it would be a custom sensor - see the docs.

You need to redefine these sensors in `mysensors.py`, set `RATED_POWER` to 5500 

```python
    NumberRWSensor(256, "Prog1 power", WATT, max=5500),
    NumberRWSensor(257, "Prog2 power", WATT, max=RATED_POWER),
    NumberRWSensor(258, "Prog3 power", WATT, max=RATED_POWER),
    NumberRWSensor(259, "Prog4 power", WATT, max=RATED_POWER),
    NumberRWSensor(260, "Prog5 power", WATT, max=RATED_POWER),
    NumberRWSensor(261, "Prog6 power", WATT, max=RATED_POWER),
```

e1ace, 2023-09-09:
[image]

Thanks. The redefined sensors works.

closed 2023-09-09 by e1ace