kellerza/sunsynk #182: The Grid Connected sensor unknown state after HA restart

ech0-py, 2023-09-10:
### You Inverter
Deye 8kw 1phase (SG01 generation)

### Addon 
Sunsynk/Deye Inverter Add-on (multi)
Current version: 0.5.9

### Describe the issue/bug and what you expect
After one of the recent updates (not sure which one) the Grid Connected sensor goes to the "unknown" state after HA restart, while mqtt-explorer shows that sensor is present and availability topic is present.
Reloading of MQTT integration do nothing, restarting of sunsynk addon helps to resolve the issue.

Not sure where I can see what's going wrong

### Logs (if applicable)

```logs

07:54:39 INFO    Schedules:
+----------------+-----+------+--------+-----------+----------+------------+
|      Key       | src | Read | Report | Change by | Change % | Change any |
+----------------+-----+------+--------+-----------+----------+------------+
|   date_time    |     |  60  |   60   |           |          |    True    |
|       rw       |  *  |  15  |  300   |           |          |    True    |
|       w        |  *  |  5   |   5    |     80    |          |            |
|      kwh       |     | 300  |  300   |           |          |            |
|    any_unit    |     |  15  |  300   |           |          |            |
|    no_unit     |     |  15  |  300   |           |          |    True    |
| grid_connected |  +  |  5   |   5    |           |          |    True    |
|       °c       |  +  |  5   |   10   |           |          |    True    |
+----------------+-----+------+--------+-----------+----------+------------+

```
```
homeassistant/binary_sensor/2301132069/grid_connected

{"unique_id": "2301132069_grid_connected", "device": {"identifiers": ["2301132069"], "manufacturer": "Sunsynk", "model": "8kW Inverter (**2069)", "name": "DY"}, "state_topic": "SUNSYNK/status/2301132069/grid_connected", "name": "Grid Connected", "availability": [{"topic": "SUNSYNK/status/2301132069/availability"}], "object_id": "dy_grid_connected"}
```
[image]

[image]

kellerza, 2023-09-11:
My guess would be this has to do with the MQTT retain Retain flag

What this means is that your MQTT explorer was open when the message was received on MQTT, but it's not kept on the server. If you restart MQTT explorer you will see the same result as you see with restarting HASS. The value will not exist (Unknown)

The value needs to be reported again. In theory `no_unit` should be reported every 5 min...  not sure why the HASS state mentions 6 min unknown, since there should have been a 5min in that 6min (they will not be synced)

ech0-py, 2023-09-11:
But I have configured schedule to report every 5sec (full addon config I've added to the Addon configuration... expand section, that not obvious I guess), thus I though everything should be ok, but it's not

```
SCHEDULES:
  - KEY: grid_connected
    READ_EVERY: 5
    REPORT_EVERY: 5
    CHANGE_ANY: true
```
I made another longer test to show that even mosquitto restart doesn't help and entity was in unknown state for more than 20 minutes long
[image]

kellerza, 2023-09-12:
Mosquitto restart won't help. Probably only the addon restart

Can you try setting `change_any` to false? or removing it from the schedule?

ech0-py, 2023-09-12:
Still the same in both cases:
 - change_any: false (or remove the grid_connected from the schedule)
 - restart HA 
 - sensor becomes unknown
 - wait for 5+ minutes until every other sensor get value
 - sensor is still unknown  

Sure, only the addon restart helps

N1c084, 2023-09-12:
Same for me.
But just change the grid state in real from on to off and the state of
entity appear clearly now.

kellerza, 2023-09-12:
I've added a possible fix in the edge branch, only unit tested for now

ech0-py, 2023-09-12:
> But just change the grid state in real from on to off and the state of entity appear clearly now.

I can confirm. If in unknown state I turn off the inverter and then turn on it the entity becomes changes the state accordingly

kellerza, 2023-09-12:
For some reason the `BinarySensor` required an actual change (by any means) to work. This sensors is the only Binary Sensor today.

We have so much load-shedding in South Africa I probably would have never picked up this bug 🤣

ech0-py, 2023-09-12:
I confirm that on the edge everything now works as expected
From my perspective, what's changed:
 - SUNSYNK/status/<serial>/grid_connected now contains ON/OFF instead of 1/0
 - if I change the sensor's state manually it returned back to the actual state
[image]

I appreciate your efforts, sir, god bless you <3

N1c084, 2023-09-12:
🤣in France very rare grid outage .
But I force every day my inverter to  go off -on grid depending of battery
soc, so I see the unknow to ok value when the state change appear. Binary
start at first change

ech0-py, 2023-09-12:
> But I force every day my inverter to go off -on grid depending of battery soc

Sorry I know this is off topic, but can you describe how you achieve that? Is that manual way (turn off the breaker by hands), automatic (smart zigbee/wifi relay/breaker) or mb even some kind of software (writing to the modbus registry, for example).
I'm trying found ways to achieve low energy consumption from the grid/battery during the night

N1c084, 2023-09-12:
Just change minimum frequency to a value outage your grid
For example 50Hz grid
> min frequency set to 52Hz and then inverter go alone to off grid by
opening the internal grid input relay

closed 2023-09-12 by kellerza