kellerza/sunsynk #186: Addon writes incorrect values to the modbus registers

ech0-py, 2023-09-14:
### Issue related to

Sunsynk  Home Assistant Add-On

### Describe the issue/bug
Description moved under the logs section

### Your environment
- Home Assistant version: 2023.9.1
- Sunsynk-multi addon: 0.6.4
- Modbus addon: 2021.10.1 
- Inverter: Deye 8k
- Adaptor: RS485 waveshare

### You configuration
```python
from sunsynk.rwsensors import NumberRWSensor

SENSORS = SensorDefinitions()
SENSORS += NumberRWSensor(290, "Grid frequency min", "Hz", 0.01)
```

```yaml
DRIVER: umodbus
INVERTERS:
  - SERIAL_NR: "<hidden>"
    HA_PREFIX: DY
    MODBUS_ID: 1
    PORT: tcp://homeassistant.local:502
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - grid_frequency_min
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: rw
    READ_EVERY: 30
    REPORT_EVERY: 30
    CHANGE_ANY: true
    CHANGE_BY: 0
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: box
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: <hidden>
MQTT_PASSWORD: <hidden>
DEBUG: 5
DEBUG_DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ0259H2-if00-port0
```

### Logs

```logs
Sunsynk-multi addon:

2023-09-14 16:13:33,378 INFO    ha_addon_sunsynk_multi.a_sensor Queue update grid_frequency_min=48
2023-09-14 16:13:34,154 INFO    sunsynk.sunsynk Writing sensor grid_frequency_min=48 Registers:{290=0x0064} 
2023-09-14 16:13:34,375 DEBUG   sunsynk.sunsynk Time taken to fetch 1 registers starting at 290 : 0.10s
2023-09-14 16:13:34,375 DEBUG   sunsynk.sunsynk Request registers: [290] glen=1. Response [100] len=1. regs={290: 100}
2023-09-14 16:13:34,375 DEBUG   sunsynk.sensors grid_frequency_min=1Hz (100,)
2023-09-14 16:13:34,375 DEBUG   sunsynk.state register {3: 12851, 4: 12337, 5: 12595, 6: 12848, 7: 13881, 16: 14464, 17: 1, 290: 100} = 1 (old=1)
```
```logs
Modbus TCP to Modbus RTU Gateway Add-on:

14 Sep 2023 19:13:34 conn[172.30.33.1]: request: [01][10][01][22][00][01][02][00][64]
```

### Describe the issue/bug
When I'm trying to edit value in the number box the addon "reads" that value normally, but "sends" something different.  
As shown in logs above I type the new value as "48" and hit enter, but addon sent "100" instead of it (last bytes [00][64] 0x0064 == 100)

**Expected behavior**
Request to Modbus Gateway should look like 
[01][10][01][22][00][01][02]**[12][c0]** (where 0x12c0 is 4800)

ech0-py, 2023-09-14:
I confirm that the Modbus add-on is working properly

ech0-py, 2023-09-14:
Mb I'm wrong, but it seems there is the root cause:
https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/rwsensors.py#L74
```python
        fval = float(value)  # 48.0
        minv = resolve_num(resolve, self.min, 0)  # 0
        maxv = resolve_num(resolve, self.max, 100)  # 100
        val = int(max(minv, min(maxv, fval / abs(self.factor))))
        # max(0, (min(100, 4800.0)) == 100
```

kellerza, 2023-09-15:
Thanks @ech0-py - I think you got it, in the commit I moved the factor to apply to min&max as well.

It should deploy on the edge branch until a new version is released.

ech0-py, 2023-09-15:
Сonfirm, the value writing of number-rw's looks great on the edge.
```logs
07:03:21 INFO    Queue update grid_frequency_min=48
07:03:22 INFO    Writing sensor grid_frequency_min=48 Registers:{290=0x12c0} 
07:03:31 INFO    Queue update battery_float_voltage=53.3
07:03:32 INFO    Writing sensor battery_float_voltage=53.3 Registers:{203=0x14d2} 
```
@kellerza thanks again <3