kellerza/sunsynk #188: config for multiple inverters

richjgould, 2023-09-20:
Just had a Sunsynk 8.8kw inverter installed in parallel to my existing 3.6kw. Add-on was working fine with just the 3.6kw.

Can I clarify if this config is correct:

```
- SERIAL_NR: "2205xxxxxx"
  HA_PREFIX: 3K6
  MODBUS_ID: 1
  PORT: tcp://homeassistant.local:502
  
- SERIAL_NR: "2304xxxxxx"
  HA_PREFIX: 8K8
  MODBUS_ID: 2
  PORT: tcp://homeassistant.local:502
```
I have set the inverters as master/slave on the devices and the Modbus ports to match. Cat5 cable between the parallel sockets of the inverters.

Thanks
Richard

richjgould, 2023-09-26:
Is there anywhere else I could be signposted for further advice or examples of this running on the multi-inverter add-on?

I now have a RS485 adaptor per inverter.

Is the config below correct when dealing with multiple RS485 adaptors in once instance of the add-on?

`
- SERIAL_NR: "220528xxxx"
  HA_PREFIX: 3k6
  MODBUS_ID: 1
  PORT: tcp://homeassistant.local:502
  DEVICE: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5185022356-if00

- SERIAL_NR: "230405xxxx"
  HA_PREFIX: 8k8
  MODBUS_ID: 2
  PORT: tcp://homeassistant.local:502
  DEVICE: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00
`

Thanks
Richard

closed 2023-09-26 by richjgould

richjgould, 2023-09-26:
Is there anywhere else I could be signposted for further advice or examples of this running on the multi-inverter add-on?

I now have an RS485 adaptor per inverter.

Is the config below correct when dealing with multiple RS485 adaptors in once instance of the add-on?

```
- SERIAL_NR: "2205xxxx"
  HA_PREFIX: 3k6
  MODBUS_ID: 1
  PORT: tcp://homeassistant.local:502
  DEVICE: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5185022356-if00

- SERIAL_NR: "2304xxxx"
  HA_PREFIX: 8k8
  MODBUS_ID: 2
  PORT: tcp://homeassistant.local:502
  DEVICE: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00
```

Thanks
Richard

kellerza, 2023-09-26:
```yaml
DRIVER: pymodbus
INVERTERS:
- SERIAL_NR: "2205xxxx"
  HA_PREFIX: 3k6
  MODBUS_ID: 1
  PORT: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5185022356-if00
- SERIAL_NR: "2304xxxx"
  HA_PREFIX: 8k8
  MODBUS_ID: 2
  PORT: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00
```

kellerza, 2023-09-26:
Do you still have mbusd running on one of those ports?

richjgould, 2023-09-26:
Thanks Kellerza.

Stopped the mbusd add-on and now the first inverter reads with the above config but the second one still times out.

```
023-09-26 18:25:58,742 INFO    ha_addon_sunsynk_multi.a_inverter Connecting to /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00
2023-09-26 18:25:58,742 INFO    sunsynk.pysunsynk PyModbus 3.4.1 Serial: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00
2023-09-26 18:25:58,743 DEBUG   pymodbus.logging Connecting to /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00.
2023-09-26 18:25:58,743 DEBUG   pymodbus.logging Connecting comm
2023-09-26 18:25:58,748 DEBUG   pymodbus.logging Connected to comm
2023-09-26 18:25:58,749 DEBUG   pymodbus.logging callback_connected called
2023-09-26 18:25:58,750 INFO    ha_addon_sunsynk_multi.a_inverter Reading startup sensors Prog6 Time, Prog2 Time, Serial, Battery Low Capacity, Prog5 Time, Prog1 Time, Prog3 Time, Prog4 Time, Rated power
2023-09-26 18:25:58,759 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0x3 0x0 0x5 0x75 0xfa
2023-09-26 18:25:58,760 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:01,775 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout 
2023-09-26 18:26:01,890 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0x3 0x0 0x5 0x75 0xfa
2023-09-26 18:26:01,891 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:04,906 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout 
2023-09-26 18:26:05,023 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0x3 0x0 0x5 0x75 0xfa
2023-09-26 18:26:05,026 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:08,037 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout 
2023-09-26 18:26:08,140 WARNING ha_addon_sunsynk_multi.a_inverter Retrying individual sensors: ['prog6_time', 'prog2_time', 'serial', 'battery_low_capacity', 'prog5_time', 'prog1_time', 'prog3_time', 'prog4_time', 'rated_power']
2023-09-26 18:26:08,174 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xff 0x0 0x1 0xb4 0x9
2023-09-26 18:26:08,176 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:11,188 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Prog6 Time
2023-09-26 18:26:11,325 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xfb 0x0 0x1 0xf5 0xc8
2023-09-26 18:26:11,326 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:14,340 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Prog2 Time
2023-09-26 18:26:14,482 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0x3 0x0 0x5 0x75 0xfa
2023-09-26 18:26:14,483 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:17,493 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Serial
2023-09-26 18:26:17,629 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xdb 0x0 0x1 0xf4 0x2
2023-09-26 18:26:17,631 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:20,650 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Battery Low Capacity
2023-09-26 18:26:20,787 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xfe 0x0 0x1 0xe5 0xc9
2023-09-26 18:26:20,789 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:23,801 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Prog5 Time
2023-09-26 18:26:23,938 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xfa 0x0 0x1 0xa4 0x8
2023-09-26 18:26:23,939 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:26,954 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Prog1 Time
2023-09-26 18:26:27,094 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xfc 0x0 0x1 0x44 0x9
2023-09-26 18:26:27,096 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:30,108 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Prog3 Time
2023-09-26 18:26:30,246 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0xfd 0x0 0x1 0x15 0xc9
2023-09-26 18:26:30,247 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:33,254 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Prog4 Time
2023-09-26 18:26:33,392 DEBUG   pymodbus.logging send: 0x2 0x3 0x0 0x10 0x0 0x2 0xc5 0xfd
2023-09-26 18:26:33,394 DEBUG   pymodbus.logging Adding transaction 2
2023-09-26 18:26:36,408 ERROR   ha_addon_sunsynk_multi.a_inverter Timeout Rated power
2023-09-26 18:26:36,512 CRITICAL ha_addon_sunsynk_multi.a_inverter Could not read sensors: ['Prog6 Time', 'Prog2 Time', 'Serial', 'Battery Low Capacity', 'Prog5 Time', 'Prog1 Time', 'Prog3 Time', 'Prog4 Time', 'Rated power']
2023-09-26 18:26:36,513 INFO    ha_addon_sunsynk_multi.a_inverter ############################################################
2023-09-26 18:26:36,513 INFO    ha_addon_sunsynk_multi.a_inverter No response on the Modbus interface /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00, see https://kellerza.github.io/sunsynk/guide/fault-finding
2023-09-26 18:26:36,514 INFO    ha_addon_sunsynk_multi.a_inverter ############################################################
2023-09-26 18:26:36,514 CRITICAL __main__ This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
```

Unfortunately this then causes the add-on to restart which I suppose would not necessarily be a required action in a multi-inverter set-up?

I need to change some of the config settings to only pull certain registers from the master inverter I think.

BW
Richard

kellerza, 2023-09-26:
Try `SENSORS_FIRST_INVERTER`

richjgould, 2023-09-26:
Thanks.

Yep my config now looks like this:

```
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2205******"
    HA_PREFIX: ss3k6
    MODBUS_ID: 1
    PORT: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5185022356-if00
  - SERIAL_NR: "2304*******"
    HA_PREFIX: ss8k8
    MODBUS_ID: 2
    PORT: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5586009732-if00
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - total_pv_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - battery_soc
  - battery_power
  - battery_current
  - battery_voltage
  - inverter_power
  - grid_power
  - load_power
  - PV1_power
  - PV2_power
  - day_active_energy
  - day_battery_charge
  - day_battery_discharge
  - day_grid_import
  - day_grid_export
  - day_load_energy
  - day_PV_energy
  - inverter_voltage
  - inverter_current
  - inverter_frequency
  - load_frequency
  - PV1_voltage
  - PV1_current
  - PV2_voltage
  - PV2_current
SENSORS_FIRST_INVERTER:
  - load_limit
  - grid_connected_status
  - priority_load
  - use_timer
  - grid_charge_enabled
  - control_mode
  - grid_frequency
  - grid_connected
  - battery_charging_voltage
  - allow_export
  - prog1_time
  - prog2_time
  - prog3_time
  - prog4_time
  - prog5_time
  - prog6_time
  - prog1_power
  - prog2_power
  - prog3_power
  - prog4_power
  - prog5_power
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
  - solar_export
  - overall_state
  - battery_temperature
  - dc_transformer_temperature
  - radiator_temperature
  - aux_power
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 60
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt
MQTT_PASSWORD: ********
DEBUG: 1
DEBUG_DEVICE: /dev/serial/by-id/usb-1a86_USB_Single_Serial_5185022356-if00
```

However still getting the same errors as above. The add-on is trying to read prog1_time from the second inverter even though that register is only in the first inverter section? This may be entirely unrelated to why I can't get this working but thought I'd share.

BW
Richard

richjgould, 2023-09-26:
I suspect I may be dealing with a hardware/wiring issue and will look again at how I've wired up to the inverter.

However any glaring problems with the above config would be very welcomed.

Best Wishes
Richard