kellerza/sunsynk #192: Data reporting randomly pause for a couple of minutes

adnovea, 2023-09-30:
### Issue related to

Sunsynk / pymodbus Home Assistant Add-On

### Describe the issue/bug

Data reporting pause randomly for about a quarter of an hour. I notice this issue through the inverter Date Time update.
I left the date_time entity as defined by default and copied all the sensors needed into the /share/hass-addon-sunsynk/mysensors.py file for easy debugging.

**Expected behavior**
I have setup the SCHEDULE parameters for a reporting twice a minute.
Everything works for a undetermined period of time (_half an hour to a couple of hours_) then the entities update pause for about a quarter of an hour.

### Your environment

- Home Assistant version: 2023.9.3
- Addon:
  - Name: Sunsynk/Deye Inverter Add-on (multi)
  - Version: 0.6.4
- Inverter:
  - Make: Deye
  - Model: Sun 5K SG03LP1-EU
  - Firmware: 0000-C366 / 0000-3384
- Adaptor details:
  - solarman LSE-3 (Ethernet version)

### You configuration

```
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "XXXXXXXX"
    HA_PREFIX: SunSynk
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "21XXXXXXXXX"
    PORT: tcp://192.168.XXX.XXX:8899
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - mysensors
  - date_time
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Deye
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 24
SCHEDULES:
  - KEY: date_time
    READ_EVERY: 15
    REPORT_EVERY: 30
    CHANGE_ANY: true
  - KEY: RW
    READ_EVERY: 15
    REPORT_EVERY: 30
    CHANGE_ANY: true
  - KEY: W
    READ_EVERY: 15
    REPORT_EVERY: 30
    CHANGE_BY: 80
  - KEY: KWH
    READ_EVERY: 30
    REPORT_EVERY: 30
  - KEY: any_unit
    READ_EVERY: 30
    REPORT_EVERY: 30
  - KEY: no_unit
    READ_EVERY: 15
    REPORT_EVERY: 30
    CHANGE_ANY: true
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
[... 88 lines cut ...]
07:31:24 ERROR   Fatal error: protocol.data_received() call failed.
protocol: <pymodbus.client.tcp.AsyncModbusTcpClient object at 0x7f89622890>
transport: <_SelectorSocketTransport fd=10 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1009, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transport/transport.py", line 289, in data_received
    self.datagram_received(data, None)
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transport/transport.py", line 303, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 219, in callback_data
    self.framer.processIncomingPacket(data, self._handle_response, slave=0)
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/base.py", line 137, in processIncomingPacket
    self.frameProcessIncomingPacket(single, callback, slave, **kwargs)
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 147, in frameProcessIncomingPacket
    self._process(callback, tid)
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 154, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
07:31:24 ERROR   ConnectionException in read_sunsynk: Modbus Error: [Connection] Connection lost during request

```

Thanks for this very complete add-on and for your help to tackle this issue.

kellerza, 2023-10-06:
Could probably try to suppress that error, but its likely the dongle disconnected for some reason

Today the retry mechanism should be a restart of the app with the addon’s watchdog

adnovea, 2023-10-06:
Dear Johann,
Thanks for your quick reply and for the excellent work you achieved 🥇 
I know how much effort it takes to maintain open-source devs!

Unfortunately, I cannot consider the dongle disconnection as the root cause of my problem because I ran the [Stephan Joubert](https://github.com/StephanJoubert/home_assistant_solarman)'s Solarman Integration for a couple of months without issues using my LSE-3 dongle (_but Stephan's integration has no inverter control possibilities_).

I tried to tackle the issue but I miss some pieces of information such as :
- Where could the full log file be found (_only tail is displayed on the log page_)
- Where is the Add-ons installed (_could I access the files using HA's Terminal_ ?)

I also suspected the Mosquitto broker to be responsible, but using MQTT Explorer I have abandoned this possibility for the time being. Moreover, I have an other add-on using MQTT without issues.

The snapshots below shows random long pause (15 mins) over the data reading.
It very clear on the 2nd picture that shows the date_time updates. The 4 pauses seems to be somewhat periodic !

I have removed my "custom sensors" and changed my Schedule  to :
```
SCHEDULES:
  - KEY: date_time
    READ_EVERY: 30
    REPORT_EVERY: 30
    CHANGE_ANY: true
  - KEY: RW
    READ_EVERY: 30
    REPORT_EVERY: 30
    CHANGE_ANY: true
  - KEY: W
    READ_EVERY: 30
    REPORT_EVERY: 30
    CHANGE_BY: 80
  - KEY: KWH
    READ_EVERY: 30
    REPORT_EVERY: 30
  - KEY: any_unit
    READ_EVERY: 30
    REPORT_EVERY: 30
  - KEY: no_unit
    READ_EVERY: 30
    REPORT_EVERY: 30
    CHANGE_ANY: true
```
Thanks for your help

[image]

[image]

kellerza, 2023-10-06:
At some point I added some improvements to the dev version of the addon & you could try using that for a while?

adnovea, 2023-10-09:
Thanks Johann for your reply.
I tried the beta version with no success. I cannot understand why the add-on pauses so long making the data unusable.
I had to fallback to Stephen's add-on to get relevant data records.
I hope this problem will be solved. However,I have no idea for the time being.

purcell-lab, 2024-03-09:
Similar issue here:

[image]

I have an hourly automation to restart, but would like to use a better trigger.
[image]

kellerza, 2024-03-21:
Only noticed now you are using the pymodbus driver, while trying to connect to the solarman dongle.
Please use the solarman driver

```yaml
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "XXXXXXXX"
    HA_PREFIX: SunSynk
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "21XXXXXXXXX"
    PORT: tcp://192.168.XXX.XXX:8899
```

closed 2024-03-21 by kellerza