kellerza/sunsynk #195: Missing Sunsynk Inverter and Battery Temperature

ai001, 2023-10-11:
### Issue related to the Python sunsynk library

### Describe the issue/bug
Not really a bug but a request to make available temperature values of various component from the Sunsynk inverters where it can be used to trigger additional cooling fan on and off.
### Expected behavior

A request to make available temperature values of various component from the Sunsynk inverters where it can be used to trigger additional cooling fan on and off.

### Logs (if applicable)

```txt
paste logs in 3x backticks...
```

TomBeech85, 2023-10-13:
I came here to raise this feature request. I want it to make sure the I can get an alert if the batteries get too cold during the winter months

iBeech, 2023-10-18:
> ### Issue related to the Python sunsynk library
> ### Describe the issue/bug
> Not really a bug but a request to make available temperature values of various component from the Sunsynk inverters where [...]

You can add two sensors to achieve this:

- dc_transformer_temperature
- radiator_temperature

kellerza, 2023-10-23:
In the past I've tried adding more temperature sensors, without luck

You can experiment with new sensors/registers using custom sensors - see https://kellerza.github.io/sunsynk/reference/mysensors

steve-prentice, 2023-11-17:
Also here to ask the same...

battery_temp_182: sensor.ss_battery_temperature

Is mentioned in the example template: https://kellerza.github.io/sunsynk/examples/templates

But doesn't seem to exist, I can't see it in 'Sunsynk/Deye Inverter Add-on (multi)' logs as a sensor that's being read?

I've read the Custom Sensors doc, but it blows my mind as a newbie, and I'd hope that it should be there without needing to add it as custom in any case?

I'd also like to get inverter_voltage_154: sensor.ss_inverter_voltage which is missing as well as far as I can see?

steve-prentice, 2023-11-18:
I've read the documentation in way more depth now than I think was needed for this... so if anyone else is wanting battery temp and not finding it easy, edit your Configuration tab in HA - I chose to Edit in YAML - and add to your Sensors area, for example I added this:

SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - battery_temperature
  - inverter_voltage
  - grid_current

closed 2024-02-28 by kellerza