kellerza/sunsynk #206: One inverter - Two Home Assistants

juanka99, 2023-11-06:
Hi,

My question is simple:

Could it possible to use an rj45 splitter to use the inverter data for two different Home Assistants at the same time?

[image]

Thanks!

juanka99, 2023-11-07:
Hi @kellerza. 
Is it possible that if I try to do this it will cause a problem in the BMS-485 connection of the inverter?
In Spain we say that experiments should be done with soda...

reedy, 2023-11-21:
> I'm not sure this will work. Electrically the RS485 can handle a "bus" - so multiple senders&receivers on a wire.

Indeed. Those splitters basically just make it so you can use 4 of the wires for ethernet on one device, and 4 for another. It doesn't split/share any of the individual wires.

Can be seen better documented on things like https://www.instructables.com/How-to-make-your-own-Ethernet-%22splitter%22/

kellerza, 2024-07-05:
See a proposal for the MQTT solution - #297

closed 2024-07-05 by kellerza