kellerza/sunsynk #208: SUN-29.9/30/35/40/50K-SG01HP3-EU

JanKraslice, 2023-11-11:
Will you add support for the HV version of sunsynk/deye SUN-29.9/30/35/40/50K-SG01HP3-EU?

kellerza, 2023-11-14:
If you have a document with the register definitions & willing to test we could add it

JanKraslice, 2023-11-14:
Hello, the error is for example in the battery voltage HV 

The battery voltage is from:

124.8~175.2 / 332.8 ~467.2 / 499.2~700V

See https://deye.com/cs/product/energy-storage-system-supplier-bos-g/

 

Furthermore, the value with PV is not displayed correctly 

 

Other communications including registers scheduling, off on sales are identical.

 

The registers are maybe the same, isn't the error just in the decimal point - miscalculation? 

 

Thank you very much

 

Barinka Jan

JanKraslice, 2023-11-14:
Hello, the error is for example in the battery voltage HV 

The battery voltage is from:

124.8~175.2 / 332.8 ~467.2 / 499.2~700V

See https://deye.com/cs/product/energy-storage-system-supplier-bos-g/

 

Furthermore, the value with PV is not displayed correctly 

 

Other communications including registers scheduling, off on sales are identical.

 

The registers are maybe the same, isn't the error just in the decimal point - miscalculation? 

 

Thank you very much

JanKraslice, 2023-11-18:
> Pokud máte dokument s definicemi registru a chcete jej otestovat, můžeme jej přidat

Hello, the registers are on your email

JanKraslice, 2023-12-14:
I didn't get a reply from you I'm sending the registries of the latest version. It is for low voltage (LV) and high voltage (HV) variant 
It is the version of the last year 23
[Uploading MODBUS RTU_3PLV_HV_Protocol_V104.pdf…]()

rixxxx, 2023-12-29:
I have a patch to add support for the HV inverters. I'll send a pull request soon.

rixxxx, 2023-12-29:
#218

kharnt0x, 2024-02-07:
@JanKraslice are you able to repost the link to that protocol doc? The link is just pointing back at this issue

kharnt0x, 2024-02-08:
PR here to add a lot more things:

https://github.com/kellerza/sunsynk/pull/233

kellerza, 2024-02-28:
Thanks @kharnt0x  & @rixxxx

closed 2024-02-28 by kellerza

JanKraslice, 2024-03-12:
Hello why even after your update the HV DEYE 30K still shows the battery voltage value for the LV ( low voltage) inverter