kellerza/sunsynk #210: Load port power

cc120689, 2023-11-23:
Hello again John, this is not a bug, it is more of a question to implement a new sensor or calculate it.  I would like to know what would be the sensor that gives the exact measurement of the power in the physical load port of the inverter, normal sensors do not change if the inverter is connected as a grid inverter or as an isolated one, however the deye cloud application identifies the data from UPS power.  Can you think of how to calculate it?  thank you

kellerza, 2023-12-27:
Today there are some calculated sensors - or MathSensors - see https://github.com/kellerza/sunsynk/blob/655ce5c0a2a5fe8edd6139398ee02a4ccb6156b0/src/sunsynk/definitions.py#L100 for some examples

Of you want to build a new one, try to view all the interested registers at the same time to see how they are related. This works really nice in the template developer tools once you’ve added the individual sensors (or even dev tools - states, if you can filter on a similar name)

cc120689, 2023-12-27:
ok, thank you very much, I'm going to try and see if I can figure it out, I'll add them in mysensors.py, right?  Greetings and Merry Christmas!

kellerza, 2023-12-29:
Yes, add them in mysensors, as an example:

```python
SENSORS += [
Sensor(175, "register 175", WATT),
Sensor(169, "register 169", WATT),
Sensor(166, "register 166", WATT),
]
```

Then also make sure your schedule for WATT reads often,- i.e. every second or 2 - https://kellerza.github.io/sunsynk/reference/schedules

Then you can play with templates to do a calc until you find the combination that works. Once you find it create a `MathSensor` or create you own custom sensor to do the calc between the registers if this is more than + & -

cc120689, 2023-12-29:
Ok thanks!!

kellerza, 2024-01-02:
You will have to look at the code. Sensors.py likely

the mathsensor simply adds registers together. Some + some -, depending on the sign of the factor

cc120689, 2024-01-03:
Ah ok only + or -  are possible

closed 2026-08-15 by kellerza