kellerza/sunsynk #215: Upgrading from 0.6.4 to 0.6.5 fails

jrkk0925, 2023-12-09:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

A clear and concise description of what the bug is.

Updating to v0.6.5 returns an error
 
Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:0.6.5: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=0.6.5&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")

**Expected behavior**
A clear and concise description of what you expected to happen.

### Your environment

Core 2023.12.1
Supervisor 2023.11.6
Operating System 11.2
Frontend 20231208.2

### Logs

```Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:138
Integration: Home Assistant Supervisor ([documentation](https://www.home-assistant.io/integrations/hassio), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+hassio%22))
First occurred: 11:52:20 AM (1 occurrences)
Last logged: 11:52:20 AM

Failed to to call /store/addons/4bb530c7_hass-addon-sunsynk-multi/update - Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:0.6.5: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=0.6.5&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")

theowinter, 2023-12-09:
Same for me:

`Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:0.6.5: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=0.6.5&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Famd64&platform=linux%2Famd64: Internal Server Error ("manifest unknown")`

rogeralexanderdean, 2023-12-09:
Same issue here

[image]

kellerza, 2023-12-09:
It should now be available

theowinter, 2023-12-09:
Da'sy! Sukses! Dankie Johann.

closed 2023-12-27 by kellerza