kellerza/sunsynk #224: Confusion over adding PV2 current and voltage sensors

cliffcourt, 2024-01-27:
I have added this HA add-on to view my Deye 8kW inverter and related parameters. A big thank you to @kellerza for the work he put into this. Having added it using the github repository, I'm now viewing the data using the sunsynk-power-flow-card off HACS.

My problem is the PV2.current and PV2.voltage values are not coming through. Only the PV2.power value is coming through, although all 3 values are coming through for PV1. In reading Issue 214, I saw @kellerza responded to this (https://github.com/kellerza/sunsynk/issues/214#issuecomment-1848419032) by indicating that these parameters (PV2.current and PV2.voltage) might not be in the current definitions file (I assume this is the definitions.py file in the github repo) and can be added as custom sensors using mysensors described in the docs.

So I looked at the docs and can see how to add new parameters as custom sensors with mysensors configuration. The problem is that when I look at the definitions.py file in the github for this add-on, these sensors are defined there (See below:
[image]

So I'm now confused a) why these 2 missing parameters are not coming through already since they exist in the definitions.py file, and b) if I do add them using the custom sensors method, will they not then clash with the ones that are meant to be there.

Finally, in looking at the MQTT payload, for the PV parameters, I see the 3 PV1 parameters, but only the power parameter for PV2 (i.e. PV2 voltage and current are missing), so I'm unsure if this is because they just aren't coming in from the inverter at all, or I'm only seeing the payload with the currently defined batch of sensors (which seem to exclude the 2 missing parameters. (see below)
[image]

I'm still very new to HA and figuring out where everything. So any assistance to point me in the right direction will be much appreciated.

kellerza, 2024-01-27:
Can you post your addon config?

the yaml version of it

cliffcourt, 2024-01-28:
Thanks Kellerza. Here is the YAML for the SS card:

type: entities
entities:
  - entity: sensor.ss_aux_power
    name: AUX power
  - entity: sensor.ss_battery_current
    name: Battery current
  - entity: sensor.ss_battery_power
    name: Battery power
  - entity: sensor.ss_battery_soc
    name: Battery SOC
  - entity: sensor.ss_battery_voltage
    name: Battery voltage
  - entity: sensor.ss_cb_stats
    name: Callback stats
  - entity: sensor.ss_day_battery_charge
    name: Day Battery Charge
  - entity: sensor.ss_day_battery_discharge
    name: Day Battery Discharge
  - entity: sensor.ss_day_grid_export
    name: Day Grid Export
  - entity: sensor.ss_day_grid_import
    name: Day Grid Import
  - entity: sensor.ss_day_load_energy
    name: Day Load Energy
  - entity: sensor.ss_day_pv_energy
    name: Day PV Energy
  - entity: sensor.ss_essential_power
    name: Essential power
  - entity: binary_sensor.ss_grid_connected
    name: Grid Connected
  - entity: sensor.ss_grid_ct_power
    name: Grid CT power
  - entity: sensor.ss_grid_frequency
    name: Grid frequency
  - entity: sensor.ss_grid_power
    name: Grid power
  - entity: sensor.ss_grid_voltage
    name: Grid voltage
  - entity: sensor.ss_inverter_current
    name: Inverter current
  - entity: sensor.ss_inverter_power
    name: Inverter power
  - entity: sensor.ss_load_frequency
    name: Load frequency
  - entity: sensor.ss_non_essential_power
    name: Non-Essential power
  - entity: sensor.ss_overall_state
    name: Overall state
  - entity: sensor.ss_pv1_current
    name: PV1 current
  - entity: sensor.ss_pv1_power
    name: PV1 power
  - entity: sensor.ss_pv1_voltage
    name: PV1 voltage
  - entity: sensor.ss_pv2_power
    name: PV2 power
  - entity: sensor.ss_timeout
    name: RS485 timeout
  - entity: sensor.ss_total_battery_charge
    name: Total Battery Charge
  - entity: sensor.ss_total_battery_discharge
    name: Total Battery Discharge
  - entity: sensor.ss_total_grid_export
    name: Total Grid Export
  - entity: sensor.ss_total_grid_import
    name: Total Grid Import
  - entity: sensor.ss_total_pv_energy
    name: Total PV Energy
title: SS

I tried adding -entity:sensor.ss_pv2_voltage and name PV2 Voltage, but it gives an error saying can't find the entities.

kellerza, 2024-01-28:
And the addon’s config?

cliffcourt, 2024-01-28:
@kellerza Thanks, I have managed to resolve my problem by using your custom sensors process. Thanks to your prompting (even though it may not be obvious how), it allowed me to realise that since I was not seeing the pv2 voltage or current in the MQTT payload, they were almost certainly not in the definitions that I installed, and therefore it was logical to try the custom sensor process as per your documentation.

So after following the custom sensors instructions, making and fixing a bunch of syntax errors on my side in the mysensors.py file, and restarting the add-on, I'm very happy to report that both pv2 current and voltage parameters are now coming through and are visible in my sunsynk-power-flow-card.

Many thanks again for your add-on and also your very rapid response and willingness to assist me when I was stuck!

kellerza, 2024-01-29:
👍 as per your first message, if its already in the definitions, you should simply add it to the addon config

glad you got it working!

closed 2024-01-29 by kellerza