kellerza/sunsynk #226: Can't install on Docker

championc, 2024-01-31:
I'm trying to install on docker, using the Docker Compose example in the documentation, but it creates a OPTIONS.YAML Folder / Directory

The logfile is showing this

[code]
s6-rc: info: service legacy-services successfully started
17:29:47 INFO    Loading configuration: /data/options.yaml
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 79, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 61, in main
    init_options()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/options.py", line 87, in init_options
    txt = fpath.read_text(encoding="utf-8")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/pathlib.py", line 1058, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/pathlib.py", line 1044, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
IsADirectoryError: [Errno 21] Is a directory: '/data/options.yaml'
[17:29:47] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
17:29:52 INFO    Loading configuration: /data/options.yaml
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 79, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 61, in main
    init_options()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/options.py", line 87, in init_options
    txt = fpath.read_text(encoding="utf-8")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/pathlib.py", line 1058, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/pathlib.py", line 1044, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
IsADirectoryError: [Errno 21] Is a directory: '/data/options.yaml'
[17:29:52] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
17:29:57 INFO    Loading configuration: /data/options.yaml
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 79, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 61, in main
    init_options()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/options.py", line 87, in init_options
    txt = fpath.read_text(encoding="utf-8")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/pathlib.py", line 1058, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/pathlib.py", line 1044, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
IsADirectoryError: [Errno 21] Is a directory: '/data/options.yaml'
[17:29:57] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
[/code]

championc, 2024-02-01:
I want to manually install a copy of OPTIONS.YAML - but how can I find the folder / directory that the add-on is looking for ?  I have copied it to several locations, but every time it says

09:56:44 INFO    Loading configuration: /data/options.yaml
09:56:44 ERROR   No configuration file found

rixxxx, 2024-02-05:
If you are using docker-compose.yaml from https://kellerza.github.io/sunsynk/guide/standalone-deployment then options.yaml file should be in the same directory where you run docker-compose. Probably the same directory where docker-compose.yaml is. You can change ${PWD} to some other directory if needed.

kellerza, 2024-02-06:
Thx @rixxxx 

This (from the docker-compose file, maps the options file in the local dir into the container at `/data/options.yaml`

```yaml
    volumes:
      - ${PWD}/options.yaml:/data/options.yaml
 ```

You could also try
```yaml
    volumes:
      - options.yaml:/data/options.yaml
 ```

If `options.yaml` does not exist, it WILL create a directory!!

closed 2024-02-06 by kellerza