kellerza/sunsynk #228: Readings start timing out after a few reads

heinrich321, 2024-02-05:
### Issue related to the Python sunsynk library

### Describe the issue/bug

Reading the inverter using the Waveshare RS485 module, works fine for a few minutes somethimes a few hours then starts timing out and throwing Frame not ready errors.

Inverter used: [Sunsynk 50kW HV Three Phase Hybrid Inverter](https://shop.crystalcom.co.za/products/sunsynk-50kw-hv-three-phase-hybrid-inverter)

It then starts working again when I restart the service.

### Config

```
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2303218580"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: three-phase-hv
SENSORS:
  - energy_management
  - power_flow_card
  - pv_power
  - load_power
  - total_active_energy
  - pv2_power
  - pv3_power
  - battery_low_capacity
  - grid_l1_power
  - grid_l2_power
  - grid_l3_power
  - grid_l1_voltage
  - grid_l2_voltage
  - grid_l3_voltage
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_BY: 80
  - KEY: RW
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: any_unit
    READ_EVERY: 30
    REPORT_EVERY: 60
    CHANGE_BY: 80
NUMBER_ENTITY_MODE: auto
MQTT_HOST: 10.0.0.151
MQTT_PORT: 1883
MQTT_USERNAME: xxxxx
MQTT_PASSWORD: xxxxx
DEBUG: 1
```

### Logs (if applicable)

```

2024-02-05 06:09:59,140 DEBUG   pymodbus.logging Processing: 0x3 0x2 0xb 0x6e 0x3e 0x98
2024-02-05 06:09:59,140 DEBUG   pymodbus.logging Frame - not ready
2024-02-05 06:10:00,052 ERROR   ha_addon_sunsynk_multi.errors (1 in 5 min) OSError in read_ss: timeout reading 3 registers from 115; timeout reading 2 registers from 141; timeout reading 14 registers from 146; timeout reading 12 registers from 166; timeout reading 1 registers from 194; timeout reading 1 registers from 500; timeout reading 2 registers from 506; timeout reading 13 registers from 514; timeout reading 1 registers from 529; timeout reading 2 registers from 534; timeout reading 5 registers from 587; timeout reading 3 registers from 598; timeout reading 1 registers from 609; timeout reading 1 registers from 619; timeout reading 4 registers from 622; timeout reading 1 registers from 636; timeout reading 1 registers from 653; timeout reading 6 registers from 672 poll_need_to_read
2024-02-05 06:10:00,052 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/timeout, 200
2024-02-05 06:10:00,053 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/cb_stats, 0
2024-02-05 06:10:00,054 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/cb_stats_attr, {"period": "120s", "long_calls": "0/0", "long_calls_avg": "0s", "slips": "120/120", "slips_avg": "70.50s", "busy": 120}
2024-02-05 06:10:01,081 DEBUG   pymodbus.logging send: 0x1 0x3 0x2 0x8d 0x0 0x1 0x15 0x99
2024-02-05 06:10:01,081 DEBUG   pymodbus.logging Adding transaction 1
2024-02-05 06:10:01,124 DEBUG   pymodbus.logging recv: 0x1 0x3 0x2 old_data:  addr=None
2024-02-05 06:10:01,124 DEBUG   pymodbus.logging Processing: 0x1 0x3 0x2
2024-02-05 06:10:01,124 DEBUG   pymodbus.logging Frame - not ready
2024-02-05 06:10:01,140 DEBUG   pymodbus.logging recv: 0x15 0x32 0x37 0x1 old_data:  addr=None
2024-02-05 06:10:01,140 DEBUG   pymodbus.logging Processing: 0x15 0x32 0x37 0x1
2024-02-05 06:10:01,140 DEBUG   pymodbus.logging Frame - not ready

```

After retart logs:

```
2024-02-05 06:15:15,722 DEBUG   sunsynk.sensors prog6_capacity=45% (45,)
2024-02-05 06:15:15,722 DEBUG   sunsynk.state register {3: 12851, 4: 12339, 5: 12849, 6: 14389, 7: 14384, 20: 41248, 21: 7, 115: 15, 116: 20, 117: 25, 148: 100, 149: 500, 150: 900, 151: 1300, 152: 1700, 153: 2100, 141: 0, 142: 1, 146: 255, 147: 0, 154: 5000, 155: 5000, 156: 5000, 157: 5000, 158: 5000, 159: 5000, 166: 45, 167: 40, 168: 50, 169: 50, 170: 50, 171: 45, 172: 1, 173: 1, 174: 1, 175: 1, 176: 1, 177: 1, 194: 0, 500: 2, 506: 0, 507: 0, 514: 6, 515: 153, 516: 25832, 517: 0, 518: 25359, 519: 0, 520: 40, 521: 0, 522: 40265, 523: 0, 524: 10, 525: 0, 526: 196, 529: 25, 534: 25191, 535: 1, 587: 5275, 588: 40, 589: 40, 590: 65523, 591: 65511, 598: 2351, 599: 2355, 600: 2382, 609: 4979, 619: 32, 622: 773, 623: 767, 624: 773, 625: 2313, 636: 3090, 653: 5403, 672: 138, 673: 110, 674: 141, 675: 0, 676: 5571, 677: 24} = 45 (old=45)
2024-02-05 06:15:15,722 DEBUG   sunsynk.state register {3: 12851, 4: 12339, 5: 12849, 6: 14389, 7: 14384, 20: 41248, 21: 7, 115: 15, 116: 20, 117: 25, 148: 100, 149: 500, 150: 900, 151: 1300, 152: 1700, 153: 2100, 141: 0, 142: 1, 146: 255, 147: 0, 154: 5000, 155: 5000, 156: 5000, 157: 5000, 158: 5000, 159: 5000, 166: 45, 167: 40, 168: 50, 169: 50, 170: 50, 171: 45, 172: 1, 173: 1, 174: 1, 175: 1, 176: 1, 177: 1, 194: 0, 500: 2, 506: 0, 507: 0, 514: 6, 515: 153, 516: 25832, 517: 0, 518: 25359, 519: 0, 520: 40, 521: 0, 522: 40265, 523: 0, 524: 10, 525: 0, 526: 196, 529: 25, 534: 25191, 535: 1, 587: 5275, 588: 40, 589: 40, 590: 65523, 591: 65511, 598: 2351, 599: 2355, 600: 2382, 609: 4979, 619: 32, 622: 773, 623: 767, 624: 773, 625: 2313, 636: 3090, 653: 5403, 672: 138, 673: 110, 674: 141, 675: 0, 676: 5571, 677: 24} = On (old=On)
2024-02-05 06:15:15,722 DEBUG   sunsynk.sensors prog6_power=50000W (5000,)
2024-02-05 06:15:15,722 DEBUG   sunsynk.state register {3: 12851, 4: 12339, 5: 12849, 6: 14389, 7: 14384, 20: 41248, 21: 7, 115: 15, 116: 20, 117: 25, 148: 100, 149: 500, 150: 900, 151: 1300, 152: 1700, 153: 2100, 141: 0, 142: 1, 146: 255, 147: 0, 154: 5000, 155: 5000, 156: 5000, 157: 5000, 158: 5000, 159: 5000, 166: 45, 167: 40, 168: 50, 169: 50, 170: 50, 171: 45, 172: 1, 173: 1, 174: 1, 175: 1, 176: 1, 177: 1, 194: 0, 500: 2, 506: 0, 507: 0, 514: 6, 515: 153, 516: 25832, 517: 0, 518: 25359, 519: 0, 520: 40, 521: 0, 522: 40265, 523: 0, 524: 10, 525: 0, 526: 196, 529: 25, 534: 25191, 535: 1, 587: 5275, 588: 40, 589: 40, 590: 65523, 591: 65511, 598: 2351, 599: 2355, 600: 2382, 609: 4979, 619: 32, 622: 773, 623: 767, 624: 773, 625: 2313, 636: 3090, 653: 5403, 672: 138, 673: 110, 674: 141, 675: 0, 676: 5571, 677: 24} = 50000 (old=50000)
2024-02-05 06:15:15,723 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/inverter_power, 3236
2024-02-05 06:15:15,724 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/grid_l2_power, 1225
2024-02-05 06:15:15,725 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/load_power, 6603
2024-02-05 06:15:15,726 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/grid_power, 3367
2024-02-05 06:15:15,729 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2303218580/grid_l3_power, 1340
```

rixxxx, 2024-02-05:
I'm using solarman diver and connect directly over tcp to the logger. I have to do some wiring before I can test RS485. Can you try to use other pins on RS485 port? 
[image]

heinrich321, 2024-02-05:
Upon further investigation im relatively certain its a buffer / frame related issue. I have tried changing the BATCH size down to 1 and up to 60 with larger gaps as well as lower read frequency.

1. It never recovers when it enters this state
2. The data which is received from the RS485 dongle, when manually stitched together checks out (CRC and data)

Example:

Sent data: 0x1 0x3 0x0 0x73 0x0 0x3 0xf4 0x10

01 | Slave address | 0x01 (1)
03 | Function code | 0x03 (3) - Read Holding Registers
00 73 | Starting address | Physical: 0x0073 (115)Logical: 0x0074 (116)
00 03 | Quantity | 0x0003 (3)
F4 10 | CRC | 0xF410 (62480)

Response:  0x1 0x3 0x6 0x0 0xf 0x0 0x14 0x0 0x19 0xf4 0xba

01 | Slave address | 0x01 (1)
03 | Function code | 0x03 (3) - Read Holding Registers
06 | Byte count | 0x06 (6)
00 0F 00 14 00 19 | Register value | 0x000F (15), 0x0014 (20), 0x0019 (25)
F4 BA | CRC | 0xF4BA (62650)

```
2024-02-05 18:30:56,063 DEBUG   pymodbus.logging send: 0x1 0x3 0x0 0x73 0x0 0x3 0xf4 0x10
2024-02-05 18:30:56,064 DEBUG   pymodbus.logging Adding transaction 1
2024-02-05 18:30:56,104 DEBUG   pymodbus.logging recv: 0x1 0x3 0x6 0x0 0xf 0x0 0x14 0x0 0x19 old_data:  addr=None
2024-02-05 18:30:56,104 DEBUG   pymodbus.logging Processing: 0x1 0x3 0x6 0x0 0xf 0x0 0x14 0x0 0x19
2024-02-05 18:30:56,105 DEBUG   pymodbus.logging Frame - not ready
2024-02-05 18:30:56,120 DEBUG   pymodbus.logging recv: 0xf4 0xba old_data:  addr=None
2024-02-05 18:30:56,120 DEBUG   pymodbus.logging Processing: 0xf4 0xba
2024-02-05 18:30:56,121 DEBUG   pymodbus.logging Frame - not ready
```

@kellerza (Johann) I can only think that something is funky with the buffer / frame handling. Any help greatly appreciated. Very strange that it works when I restart it, then ends up in this state where I can see valid data coming in, but the plugin cannot parse the frames.

heinrich321, 2024-02-05:
@kellerza think I found the issue.

Pymodbus recently fixed an issue in their package. Your pre-built release from Dec still contains this old version

kellerza, 2024-02-06:
@heinrich321 have you tried using mbusd? I use that and have never had any issues with buffers

I bumped the pymodbus version and the dev addon should build with the new pymodbus
https://github.com/kellerza/sunsynk/commit/74f32247b516b87bf573d25c7870bffc5e9657bd

heinrich321, 2024-02-06:
Thank @kellerza Johann, I should probably have tried that. Assumed the other two drivers are tcp only. Ive switched to umodbus, will let you know how it goes. Thanks for bumbing the version of pymodbus, although im nto sure it will solve the isue. Seems like the frame issue might still be an open one on pymodbus.

heinrich321, 2024-02-07:
Switching to umodbus resolved the isue thanks!

closed 2024-02-07 by heinrich321