kellerza/sunsynk #230: Standalone, where do I define the USB device?

gurmukhp, 2024-02-06:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

I've copied over options.yaml and the error I'm getting is:
`sunsynk-multi-1  | 17:57:45 INFO    No response on the Modbus interface tcp://mbusd:502, see https://kellerza.github.io/sunsynk/guide/fault-finding`

I haven't defined the dev/USB** device anywhere, I've just used the bundled mbus.

Do I need to define the USB device anywhere or should it 'just work'? I'm using the WAVE to USB adaptor.

**Expected behavior**
Connect successfullly

### Your environment

- Home Assistant version: 2023.8.x
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter:
  - Make: Sunsynk
  - Model: Ecco 3.5
  - Firmware: 
- Adaptor details:
  - RS485 WAVE to USB

### Your configuration

```yaml
---
DRIVER: umodbus
INVERTERS:
- SERIAL_NR: '007'
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: tcp://mbusd:502
SENSOR_DEFINITIONS: single-phase
SENSORS:
- total_pv_energy
- total_battery_charge
- total_battery_discharge
- total_grid_export
- total_grid_import
- battery_soc
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_SENSORS_BATCH_SIZE: 60
NUMBER_ENTITY_MODE: auto
MQTT_HOST: 192.168.68.53
MQTT_PORT: 1883
MQTT_USERNAME: ''
MQTT_PASSWORD: ''
DEBUG: 0
DEBUG_DEVICE: "/dev/ttyAMA0"
```

### Logs

```logs

sunsynk-multi-1  | 17:57:44 INFO    Using Single phase sensor definitions.
sunsynk-multi-1  | 17:57:44 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
sunsynk-multi-1  | 17:57:44 INFO    Connecting to tcp://mbusd:502
sunsynk-multi-1  | 17:57:44 INFO    Reading startup sensors Serial, Rated power
sunsynk-multi-1  | 17:57:44 ERROR   gaierror reading 15 registers from 3: [Errno -2] Name does not resolve
sunsynk-multi-1  | 17:57:44 ERROR   gaierror reading 15 registers from 3: [Errno -2] Name does not resolve
sunsynk-multi-1  | 17:57:44 ERROR   gaierror reading 15 registers from 3: [Errno -2] Name does not resolve
sunsynk-multi-1  | 17:57:44 WARNING Retrying individual sensors: ['serial', 'rated_power']
sunsynk-multi-1  | 17:57:45 ERROR   gaierror reading 5 registers from 3: [Errno -2] Name does not resolve Serial
sunsynk-multi-1  | 17:57:45 ERROR   gaierror reading 2 registers from 16: [Errno -2] Name does not resolve Rated power
sunsynk-multi-1  | 17:57:45 CRITICAL Could not read sensors: ['Serial', 'Rated power']
sunsynk-multi-1  | 17:57:45 INFO    ############################################################
sunsynk-multi-1  | 17:57:45 INFO    No response on the Modbus interface tcp://mbusd:502, see https://kellerza.github.io/sunsynk/guide/fault-finding
sunsynk-multi-1  | 17:57:45 INFO    ############################################################
sunsynk-multi-1  | 17:57:45 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
```

kellerza, 2024-02-06:
The USB port should be configured where you have the USB device plugged in.

If you refer to mbusd, you have to either run mbusd natively or on Docker. In this case you specify the USB config in mbusd

Otherwise you need to identify, and configure the USB port in the options.yaml

gurmukhp, 2024-02-06:
Thanks for the quick reply, I'm not running any additional services for mbus, just the hass-addon-sunsynk-multi image. Should it be something like this?

```
DRIVER: umodbus
INVERTERS:
  - PORT: serial:///dev/ttyUSB0
 ```

kellerza, 2024-02-06:
Yes, that looks better. I would use the pymodbus driver though, not umodbus

gurmukhp, 2024-02-06:
Thanks, do I need to run any additional services in Docker or is just the main Sunsynk one enough?

gurmukhp, 2024-02-06:
I got it working, I had to pass in the device to the docker compose file too::

```
services:
  sunsynk-multi:
    restart: unless-stopped
    image: ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:stable
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    volumes:
      - ./options.yaml:/data/options.yaml
 ```

```
---
DRIVER: pymodbus
INVERTERS:
- SERIAL_NR: '007'
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: /dev/ttyUSB0
```

I could document this and raise a PR if you want?

kellerza, 2024-02-12:
A PR would be welcome & help people in the future. Thanks!

gurmukhp, 2024-02-13:
PErfect, I'll do that! thanks

closed 2024-02-13 by gurmukhp

championc, 2024-02-18:
That fixed my issue too.  I was resorting to using TCP access via the Logger, but I wanted to use a cable.

gurmukhp, 2025-10-08:
Apologies, I didn't get round to this. I ended up switching to Solar Assistant because this integration kept going offline (and nothing was sent to MQTT).

If you have a docker setup, you will have a docker-compose.yml file. The config I pasted should go into that. If you don't run via Docker, then it might be worth opening a new bug.