kellerza/sunsynk #232: Incorrect Grid power, Grid CT power, and Non-Essential power

alistairts, 2024-02-07:
Deye 5kW Hybrid Inverter
Firmware: MAIN：3382-1515
Deye Inverter Add-on (multi)

Grid power is the power going passing through the GRID port of the inverter, Grid CT power is the passing through the CT, and Non-Essential power is the power which T's off between the inverters GRID port and the CT.

What I am observing on my integration is that Grid power equals Grid CT power when Non-Essential power is none-zero.

Grid CT power should be equal to the sum of both Grid power and Non-Essential power.

kellerza, 2024-02-07:
What is your load_limit set to?

alistairts, 2024-02-08:
Hi Johann

Thanks for getting back to me.

My load limit is set to "Zero Export".

Kind regards
Alistair

alistairts, 2024-02-08:
Hi Johann

I've noticed now that my essential loads power is also not correct, and is
somehow linked to the non-essential power.

Below are two screenshots showing with my geyser turned on, and then with
my geyser turned off.

[image: image.png]

[image: image.png]

Kind regards

kellerza, 2024-02-12:
@alistairts, I don't see any image. 

Something about your wiring/setup requires the sensor to be different from the norm. My suggestion would be that you create your own custom sensor/mysensors to correctly show these values.

Once you have a working sensor, we could see if there is some register to read determine what to use.

You can see the current definitions and some history for essential power here, the rest of the file contains non-essential etc - https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py#L92-L108

alistairts, 2024-02-12:
> Hi Johaan
>
> I guess github doesn't allow sending of pictures.
>
> I will do as you've suggested, but how do I get to your code? as your
> add-on does not show up in my custom_componets file.
>
> Kind regards

Hi Johan

I've been reading up on your custom sensors page, and I can't seem to get
the mysensors.py file to work.
Where exactly must these folders "/share/hass-addon-sunsynk/" go in my HA?
I've put it under my main CONFIG folder, but your addon is not importing it.

It would be easier to simply edit your sensor definitions file, where can I
find that to edit it?

Kind regards

RhysMcW, 2024-02-13:
> I've noticed now that my essential loads power is also not correct, and is somehow linked to the non-essential power.

I'm using the sensor "Essential 1 power" for my essential power, which excludes the non-essential in the calc

with regards to your grid power - I'm using sensor "Grid LD power" as that is basically the CT reading minus the non-essential, which could be a negative value if your inverter is feeding to the non-essential load

alistairts, 2024-02-13:
Thank you RhysMcW, that did the trick and I didn't have to create any custom sensors.
Replying on the website and not via email allows me to attach pictures, attached is a picture showing all powers balance out, but that consumed by the inverter...
Thanks again Johann for the work you put into this integration.
[image]

kellerza, 2024-02-28:
Looking at my config today I also use `essential_1_power`

Adding this to the Template Developer Tool in HASS
```jinja2
essential_1_power + non_essential_power
{{ states.sensor.ss_essential_1_power.state }} + {{ states.sensor.ss_non_essential_power.state }} = {{ int( states.sensor.ss_essential_1_power.state) + int(states.sensor.ss_non_essential_power.state) }}

essential_power: {{ states.sensor.ss_essential_power.state }}
     load_power: {{ states.sensor.ss_load_power.state }}
```

gives me

```
essential_1_power + non_essential_power
2341 + 3284.833 = 5625

essential_power: 5595
     load_power: 5595
```

closed 2024-02-28 by kellerza