kellerza/sunsynk #238: Mbusd Docker Container

championc, 2024-02-18:
Is there details of how to create an mbusd docker container please ?  I want to be able to share the /dev/ttyUSB0 port with the sunsynk-multi, node-red and homeassistant docker containers, and it seems like the way to go

kellerza, 2024-02-19:
Try the instructions https://kellerza.github.io/sunsynk/guide/standalone-deployment#mbusd

championc, 2024-02-19:
Ok, I'm doing something wrong

My docker-compose.yaml file contains the following

[code]
  mbusd:
    container_name: mbusd
    restart: unless-stopped
    image: ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:stable
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    volumes:
      - /dev/ttyUSB0:/dev/ttyUSB0
[/code]

I got the following errors after running docker compose up mbusd

[code]
root@Solar-M93:/opt# docker compose up mbusd
[+] Running 1/0
 ✔ Container mbusd  Created                                                                                                                                                                               0.0s
Attaching to mbusd
mbusd  | s6-rc: info: service s6rc-oneshot-runner: starting
mbusd  | s6-rc: info: service s6rc-oneshot-runner successfully started
mbusd  | s6-rc: info: service fix-attrs: starting
mbusd  | s6-rc: info: service fix-attrs successfully started
mbusd  | s6-rc: info: service legacy-cont-init: starting
mbusd  | s6-rc: info: service legacy-cont-init successfully started
mbusd  | s6-rc: info: service legacy-services: starting
mbusd  | services-up: info: copying legacy longrun sunsynk (no readiness notification)
mbusd  | s6-rc: info: service legacy-services successfully started
mbusd  | 09:39:53 ERROR   No configuration file found
mbusd  | 09:39:53 INFO    sunsynk library version: 0.6.4
mbusd  | 09:39:53 INFO    Schedules:
mbusd  | +-----------+-----+------+--------+-----------+----------+------------+
mbusd  | |    Key    | src | Read | Report | Change by | Change % | Change any |
mbusd  | +-----------+-----+------+--------+-----------+----------+------------+
mbusd  | | date_time |     |  60  |   60   |           |          |    True    |
mbusd  | |     rw    |     |  5   |  300   |           |          |    True    |
mbusd  | |     w     |     |  5   |   60   |     80    |          |            |
mbusd  | |    kwh    |     | 300  |  300   |           |          |            |
mbusd  | |  any_unit |     |  15  |  300   |           |          |            |
mbusd  | |  no_unit  |     |  15  |  300   |           |          |    True    |
mbusd  | +-----------+-----+------+--------+-----------+----------+------------+
mbusd  | 09:39:53 INFO    Using Single phase sensor definitions.
mbusd  | 09:39:53 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
mbusd  | Traceback (most recent call last):
mbusd  |   File "<frozen runpy>", line 198, in _run_module_as_main
mbusd  |   File "<frozen runpy>", line 88, in _run_code
mbusd  |   File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 79, in <module>
mbusd  |     main()
mbusd  |   File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 74, in main
mbusd  |     loop.run_until_complete(main_loop(loop))
mbusd  |   File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
mbusd  |     return future.result()
mbusd  |            ^^^^^^^^^^^^^^^
mbusd  |   File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 31, in main_loop
mbusd  |     MQTT.availability_topic = f"{SS_TOPIC}/{OPT.inverters[0].serial_nr}/availability"
mbusd  |                                             ~~~~~~~~~~~~~^^^
mbusd  | IndexError: list index out of range
mbusd  | [09:39:53] WARNING: Halt add-on with exit code 1
mbusd  | s6-rc: info: service legacy-services: stopping
mbusd  | s6-rc: info: service legacy-services successfully stopped
mbusd  | s6-rc: info: service legacy-cont-init: stopping
mbusd  | s6-rc: info: service legacy-cont-init successfully stopped
mbusd  | s6-rc: info: service fix-attrs: stopping
mbusd  | s6-rc: info: service fix-attrs successfully stopped
mbusd  | s6-rc: info: service s6rc-oneshot-runner: stopping
mbusd  | s6-rc: info: service s6rc-oneshot-runner successfully stopped
mbusd exited with code 0
[/code]

championc, 2024-02-19:
Ok, I see my first problem

What is the correct image to use

I try
https://ghcr.io/v2/kellerza/hass-addon-mbusd/amd64:stable

But this does not work

championc, 2024-02-19:
Ok, something not right

All looks OK with mbusd
**_s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started_**

But from Sunsynk-Multi
 **_No response on the Modbus interface tcp://mbusd:502_**

kellerza, 2024-02-22:
What is 192.168.1.22? The host address?
What do you use in the addon?

kellerza, 2024-02-22:
If you move away from docker name resolution (tcp://mbusd:502) - you probably want to expose the ports

closed 2024-06-18 by kellerza