kellerza/sunsynk #246: Prog{x} Time select list

RhysMcW, 2024-02-21:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

The Prog{x} Time select lists populate with the time in 15 minute increments. The Sunsynk system mode settings adjust the time in 5 minute increments

**Expected behavior**
Prog{x} Time select lists to populate in 5 minute increments, instead of the 15 minute interval it currently uses

### Your environment

- Home Assistant version: 2024.2.2
- Addon:
  - Name: sunsynk-multi
  - Version: 0.6.5
- Inverter:
  - Make: Sunsynk
  - Model: 5.5k
  - Firmware: COMM: E426 MCU:3382-1515
- Adaptor details:
  - RS485
  - RS485 USB to RS485 - 3 Pin from https://www.robotics.org.za/RS485-3P?search=rs485-3p

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "1234"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - Battery temperature
  - Battery Shutdown Capacity
  - DC transformer temperature
  - PV2 voltage
  - PV2 current
  - Radiator temperature
  - Load power
  - Date Time
  - Essential 1 power
  - Load Limit
  - Grid LD power
  - mysensors
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: true
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
  - KEY: grid_voltage
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: true
    CHANGE_BY: 5
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: box
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: 
MQTT_PASSWORD: 

```

### Logs

```logs

paste logs in 3x backticks...

```

kellerza, 2024-02-21:
But does it make any sense to have these fine-grained intervals?

The Inverter clock is typically running freely (despite attempts to write, it's not that successful)

With 15 minutes, most of these dropdown menus already scroll on a 4K screen, 5 minutes will make it 3x as long,- is this really usable?

RhysMcW, 2024-02-21:
I have various functions that adjust some of the times based on things like sunrise and sunset, and adjust by the 5 minute interval.
This was not a problem for me when I was using NodeRed and it's modbus to update the inverter daily but had to disable the NodeRed modbus to enable this add-on.
With the Prog{x} Time select lists only having 15 minutes intervals I'm not able to change the value to something not in the select list.
I understand this would create a longer list to select from, maybe then parameter driven from the config so users can select their own interval...

RhysMcW, 2024-02-25:
I've been trying with the custom sensors.
I defined one as "Sensor(255, "ProgZ6 Time")" which shows up as sensor.ss_progz6_time but does not update when I try from Nodered, sort of makes sense as the "Sensor" is defined under the sunsynk.sensors which look to be read-only.
I then defined one as "RWSensor(255, "ProgZ6 TimeZ")" which shows up as text.ss_progz6_timez but when I try set the value from Nodered I get the following in the log, and the entity does not change to the value I tried to set it to (1825 as used by the inverter register)...
_16:49:27 INFO    Queue update progz6_timez=1825
16:49:28 ERROR   NotImplementedError in read_ss:_ 
or if I try set it as a "time" format 18:25, then the log shows
_17:52:20 INFO    Queue update progz6_timez=18:25
17:55:00 ERROR   (1 in 5 min) NotImplementedError in read_ss:_

Any ideas as to what else I could try, please?

closed 2024-02-28 by kellerza