kellerza/sunsynk #247: I have just updated sunsynk add on, and am now getting 2 or 3 decimal places for all values.

f1wade, 2024-02-22:
I have just updated sunsynk add on, and am now getting 2 or 3 decimal places for all values.

Is there a way I can change this default to be 0 decimal places, like it use to be?

I posted the below in discussions, but haven't seen any response.
and looking around the definitions, I can't see anything that suggests this has changed in there, unless its changed more internal to the add-on.

any help greatly appreciated?

_Originally posted by @f1wade in https://github.com/kellerza/sunsynk/discussions/242_

f1wade, 2024-02-22:
@RhysMcW that works for the dashboards, but not for markdown.

I don't understand why they where all showing whole numbers mostly and occationally 1 decimal place before the update I did to both HA os, core and sunsynk the other day, moving up about 6 months of updates, so not sure exactly what change has triggered this.

kellerza, 2024-02-27:
Is 1 decimal place a good enough suggestion for most sensors?

Hoping this will make a difference - https://github.com/kellerza/sunsynk/commit/0cbfd610012d7c91607c5daa52b56de388168875

closed 2024-06-19 by kellerza