kellerza/sunsynk #251: Unable to successfully connect to Sunsynk inverter

gurmukhp, 2024-02-27:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

With a fresh install, I am unable to connect to the Sunsynk via RS485. If I  define the serial number as an empty string, I get the following:

```
sunsynk-multi-1  | s6-rc: info: service s6rc-oneshot-runner: starting
sunsynk-multi-1  | s6-rc: info: service s6rc-oneshot-runner successfully started
sunsynk-multi-1  | s6-rc: info: service fix-attrs: starting
sunsynk-multi-1  | s6-rc: info: service fix-attrs successfully started
sunsynk-multi-1  | s6-rc: info: service legacy-cont-init: starting
sunsynk-multi-1  | s6-rc: info: service legacy-cont-init successfully started
sunsynk-multi-1  | s6-rc: info: service legacy-services: starting
sunsynk-multi-1  | services-up: info: copying legacy longrun sunsynk (no readiness notification)
sunsynk-multi-1  | s6-rc: info: service legacy-services successfully started
sunsynk-multi-1  | 17:55:55 INFO    Loading configuration: /data/options.yaml
sunsynk-multi-1  | 17:55:55 INFO    sunsynk library version: 0.6.4
sunsynk-multi-1  | 17:55:55 INFO    Schedules:
sunsynk-multi-1  | +-----------+-----+------+--------+-----------+----------+------------+
sunsynk-multi-1  | |    Key    | src | Read | Report | Change by | Change % | Change any |
sunsynk-multi-1  | +-----------+-----+------+--------+-----------+----------+------------+
sunsynk-multi-1  | | date_time |     |  60  |   60   |           |          |    True    |
sunsynk-multi-1  | |     rw    |     |  5   |  300   |           |          |    True    |
sunsynk-multi-1  | |     w     |     |  5   |   60   |     80    |          |            |
sunsynk-multi-1  | |    kwh    |     | 300  |  300   |           |          |            |
sunsynk-multi-1  | |  any_unit |     |  15  |  300   |           |          |            |
sunsynk-multi-1  | |  no_unit  |     |  15  |  300   |           |          |    True    |
sunsynk-multi-1  | +-----------+-----+------+--------+-----------+----------+------------+
sunsynk-multi-1  | 17:55:55 INFO    Using Single phase sensor definitions.
sunsynk-multi-1  | 17:55:55 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
sunsynk-multi-1  | 17:55:55 INFO    Connecting to /dev/ttyUSB0
sunsynk-multi-1  | 17:55:55 INFO    PyModbus 3.5.2 Serial: /dev/ttyUSB0
sunsynk-multi-1  | 17:55:55 INFO    Reading startup sensors Serial, Rated power
sunsynk-multi-1  | 17:56:06 ERROR   timeout reading 15 registers from 3
```

If I set anything else as the serial number, I get the following:

```
sunsynk-multi-1  | 18:00:23 INFO    PyModbus 3.5.2 Serial: /dev/ttyUSB0
sunsynk-multi-1  | 18:00:23 INFO    Reading startup sensors Serial, Rated power
sunsynk-multi-1  | 18:00:23 INFO    ############################################################
sunsynk-multi-1  | 18:00:23 INFO        Serial number mismatch. Expected 007, got 
sunsynk-multi-1  | 18:00:23 INFO    ############################################################
sunsynk-multi-1  | 18:00:23 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
```

I can get the serial number from the app, but even if I add that, it doesn't seem to work.

Is there anything obvious wrong?

### Your environment

- Home Assistant version: 2023.8.x
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter:
  - Make: Sunsynk
  - Model: Ekko
- Adaptor details:
  - RS485/RS232 usb to rs485

### You configuration

```yaml

---
DRIVER: pymodbus
INVERTERS:
- SERIAL_NR: '007'
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
- total_pv_energy
- total_battery_charge
- total_battery_discharge
- total_grid_export
- total_grid_import
- battery_soc
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_SENSORS_BATCH_SIZE: 60
NUMBER_ENTITY_MODE: auto
MQTT_HOST: 192.168.68.68
MQTT_PORT: 1883
MQTT_USERNAME: ''
MQTT_PASSWORD: ''
DEBUG: 0
DEBUG_DEVICE: "/dev/ttyAMA0"
```

gurmukhp, 2024-02-27:
Sometimes the serial number shows up as B)e or sometimes it's B__, always changes or doesn't show?

kellerza, 2024-02-28:
Your inverter is not responding, try the fault-finding section in the docs (setting a modbus ID on the inverter, shorter wire, etc)

gurmukhp, 2024-02-28:
Thanks, I was getting CRC errors with Solar Assistant which led me to this solution

https://solarenergyconcepts.co.uk/practical-and-diy/crc-error-solar-assistant/

closed 2024-02-28 by gurmukhp