kellerza/sunsynk #254: "PV3 Sensor" definition for Sunsynk 16Kw single phase inverter

kresenn, 2024-02-29:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py)

### Sunsynk 16kW Single phase

Details about your Inverter: make/model/firmware

### Describe the issue/bug and what you expect

The definitions.py file does not contain the "PV3 Sensor" definition for a Sunsynk 16kW single phase  inverter as this inverter has 3 Mppts. 
I tried to add it as a custom Sensor but says "Unknown Sensor" in logs. 

Plea

### Logs (if applicable)

```logs

paste logs in 3x backticks...

```

kresenn, 2024-02-29:
from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()
#############
# Solar PV 3
#############
SENSORS += (
    Sensor(188, "PV3 power", WATT, -1),
    Sensor(113, "PV3 voltage", VOLT, 0.1),
    Sensor(114, "PV3 current", AMPS, 0.1),
)

kresenn, 2024-02-29:
20:29:05 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
20:29:05 INFO    Using Single phase sensor definitions.
20:29:05 ERROR   Unknown sensor specified: pv3_power
20:29:05 ERROR   Unknown sensor specified: pv3_current
20:29:05 ERROR   Unknown sensor specified: pv3_voltage
20:29:05 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
20:29:05 INFO    Connecting to /dev/ttyUSB0
20:29:05 INFO    PyModbus 3.6.4 Serial: /dev/ttyUSB0
20:29:05 INFO    Reading startup sensors Battery Low Capacity, Prog5 Time, Battery Shutdown Capacity, Serial, Prog6 Time, Prog2 Time, Prog4 Time, Rated power, Prog3 Time, Prog1 Time
20:29:05 INFO    ############################################################
20:29:05 INFO                 Inverter serial number '****0783'
20:29:05 INFO    ############################################################
20:29:05 INFO    MQTT: Connecting to mqtt@core-mosquitto:1883
20:29:06 INFO    MQTT: Connection successful
20:29:07 INFO    Read every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  5  |      battery_low_capacity, prog4_capacity, non_essential_power, aux_power,       |
|     |      prog6_charge, prog4_charge, priority_load, battery_shutdown_capacity,       |
|     |  prog6_power, prog3_power, grid_ct_power, use_timer, pv1_power, prog2_capacity,  |
|     |   grid_power, prog2_time, prog1_time, prog5_time, prog1_capacity, prog2_power,   |
|     | prog1_power, prog4_power, prog5_charge, prog1_charge, pv2_power, inverter_power, |
|     |      prog4_time, prog3_charge, prog3_time, prog3_capacity, prog5_capacity,       |
|     |    essential_power, prog6_capacity, prog6_time, prog2_charge, battery_power,     |
|     |                             prog5_power, load_limit                              |
|  15 |      pv1_voltage, pv1_current, grid_frequency, pv2_current, load_frequency,      |
|     |     grid_connected, pv2_voltage, grid_voltage, battery_soc, battery_current,     |
|     |                 overall_state, inverter_current, battery_voltage                 |
| 300 |    day_pv_energy, day_grid_import, total_grid_import, day_battery_discharge,     |
|     |        total_battery_charge, day_battery_charge, total_battery_discharge,        |
|     |   total_active_energy, day_grid_export, total_grid_export, total_load_energy,    |
|     |                         day_load_energy, total_pv_energy                         |
+-----+----------------------------------------------------------------------------------+

kellerza, 2024-02-29:
Hi, I expect your custom sensor file is either in the incorrect place/not being picked up

From you log:

```
20:29:05 INFO    Using Single phase sensor definitions.
20:29:05 ERROR   Unknown sensor specified: pv3_power
20:29:05 ERROR   Unknown sensor specified: pv3_current
20:29:05 ERROR   Unknown sensor specified: pv3_voltage
```

Where did you put it?

kresenn, 2024-02-29:
Hi Mr Kellerman

Sorry I am new to this.
Thanks for that.

Think it was a typo after closer inspection.
I put it in /share/sharehass-addon-sunsynk/mysensors.py

I changed to /share/hass-addon-sunsynk/mysensors.py

...And it works;))

Thank you so much

Regards

Kresen

Kind Regards

Kresen Naicker
0825017916

closed 2024-03-01 by kellerza