kellerza/sunsynk #263: Use_timer: Unknown register value 1

martinarva, 2024-03-22:
### Issue related to the Python sunsynk library

### Describe the issue/bug
I get use_timer errors in logs. 

Inverter: Deye-12k-SG04LP3-EU 3ph inverter
Running Sunsynk in docker.

Options.yaml
```
---
DRIVER: solarman
INVERTERS:
- SERIAL_NR: 'xxxxx'
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: tcp://xxxxxxx
  DONGLE_SERIAL_NUMBER: 'xxxxxxxxx'
SENSOR_DEFINITIONS: three-phase
SENSORS:
- energy_management
- load_limit
- battery_max_charge_current
- battery_max_discharge_current
- battery_capacity_current
- grid_charge_battery_current
- grid_charge_enabled
- power_flow_card
- battery_current
- battery_power
- battery_soc
- battery_voltage
- battery_temperature
- day_battery_charge
- day_battery_discharge
- day_grid_export
- day_grid_import
- day_load_energy
- day_pv_energy
- grid_connected
- grid_ct_power
- grid_frequency
- grid_power
- grid_current
- grid_l1_power
- grid_l2_power
- grid_l3_power
- overall_state
- priority_load
- pv1_current
- pv1_power
- pv1_voltage
- pv2_current
- pv2_power
- pv2_voltage
- load_power
- load_l1_power
- load_l2_power
- load_l3_power
- date_time
- inverter_power
- inverter_l1_power
- inverter_l2_power
- inverter_l3_power
- inverter_voltage
- inverter_frequency
- radiator_temperature
- dc_transformer_temperature
- solar_export
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Deye
READ_SENSORS_BATCH_SIZE: 60
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: 192.168.1.35
MQTT_PORT: 1883
MQTT_USERNAME: hass
MQTT_PASSWORD: ''
DEBUG: 0
DEBUG_DEVICE: "/dev/ttyAMA0"
```

### Expected behavior

A clear and concise description of what you expected to happen.

### Logs (if applicable)

```
09:56:36 INFO    Queue update use_timer=ON
09:56:37 INFO    Writing sensor use_timer=ON Registers:{146=0x00ff} 
09:57:15 INFO    Queue update use_timer=OFF
09:57:16 INFO    Writing sensor use_timer=OFF Registers:{146=0x0000} 
09:57:25 WARNING use_timer: Unknown register value 1
09:57:25 WARNING Cannot publish Use Timer: value is None
09:57:30 WARNING use_timer: Unknown register value 1
09:57:35 WARNING use_timer: Unknown register value 1
09:57:40 WARNING use_timer: Unknown register value 1
09:57:46 WARNING use_timer: Unknown register value 1
09:57:50 WARNING use_timer: Unknown register value 1
09:57:55 WARNING use_timer: Unknown register value 1
09:58:01 WARNING use_timer: Unknown register value 1
09:58:05 WARNING use_timer: Unknown register value 1
09:58:10 WARNING use_timer: Unknown register value 1
```

kellerza, 2024-03-29:
What version of the addon are you using? 

The value being written is different from the current sensor definition. Your log is also incomplete.

```
03:12:51 INFO    Queue update use_timer=OFF
03:12:53 INFO    Writing sensor use_timer=OFF Registers:{248=0x00fe} [Register 255-->0]
03:13:18 INFO    Queue update use_timer=ON
03:13:19 INFO    Writing sensor use_timer=ON Registers:{248=0x00ff} [Register 254-->1]
```

martinarva, 2024-04-04:
> What version of the addon are you using?
>
> The value being written is different from the current sensor definition. Your log is also incomplete.
>
> ```
> 03:12:51 INFO    Queue update use_timer=OFF [...]

I'm using latest stable docker image
ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:stable@sha256:b23ae46f97cd6cc0211926aaef007c16b3be6df7d0700b6d9d1e10d6a4b3507d

kellerza, 2024-04-06:
Can you try the edge addon

martinarva, 2024-04-07:
> Can you try the edge addon

Edge addon? I have HA in docker and can't have addons there.

kellerza, 2024-04-07:
Its just a different container image

closed 2024-04-08 by kellerza

martinarva, 2024-04-08:
> Its just a different container image

How should i modify the docker compose:
image: ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:stable

Like this?
image: ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:edge

kellerza, 2024-04-09:
When using Docker you also won't get update notifications and you will have to make sure you pull the image again

```
docker compose pull
```

martinarva, 2024-04-09:
> When using Docker you also won't get update notifications and you will have to make sure you pull the image again when trying to update
> 
> 
> 
> ```
> 
> docker compose pull
> 
> ```

Thanks! I have Watchdog that updates the images daily

martinarva, 2024-04-09:
Now there are more errors. Related to issue https://github.com/kellerza/sunsynk/issues/261

`     
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
05:23:37 INFO    Loading configuration: /data/options.yaml
05:23:37 INFO    sunsynk library version: 0.6.4
05:23:37 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
05:23:37 INFO    Using three phase sensor definitions.
05:23:37 INFO    Using Single phase sensor definitions.
05:23:37 ERROR   Unknown sensor specified: battery_capacity_current
05:23:37 ERROR   Unknown sensor specified: grid_l1_power
05:23:37 ERROR   Unknown sensor specified: grid_l3_power
05:23:37 ERROR   Unknown sensor specified: load_l3_power
05:23:37 ERROR   Unknown sensor specified: inverter_l1_power
05:23:37 ERROR   Unknown sensor specified: inverter_l2_power
05:23:37 ERROR   Unknown sensor specified: inverter_l3_power
05:23:37 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
05:23:37 INFO    Connecting to tcp://192.168.1.154:8899
05:23:37 INFO    Reading startup sensors Serial, Prog4 Time, Prog1 Time, Prog6 Time, Prog3 Time, Prog5 Time, Rated power, Battery Low Capacity, Prog2 Time
05:23:37 INFO    ############################################################
05:23:37 INFO                 Inverter serial number '****0392'
05:23:37 INFO    ############################################################
05:23:37 INFO    MQTT: Connecting to hass@192.168.1.35:1883
05:23:37 INFO    MQTT: Connection successful
05:23:38 INFO    Read every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  5  | prog2_capacity, prog3_charge, prog2_power, prog4_power, prog4_time, prog3_power, |
|     |  battery_power, battery_max_charge_current, prog5_time, load_power, pv1_power,   |
|     |  grid_ct_power, priority_load, battery_max_discharge_current, essential_power,   |
|     |       pv2_power, prog6_power, grid_l2_power, prog1_capacity, prog5_power,        |
|     |       prog1_charge, prog1_time, prog1_power, load_l2_power, solar_export,        |
|     |  prog5_capacity, grid_charge_battery_current, battery_low_capacity, prog2_time,  |
|     |      use_timer, non_essential_power, load_limit, grid_power, prog6_charge,       |
|     |   prog3_capacity, prog6_capacity, prog4_capacity, load_l1_power, prog2_charge,   |
|     |         aux_power, prog3_time, prog6_time, prog5_charge, inverter_power,         |
|     |                        grid_charge_enabled, prog4_charge                         |
|  15 |    pv2_voltage, overall_state, inverter_frequency, battery_soc, grid_current,    |
|     |          pv2_current, inverter_voltage, inverter_current, pv1_current,           |
|     |    dc_transformer_temperature, pv1_voltage, grid_connected, battery_current,     |
|     |       grid_voltage, battery_voltage, radiator_temperature, load_frequency,       |
|     |                       grid_frequency, battery_temperature                        |
|  60 |                                    date_time                                     |
| 300 |    total_grid_import, total_battery_charge, day_grid_export, day_load_energy,    |
|     |        total_grid_export, total_battery_discharge, day_battery_discharge,        |
|     |       day_grid_import, day_pv_energy, day_battery_charge, total_pv_energy        |
+-----+----------------------------------------------------------------------------------+
05:23:38 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  60 |   essential_power, pv2_power, non_essential_power, grid_l2_power, grid_power,    |
|     |    aux_power, battery_power, pv1_power, load_l2_power, date_time, load_power,    |
[... 8 lines cut ...]
|     |   battery_low_capacity, prog2_time, use_timer, total_grid_import, pv1_current,   |
|     |        load_limit, dc_transformer_temperature, pv1_voltage, prog6_charge,        |
|     |        total_grid_export, prog3_capacity, grid_connected, prog6_capacity,        |
[... cut ...]
05:23:40 WARNING Cannot publish Priority Load: value is None
05:23:40 WARNING Cannot publish Solar Export: value is None
05:23:40 WARNING load_limit: Unknown register value 65535
05:23:40 WARNING grid_charge_enabled: Unknown register value 65535
05:23:40 WARNING priority_load: Unknown register value 65535
05:23:40 WARNING solar_export: Unknown register value 65535
05:23:45 WARNING load_limit: Unknown register value 65535
05:23:45 WARNING grid_charge_enabled: Unknown register value 65535
05:23:45 WARNING priority_load: Unknown register value 65535
`

PR #264 (2024-04-08): Fix Use Timer for 3PH inverters
Probably fixes #263