kellerza/sunsynk #265: CRC validation failed

Excalibur83, 2024-03-27:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On using solarman parameter (Sofar-solar inverter HYD-5000-EP with WifiStick)

### Describe issue
IP static address, port 8899 (scanned and opened, ping from VM running HA returns ok)
After many attempts, changing "scheduled", "gap", "sensors"... console returns every time a CRC fault :
"

08:11:54 INFO    Connecting to tcp://192.168.0.220:8899
08:11:54 INFO    Reading startup sensors Rated power, Serial
08:11:54 ERROR   Error retry attempt(0): CRC validation failed.
08:11:56 ERROR   Error retry attempt(1): CRC validation failed.
08:11:58 ERROR   Error retry attempt(2): CRC validation failed.
08:12:00 ERROR   Error retry attempt(3): CRC validation failed.
08:12:02 ERROR   Error retry attempt(4): CRC validation failed.
(...)
08:12:15 ERROR   OSError reading 5 registers from 3: Failed to read register 3; timeout reading 2 registers from 16
"
.... and again and again till it restarts once or more

### Environment

- Home Assistant version: 2024.3.3
- Addon:
  - Name: sunsynk-multi
  - Version: 202
- Inverter:
  - Make: Sofar Solar
  - Model: HYD-5000-EP
  - Firmware: v40022-00-021
- Adaptor details:
  - WifiStick LSW3
  - Firmware: LSW3_15_270A_1.68

  

### You configuration

```yaml

DRIVER: solarman
INVERTERS:
  - SERIAL_NR: SMxxxxxxxx3
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "123456789"
    PORT: tcp://192.168.0.220:8899
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - battery_soc
  - Inverter power
  - Load power
  - PV1 power
  - PV1 voltage
  - PV2 power
  - PV2 voltage
  - Day Active Energy
  - Fault
  - Overall state
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sofar
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 5
SCHEDULES:
  - KEY: W
    READ_EVERY: 120
    REPORT_EVERY: 120
    CHANGE_ANY: false
    CHANGE_BY: 50
    CHANGE_PERCENT: false
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt-user
MQTT_PASSWORD: Mypassword
DEBUG_DEVICE: /dev/ttyS0
DEBUG: 0

```

### Logs
08:26:04 WARNING Retrying individual sensors: ['serial', 'rated_power']
08:26:15 ERROR   timeout reading 5 registers from 3 Serial
08:26:19 ERROR   Error retry attempt(0): CRC validation failed.
08:26:21 ERROR   Error retry attempt(1): CRC validation failed.
08:26:23 ERROR   Error retry attempt(2): CRC validation failed.
08:26:25 ERROR   Error retry attempt(3): CRC validation failed.
08:26:26 ERROR   timeout reading 2 registers from 16 Rated power
08:26:26 CRITICAL Could not read sensors: ['Serial', 'Rated power']
08:26:26 INFO    ############################################################
08:26:26 INFO    No response on the Modbus interface tcp://192.168.0.220:8899, see https://kellerza.github.io/sunsynk/guide/fault-finding
08:26:26 INFO    ############################################################
08:26:26 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
08:26:56 ERROR   Task was destroyed but it is pending!
task: <Task pending name='ConnKeeper' coro=<PySolarmanV5Async._conn_keeper() running at /usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5_async.py:139> wait_for=<Future pending cb=[Task.task_wakeup()]>>
[08:26:56] INFO: Service restart after closing
08:26:57 INFO    Loading configuration: /data/options.json
08:26:57 INFO    sunsynk library version: 0.6.4
08:26:57 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|     w     |  *  | 120  |  120   |     50    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
08:26:57 INFO    Using Single phase sensor definitions.
08:26:57 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
08:26:57 INFO    Connecting to tcp://192.168.0.220:8899
08:26:57 INFO    Reading startup sensors Serial, Rated power
08:26:58 ERROR   Error retry attempt(0): CRC validation failed.
08:27:00 ERROR   Error retry attempt(1): CRC validation failed.
08:27:02 ERROR   Error retry attempt(2): CRC validation failed.
08:27:04 ERROR   Error retry attempt(3): CRC validation failed.
08:27:06 ERROR   Error retry attempt(4): CRC validation failed.
08:27:08 ERROR   Error retry attempt(0): CRC validation failed.
08:27:10 ERROR   Error retry attempt(1): CRC validation failed.
08:27:12 ERROR   Error retry attempt(2): CRC validation failed.
08:27:14 ERROR   Error retry attempt(3): CRC validation failed.
08:27:16 ERROR   Error retry attempt(4): CRC validation failed.
08:27:18 ERROR   OSError reading 5 registers from 3: Failed to read register 3; OSError reading 2 registers from 16: Failed to read register 16

```

kellerza, 2024-03-27:
Not sure this Sofar is supported via the pysolarmanv5 library if I understand this workaround correctly - https://github.com/jmccrohan/pysolarmanv5/issues/18

Excalibur83, 2024-03-27:
Hi, thanks for answered.
That's the point. Reading some information, many issues come with Sofar Inverter. It doesn't even works with solarman add-on...
Concerning CRC, and after reading parts of your code, you call data sensor by number from 1 to XXX. Bus address iaw sofar's datasheet , are something like 0x02%%. Do you think this is the reason ?

kellerza, 2024-03-29:
0x02 seems like 2 in hex. No idea what they meant with the %%

kellerza, 2024-03-29:
Their workaround was to use a cabled solution on the RS232 port (RS485 will likely also work)

So doubt its the registers that is the problem

Excalibur83, 2024-03-29:
Hi,
% came from me. Meanings number of your choice.
Addresses start from 0x200 to 0x245 in hex (see attached)
I will try with RS485, but there some issues also.
Whatever be proud of your add-on, Solarman's one doesn't work neither.
You can close that topic if you want. Thanks.
[SOFARSOLAR ModBus-RTU Communication Protocol.pdf](https://github.com/kellerza/sunsynk/files/14800221/SOFARSOLAR.ModBus-RTU.Communication.Protocol.pdf)

kellerza, 2024-06-18:
@Excalibur83 did you get RS485 to work?

Excalibur83, 2024-06-19:
Hi, not yet, lack of time.

closed 2024-09-06 by kellerza