kellerza/sunsynk #268: Version b381a73 WARNING use_timer: Unknown register value 255 The Problem

JanKraslice, 2024-04-09:
Problem recurring every 15 seconds after update
09:59:01 WARNING Cannot publish Use Timer: value is None
09:59:15 WARNING use_timer: Unknown register value 255

Inverter DEYE 3 Phase High Voltage 
Deye 30K 

No problem before update

Is a rollback to previous version or quick fix possible? Please advise
[image]

JanKraslice, 2024-04-09:
After switching off and on again with the command 
use_timer OFF
use_timer ON 
The error in the log has stopped showing up, probably will want to treat for future releases
[image]

rixxxx, 2024-04-09:
Its most probably due to the bug I fixed with #264 . HA had some now non-existing wrong value cached.

kellerza, 2024-06-03:
@JanKraslice can you please check if this is still present in the latest dev version?

JanKraslice, 2024-06-05:
hello latest dev version tested and everything is fine.

Thank you very much, you are the best

JanKraslice, 2024-06-05:
hello version latest dev s (fix @rixxxx submitted) tested and all is well.
The value is written correctly 1:1
Tested with three phase HV inverter version

Thank you very much, you are the best

kellerza, 2024-06-05:
Great!
Thx @rixxxx

closed 2024-06-05 by kellerza