kellerza/sunsynk #271: Addon reading incorrect data after connection is lost

jacekowski, 2024-04-14:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

After addon has been running for a while, i get an error in the logs and garbage data starts appearing in HA.

### Your environment

My current configuration uses https://www.waveshare.com/wiki/RS232/485/422_TO_POE_ETH_(B)  in TCP server mode (UDP mode in it is useless). 

It works fine for hours/days and then it doesn't, it looks like it is cause by buffers not being properly flushed 

### Logs

```logs

23:15:32 INFO    Queue update grid_trickle_feed=-100
23:15:32 INFO    Writing sensor grid_trickle_feed=-100 Registers:{206=0xff9c}
23:16:32 INFO    Queue update grid_trickle_feed=-100
23:16:32 INFO    Writing sensor grid_trickle_feed=-100 Registers:{206=0xff9c}
23:17:32 INFO    Queue update grid_trickle_feed=-100
23:17:32 INFO    Writing sensor grid_trickle_feed=-100 Registers:{206=0xff9c}
23:17:51 ERROR   OSError in read_sunsynk: ConnectionException reading 2 registers from 13: Modbus Error: [Connection] Connection lost during request poll_need_to_read
23:18:00 WARNING Did not complete read, only read 2/4
23:18:00 WARNING Did not complete read, only read 4/1
23:18:00 WARNING Did not complete read, only read 1/15
23:18:01 WARNING meter_enable: Unknown register value 10
23:18:01 WARNING Cannot publish Meter Enable: value is None
23:18:08 WARNING Did not complete read, only read 6/1
23:18:08 WARNING Did not complete read, only read 1/4
23:18:10 WARNING Did not complete read, only read 15/1
23:18:14 WARNING Did not complete read, only read 1/2
23:18:14 WARNING Did not complete read, only read 2/6
23:18:16 WARNING Did not complete read, only read 2/1
23:18:16 WARNING Did not complete read, only read 4/6
23:18:16 WARNING Did not complete read, only read 6/4
23:18:28 WARNING Did not complete read, only read 4/1
23:18:30 WARNING Did not complete read, only read 2/1
23:18:30 WARNING Did not complete read, only read 1/15
23:18:31 WARNING meter_enable: Unknown register value 10
23:18:31 WARNING Cannot publish Meter Enable: value is None

```

MacieGx, 2024-04-23:
@jacekowski Could you please send me your configuration in VirCOM? I have similar device RS485 TO POE ETH (B) (Din mounted). But no luck with configurations. Addon can't get any data 😢

maslyankov, 2024-05-13:
> @jacekowski Could you please send me your configuration in VirCOM? I have similar device RS485 TO POE ETH (B) (Din mounted). But no luck with configurations. Addon can't get any data 😢
> 
> nvm fixed. I connected wire to the meter port instead of BMS

Can you upload a screenshot of your RS485 Gateway web config? Are you using Modbus TCP to RTU?

jacekowski, 2024-05-13:
[image]

kellerza, 2024-05-16:
What protocol options do you have under the multi-host options?

jacekowski, 2024-05-18:
Modbus TCP to RTU and None

kellerza, 2024-06-18:
An option is to use "serial-tcp://" instead of "tcp://". It will allow you to send RTU-over-TCP from the addon. Then you likely need the protocol set to None.

kellerza, 2026-08-15:
> > An option is to use "serial-tcp://" instead of "tcp://". It will allow you to send RTU-over-TCP from the addon. Then you likely need the protocol set to None.
> 
> I have just been through this exercise with a USR-W610. Going to transparent mode and serial-tcp:// made a huge difference to stability, over wifi. I still get "Did not complete read, only read 6/4" on occasion, but it recovers more gracefully, and I can use a larger batch size. Is there any reason this is not the default recommendation in the documentation?

Feel free to raise a PR with this recommendation

closed 2026-08-19 by kellerza