kellerza/sunsynk #276: Issue with the SOC figure since HA upgraded

championc, 2024-04-22:
I have been running this plugin for about 2 months without issue.  I have an Automation which reads the SOC% and potentially triggers forced export of power.

But the other day I upgraded Home Assistant from 2023 to the latest 2024 image, and since then, my automation has not worked until I rebooted my PC.

So I had more time to look into things this evening.  I ran a trace against the Automation, and got the following

_if/condition/0/entity_id/0
Executed: April 22, 2024 at 11:12:10 PM
Result:

result: false
message: value 'unavailable' is non-numeric and treated as False_

I restarted the plugin and it correctly read the SOC%

_if/condition/0

Executed: April 22, 2024 at 11:18:10 PM
Result:

result: true

if/condition/0/entity_id/0
Executed: April 22, 2024 at 11:18:10 PM
Result:

result: true
state: 70_

I think the value was arriving as an alphanumeric because I use it in Node-Red on a dashboard, and it looked OK.

So can I do anything to ensure the value type does not change, or see what a restart of the docker container fixes the issue.

All suggestions greatly received, as I cannot be doing restarts day after day

championc, 2024-04-25:
I have had to implement an automatic restart script daily to restart the docker container

kellerza, 2024-04-25:
Anything in the mqtt logs re mqtt discovery?

championc, 2024-04-27:
Thanks for replying.  I forgot until it was too late to check over the past two days, but checked just now, and the logs shows the following, continuously

18:41:10 INFO    Queue update load_limit=Zero Export
18:41:10 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
18:42:10 INFO    Queue update load_limit=Zero Export
18:42:11 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
18:43:10 INFO    Queue update load_limit=Zero Export
18:43:10 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
18:44:10 INFO    Queue update load_limit=Zero Export
18:44:10 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
18:45:10 INFO    Queue update load_limit=Zero Export
18:45:10 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 

Hope this helps

HannesVenter, 2024-05-01:
Hi All, 

I had the same issue after upgrading the Update Sunsynk/Deye Inverter Add-on (edge/dev) on 27 April 2024, reloaded backup and now SOC is back to normal. I am new to HA. I am glad to help with logs ect if you show me how to get them.

kellerza, 2024-05-01:
@HannesVenter are you using the non-edge version, or just an older one? You can have a look at the sensor state in the HA Dev Tools and for further debugging see what is being published to MQTT unsing mqtt Explorer (free & very useful windows MQTT client)

[image]

Using MQTT explorer you can see it's constantly updated (every 5 min)

[image]

And if you restart HA somewhere in this 5min period you get the correct state through HA's restore_state mechanism.
Once you get the next (non-persistent) MQTT message on this topic it updates again

The HA discovery topic (also from in MQTT explorer) - this creates the sensor that you see in the HA Dev Tools

```
{
  "unique_id": "*49_battery_soc",
  "device": {
    "identifiers": [
      "*49"
    ],
    "manufacturer": "Sunsynk",
    "model": "5kW Inverter (**4149)",
    "name": "SS"
  },
  "state_topic": "SUNSYNK/status/xx49/battery_soc",
  "name": "Battery SOC",
  "availability": [
    {
      "topic": "SUNSYNK/status/xx49/availability"
    }
  ],
  "device_class": "battery",
  "unit_of_measurement": "%",
  "state_class": "measurement",
  "object_id": "ss_battery_soc",
  "suggested_display_precision": 1
}
```

HannesVenter, 2024-05-01:
@kellerza I am using the edge version.
[image]

After the update there is no sensor for Battery SOC available. 

Under Configuration Option in the Sunsynk/Deye Inverter Add-on(edge/dev) yaml I had SOC Battery under sensors: 

SENSORS:
  - PV1_power
  - PV2_power
  - PV3_power
  - PV4_power
  - PV_power
  - SOC Battery
  - Grid_Peak_Shaving_power
  - energy_management
  - power_flow_card
  - pv2_power

In MQTT I have the following Sensors:

[image]
[image]

Not sure if this helps you?

HannesVenter, 2024-05-01:
Maybe some more info. 

I have 4 50KW HV Deye inverters connected using the Solarman dongle on MQTT with the Sunsynk/Deye Inverter Add-on(edge/dev) Add-on. They are in parallel and all 4 do not have the Battery SOC sensor after the update.

Then on the same network and same HA, I have another 50KW HV Sunsynk inverter connected with a ESP32 from Heinz Meulke. 

Could the sensor names be conflicting?

kellerza, 2024-05-01:
Do you have “soc battery” or “battery soc”?

please post the addon log just after a startup

HannesVenter, 2024-05-01:
I have SOC Battery.

Sunsynk/Deye Inverter Add-on (edge/dev)
|  15 | pv1_current, pv_power, grid_ct_power, grid_connected, inverter_power, pv1_power, |
|     |  grid_frequency, grid_power, pv1_voltage, pv3_power, overall_state, pv2_power,   |
|     |                                    pv4_power                                     |
| 300 |        total_battery_discharge, total_battery_charge, total_grid_import,         |
|     |     day_battery_charge, day_load_energy, total_grid_export, day_grid_import,     |
|     |      day_battery_discharge, day_grid_export, total_pv_energy, day_pv_energy      |
+-----+----------------------------------------------------------------------------------+
12:21:55 INFO    Report every (inverter >1)
+-----+-----------------------------------------------------------------------------+
|  s  |                                   Sensors                                   |
+-----+-----------------------------------------------------------------------------+
|  60 |  pv_power, grid_ct_power, inverter_power, pv1_power, grid_power, pv3_power, |
|     |                             pv2_power, pv4_power                            |
| 300 | day_grid_export, total_pv_energy, day_pv_energy, grid_frequency, use_timer, |
|     |  battery_low_capacity, priority_load, day_battery_charge, day_load_energy,  |
|     |  pv1_voltage, day_grid_import, day_battery_discharge, total_battery_charge, |
|     |     grid_connected, overall_state, total_battery_discharge, pv1_current,    |
|     |        total_grid_import, total_grid_export, grid_peak_shaving_power        |
+-----+-----------------------------------------------------------------------------+
12:21:55 INFO    Connecting to tcp://192.168.1.34:8899
12:21:55 INFO    Reading startup sensors Battery Low Capacity, Serial, Prog2 Time, Rated power, Prog3 Time, Prog4 Time, Prog1 Time, Prog6 Time, Prog5 Time
12:21:55 INFO    ############################################################
12:21:55 INFO                 Inverter serial number '****8440'
12:21:55 INFO    ############################################################
12:21:56 INFO    Connecting to tcp://192.168.1.35:8899
12:21:56 INFO    Reading startup sensors Battery Low Capacity, Serial, Prog2 Time, Rated power, Prog3 Time, Prog4 Time, Prog1 Time, Prog6 Time, Prog5 Time
12:21:56 INFO    ############################################################
12:21:56 INFO                 Inverter serial number '****0318'
12:21:56 INFO    ############################################################
12:21:58 WARNING use_timer: Unknown register value 255
12:21:58 WARNING Cannot publish Use Timer: value is None
12:21:58 WARNING use_timer: Unknown register value 255
12:21:58 WARNING Cannot publish Use Timer: value is None
12:21:58 WARNING use_timer: Unknown register value 255
12:21:58 WARNING Cannot publish Use Timer: value is None
12:21:58 WARNING use_timer: Unknown register value 255
12:21:58 WARNING Cannot publish Use Timer: value is None
12:22:00 WARNING use_timer: Unknown register value 255
12:22:00 WARNING use_timer: Unknown register value 255
12:22:00 WARNING use_timer: Unknown register value 255
12:22:01 WARNING use_timer: Unknown register value 255
12:22:05 WARNING use_timer: Unknown register value 255
12:22:05 WARNING use_timer: Unknown register value 255
12:22:05 WARNING use_timer: Unknown register value 255
12:22:05 WARNING use_timer: Unknown register value 255
12:22:10 WARNING use_timer: Unknown register value 255
12:22:10 WARNING use_timer: Unknown register value 255
12:22:10 WARNING use_timer: Unknown register value 255
12:22:10 WARNING use_timer: Unknown register value 255
12:22:15 WARNING use_timer: Unknown register value 255
12:22:15 WARNING use_timer: Unknown register value 255
12:22:15 WARNING use_timer: Unknown register value 255
12:22:15 WARNING use_timer: Unknown register value 255
12:22:20 WARNING use_timer: Unknown register value 255
12:22:20 WARNING use_timer: Unknown register value 255
12:22:20 WARNING use_timer: Unknown register value 255
12:22:20 WARNING use_timer: Unknown register value 255
12:22:25 WARNING use_timer: Unknown register value 255
12:22:25 WARNING use_timer: Unknown register value 255
12:22:25 WARNING use_timer: Unknown register value 255
12:22:25 WARNING use_timer: Unknown register value 255
12:22:30 WARNING use_timer: Unknown register value 255
12:22:30 WARNING use_timer: Unknown register value 255
12:22:30 WARNING use_timer: Unknown register value 255
12:22:31 WARNING use_timer: Unknown register value 255
12:22:35 WARNING use_timer: Unknown register value 255
12:22:35 WARNING use_timer: Unknown register value 255
[... 12 lines cut ...]
12:22:50 WARNING use_timer: Unknown register value 255
12:22:50 WARNING use_timer: Unknown register value 255
12:22:55 WARNING use_timer: Unknown register value 255
12:22:55 WARNING use_timer: Unknown register value 255
12:22:55 WARNING use_timer: Unknown register value 255
12:22:55 WARNING use_timer: Unknown register value 255
12:23:00 WARNING use_timer: Unknown register value 255
12:23:00 WARNING use_timer: Unknown register value 255
12:23:00 WARNING use_timer: Unknown register value 255
12:23:00 WARNING use_timer: Unknown register value 255
12:23:05 WARNING use_timer: Unknown register value 255
12:23:05 WARNING use_timer: Unknown register value 255
12:23:05 WARNING use_timer: Unknown register value 255
12:23:05 WARNING use_timer: Unknown register value 255
12:23:10 WARNING use_timer: Unknown register value 255
[... cut ...] e 255
12:23:10 WARNING use_timer: Unknown register value 255
12:23:10 WARNING use_timer: Unknown register value 255
12:23:15 WARNING use_timer: Unknown register value 255
12:23:15 WARNING use_timer: Unknown register value 255
12:23:15 WARNING use_timer: Unknown register value 255
12:23:15 WARNING use_timer: Unknown register value 255
12:23:20 WARNING use_timer: Unknown register value 255
12:23:20 WARNING use_timer: Unknown register value 255
12:23:20 WARNING use_timer: Unknown register value 255

kellerza, 2024-05-01:
“Soc battery” is not in your sensors.
If you look at the start of the log there should be an error

try changing to “battery_soc” or “battery soc”

HannesVenter, 2024-05-01:
[12:30:49] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
12:30:53 INFO    Loading configuration: /data/options.json
12:30:53 INFO    sunsynk library version: 0.6.4
12:30:53 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  15  |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
12:30:53 INFO    Using three phase HV sensor definitions.
12:30:53 INFO    Importing /share/hass-addon-sunsynk/mysensors.py...
12:30:53 INFO      custom sensors: pv1_power, pv2_power, pv3_power, pv4_power, pv_power
12:30:53 ERROR   Unknown sensor specified: battery_soc
12:30:53 ERROR   Unknown sensor specified: aux_power
12:30:53 ERROR   Unknown sensor specified: battery_current
12:30:53 ERROR   Unknown sensor specified: battery_power
12:30:53 ERROR   Unknown sensor specified: battery_soc
12:30:53 ERROR   Unknown sensor specified: battery_voltage
12:30:53 ERROR   Unknown sensor specified: essential_power
12:30:53 ERROR   Unknown sensor specified: grid_voltage
12:30:53 ERROR   Unknown sensor specified: inverter_current
12:30:53 ERROR   Unknown sensor specified: load_frequency
12:30:53 ERROR   Unknown sensor specified: non_essential_power
12:30:53 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
12:30:53 INFO    Connecting to tcp://192.168.1.32:8899
12:30:53 INFO    Reading startup sensors Battery Low Capacity, Prog3 Time, Prog1 Time, Prog5 Time, Prog6 Time, Rated power, Prog4 Time, Serial, Prog2 Time

kellerza, 2024-05-01:
@rixxxx mad a change in the latest edge version that should get rid of that use_timer error

kellerza, 2024-05-01:
You have these issues

```
12:30:53 ERROR Unknown sensor specified: battery_soc
12:30:53 ERROR Unknown sensor specified: aux_power
12:30:53 ERROR Unknown sensor specified: battery_current
12:30:53 ERROR Unknown sensor specified: battery_power
12:30:53 ERROR Unknown sensor specified: battery_soc
12:30:53 ERROR Unknown sensor specified: battery_voltage
12:30:53 ERROR Unknown sensor specified: essential_power
12:30:53 ERROR Unknown sensor specified: grid_voltage
12:30:53 ERROR Unknown sensor specified: inverter_current
12:30:53 ERROR Unknown sensor specified: load_frequency
12:30:53 ERROR Unknown sensor specified: non_essential_power
```

@rixxxx might be able to help - he knows the HV sensors better

i suspect you need to update

HannesVenter, 2024-05-01:
This is with "battery soc"

|     |        prog5_capacity, prog1_time, prog4_power, prog5_time, prog4_charge,        |
|     |       prog3_capacity, prog4_capacity, prog6_power, load_limit, prog2_time,       |
|     |             prog1_capacity, prog6_capacity, prog3_power, prog2_power             |
|  15 | grid_connected, pv_power, pv4_power, grid_ct_power, grid_frequency, pv1_current, |
|     |  pv3_power, inverter_power, grid_power, overall_state, pv1_power, pv1_voltage,   |
|     |                                    pv2_power                                     |
| 300 |     total_grid_import, day_battery_charge, day_pv_energy, total_grid_export,     |
|     |  day_load_energy, day_grid_import, total_battery_charge, day_battery_discharge,  |
|     |            day_grid_export, total_battery_discharge, total_pv_energy             |
+-----+----------------------------------------------------------------------------------+
12:33:08 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  60 |    pv_power, pv4_power, grid_ct_power, pv3_power, inverter_power, grid_power,    |
|     |                               pv1_power, pv2_power                               |
| 300 |       total_grid_import, prog2_capacity, day_battery_charge, prog1_power,        |
|     |   overall_state, prog3_time, prog6_time, grid_peak_shaving_power, pv1_voltage,   |
|     | prog3_charge, prog5_power, total_pv_energy, battery_low_capacity, day_pv_energy, |
|     |         prog1_charge, total_grid_export, prog6_charge, day_load_energy,          |
|     |   total_battery_charge, priority_load, use_timer, prog4_time, prog6_capacity,    |
|     |  prog5_charge, prog5_capacity, day_battery_discharge, prog1_time, prog4_power,   |
|     |     day_grid_import, prog5_time, prog4_charge, grid_connected, pv1_current,      |
|     |     grid_frequency, prog3_capacity, prog4_capacity, prog6_power, load_limit,     |
|     |            prog2_time, prog1_capacity, prog3_power, day_grid_export,             |
|     |                total_battery_discharge, prog2_charge, prog2_power                |
+-----+----------------------------------------------------------------------------------+
12:33:08 INFO    Connecting to tcp://192.168.1.33:8899
12:33:08 INFO    Reading startup sensors Battery Low Capacity, Rated power, Prog3 Time, Serial, Prog6 Time, Prog2 Time, Prog1 Time, Prog4 Time, Prog5 Time
12:33:08 INFO    ############################################################
12:33:08 INFO                 Inverter serial number '****0379'
12:33:08 INFO    ############################################################
12:33:09 INFO    Read every (inverter >1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  5  |     battery_low_capacity, priority_load, use_timer, grid_peak_shaving_power      |
|  15 | grid_connected, pv_power, pv4_power, grid_ct_power, grid_frequency, pv1_current, |
|     |  pv3_power, inverter_power, grid_power, overall_state, pv1_power, pv1_voltage,   |
|     |                                    pv2_power                                     |
| 300 |     total_grid_import, day_battery_charge, day_pv_energy, total_grid_export,     |
|     |  day_load_energy, day_grid_import, total_battery_charge, day_battery_discharge,  |
|     |            day_grid_export, total_battery_discharge, total_pv_energy             |
+-----+----------------------------------------------------------------------------------+
12:33:09 INFO    Report every (inverter >1)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  60 |    pv_power, pv4_power, grid_ct_power, pv3_power, inverter_power, grid_power,   |
|     |                               pv1_power, pv2_power                              |
| 300 |  total_grid_import, day_battery_charge, overall_state, grid_peak_shaving_power, |
|     |        pv1_voltage, total_pv_energy, battery_low_capacity, day_pv_energy,       |
|     |     total_grid_export, day_load_energy, total_battery_charge, priority_load,    |
|     | use_timer, day_battery_discharge, day_grid_import, grid_connected, pv1_current, |
|     |             grid_frequency, day_grid_export, total_battery_discharge            |
+-----+---------------------------------------------------------------------------------+
12:33:09 INFO    Connecting to tcp://192.168.1.34:8899
12:33:09 INFO    Reading startup sensors Battery Low Capacity, Rated power, Prog3 Time, Serial, Prog6 Time, Prog2 Time, Prog1 Time, Prog4 Time, Prog5 Time
12:33:09 INFO    ############################################################
12:33:09 INFO                 Inverter serial number '****8440'
12:33:09 INFO    ############################################################
12:33:09 INFO    Connecting to tcp://192.168.1.35:8899
[... cut ...]
12:33:35 WARNING use_timer: Unknown register value 255
12:33:35 WARNING use_timer: Unknown register value 255
12:33:35 WARNING use_timer: Unknown register value 255
12:33:40 WARNING use_timer: Unknown register value 255
12:33:40 WARNING use_timer: Unknown register value 255
12:33:40 WARNING use_timer: Unknown register value 255
12:33:40 WARNING use_timer: Unknown register value 255
12:33:45 WARNING use_timer: Unknown register value 255
12:33:45 WARNING use_timer: Unknown register value 255
12:33:45 WARNING use_timer: Unknown register value 255

HannesVenter, 2024-05-01:
Current version: 0125f42

HannesVenter, 2024-05-01:
If I roll back to ab718d8  then this is the log:

Sunsynk/Deye Inverter Add-on (edge/dev)
|  5  |        prog2_power, prog1_capacity, use_timer, prog2_charge, prog4_time,        |
|     |       prog6_capacity, prog6_time, prog5_capacity, prog3_time, prog1_power,      |
|     |   prog2_capacity, prog4_charge, prog2_time, prog5_power, battery_low_capacity,  |
|     | prog3_capacity, prog4_power, prog5_time, load_limit, prog6_power, prog6_charge, |
|     |      prog3_charge, prog4_capacity, priority_load, prog1_time, prog3_power,      |
|     |               grid_peak_shaving_power, prog1_charge, prog5_charge               |
|  15 |  grid_ct_power, grid_frequency, pv4_power, pv3_power, pv2_power, battery_power, |
|     |     pv1_voltage, battery_current, inverter_power, grid_power, overall_state,    |
|     |  pv1_current, battery_soc, pv_power, grid_connected, pv1_power, battery_voltage |
| 300 |  day_grid_import, total_grid_export, day_battery_discharge, total_grid_import,  |
|     |       day_load_energy, total_pv_energy, day_battery_charge, day_pv_energy,      |
|     |          day_grid_export, total_battery_discharge, total_battery_charge         |
+-----+---------------------------------------------------------------------------------+
12:47:06 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  60 |  grid_ct_power, pv4_power, pv3_power, pv2_power, battery_power, inverter_power,  |
|     |                         grid_power, pv_power, pv1_power                          |
| 300 |       grid_frequency, day_battery_discharge, pv1_voltage, battery_current,       |
|     | battery_soc, prog2_power, total_battery_charge, battery_voltage, prog1_capacity, |
|     |     use_timer, prog2_charge, prog4_time, prog6_capacity, total_grid_import,      |
|     |      prog6_time, overall_state, day_pv_energy, prog5_capacity, prog1_power,      |
|     |   day_grid_import, prog3_time, prog2_capacity, prog4_charge, total_pv_energy,    |
|     |  prog2_time, prog5_power, grid_connected, prog3_capacity, battery_low_capacity,  |
|     |     prog4_power, total_grid_export, prog5_time, load_limit, day_load_energy,     |
|     |   prog6_power, prog6_charge, day_battery_charge, prog3_charge, prog4_capacity,   |
|     |       day_grid_export, priority_load, prog1_time, total_battery_discharge,       |
|     |  prog3_power, pv1_current, grid_peak_shaving_power, prog1_charge, prog5_charge   |
+-----+----------------------------------------------------------------------------------+
12:47:06 INFO    Connecting to tcp://192.168.1.33:8899
12:47:06 INFO    Reading startup sensors Serial, Prog5 Time, Rated power, Prog2 Time, Prog6 Time, Prog1 Time, Battery Low Capacity, Prog4 Time, Prog3 Time
12:47:08 INFO    ############################################################
12:47:08 INFO                 Inverter serial number '****0379'
12:47:08 INFO    ############################################################
12:47:09 INFO    Read every (inverter >1)
+-----+--------------------------------------------------------------------------------+
|  s  |                                    Sensors                                     |
+-----+--------------------------------------------------------------------------------+
|  5  |    battery_low_capacity, grid_peak_shaving_power, priority_load, use_timer     |
|  15 | grid_ct_power, grid_frequency, pv4_power, pv3_power, pv2_power, battery_power, |
|     |    pv1_voltage, battery_current, inverter_power, grid_power, overall_state,    |
|     | pv1_current, battery_soc, pv_power, grid_connected, pv1_power, battery_voltage |
| 300 | day_grid_import, total_grid_export, day_battery_discharge, total_grid_import,  |
|     |      day_load_energy, total_pv_energy, day_battery_charge, day_pv_energy,      |
|     |         day_grid_export, total_battery_discharge, total_battery_charge         |
+-----+--------------------------------------------------------------------------------+
12:47:09 INFO    Report every (inverter >1)
+-----+--------------------------------------------------------------------------------+
|  s  |                                    Sensors                                     |
+-----+--------------------------------------------------------------------------------+
|  60 | grid_ct_power, pv4_power, pv3_power, pv2_power, battery_power, inverter_power, |
|     |                        grid_power, pv_power, pv1_power                         |
| 300 |      grid_frequency, day_battery_discharge, pv1_voltage, battery_current,      |
|     |         battery_soc, total_battery_charge, battery_voltage, use_timer,         |
|     |       total_grid_import, overall_state, day_pv_energy, day_grid_import,        |
|     |   total_pv_energy, grid_connected, battery_low_capacity, total_grid_export,    |
|     |      day_load_energy, day_battery_charge, day_grid_export, priority_load,      |
|     |         total_battery_discharge, pv1_current, grid_peak_shaving_power          |
+-----+--------------------------------------------------------------------------------+
12:47:09 INFO    Connecting to tcp://192.168.1.34:8899
[... cut ...]
12:47:30 WARNING use_timer: Unknown register value 255
12:47:31 WARNING use_timer: Unknown register value 255
12:47:32 WARNING use_timer: Unknown register value 255
12:47:35 WARNING use_timer: Unknown register value 255
12:47:35 WARNING use_timer: Unknown register value 255
12:47:35 WARNING use_timer: Unknown register value 255
12:47:37 WARNING use_timer: Unknown register value 255
12:47:40 WARNING use_timer: Unknown register value 255
12:47:40 WARNING use_timer: Unknown register value 255
12:47:40 WARNING use_timer: Unknown register value 255

kellerza, 2024-05-01:
It seems to be battery_1_soc for the HV inverter - https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions3phhv.py#L30

kellerza, 2024-05-01:
The warning for deprecated/changed sensors might have been broken, I updated it and maybe you will start getting new warnings on the new version.

rixxxx, 2024-05-01:
HV intvertes with multiple battery ports have battery_X_soc/power/current/etc. I have unfinished patch to have different sensors for LV and HV in sensor groups. So eg power_flow_card would bring in battery_soc on LV and battery_1_soc on HV. PR will come soon.

heinrich321, 2024-06-02:
@kellerza / @rixxxx I havent had time to properly debug yet. But ive just hooked up 4 x 12kW Three Phase inverters on the edge version and am getting the same:

08:09:29 INFO    Report every (inverter >1)
+-----+----------------------------------------------------------------------------+
|  s  |                                  Sensors                                   |
+-----+----------------------------------------------------------------------------+
|  60 |    battery_power, grid_ct_power, pv1_power, grid_power, inverter_power     |
| 300 |    battery_current, total_battery_discharge, priority_load, use_timer,     |
|     | total_grid_import, day_battery_discharge, day_load_energy, grid_frequency, |
|     |   total_battery_charge, grid_connected, overall_state, day_grid_export,    |
|     |        pv1_current, pv1_voltage, day_battery_charge, day_pv_energy,        |
|     |   battery_low_capacity, battery_soc, battery_voltage, total_grid_export,   |
|     |                      day_grid_import, total_pv_energy                      |
+-----+----------------------------------------------------------------------------+
08:09:29 INFO    Connecting to /dev/ttyACM2
08:09:29 INFO    PyModbus 3.6.4 Serial: /dev/ttyACM2
08:09:29 INFO    Reading startup sensors Prog5 Time, Prog6 Time, Prog1 Time, Rated power, Battery Low Capacity, Prog4 Time, Prog3 Time, Prog2 Time, Serial
08:09:32 INFO    ############################################################
08:09:32 INFO                 Inverter serial number '****3410'
08:09:32 INFO    ############################################################
08:09:34 WARNING use_timer: Unknown register value 255
08:09:34 WARNING use_timer: Unknown register value 255
08:09:34 WARNING Cannot publish Use Timer: value is None
08:09:34 WARNING Cannot publish Use Timer: value is None
08:09:34 WARNING use_timer: Unknown register value 255
08:09:34 WARNING Cannot publish Use Timer: value is None
08:09:35 WARNING use_timer: Unknown register value 255
08:09:38 WARNING use_timer: Unknown register value 255
08:07:20 WARNING use_timer: Unknown register value 255
08:07:25 WARNING use_timer: Unknown register value 255
08:07:30 WARNING use_timer: Unknown register value 255
08:07:35 WARNING use_timer: Unknown register value 255
08:07:40 WARNING use_timer: Unknown register value 255
08:07:45 WARNING use_timer: Unknown register value 255
08:07:53 WARNING use_timer: Unknown register value 255
08:07:58 WARNING use_timer: Unknown register value 255
08:08:02 WARNING use_timer: Unknown register value 255
08:08:02 ERROR   OSError in read_ss2: timeout reading 5 registers from 587; timeout reading 1 registers from 609; timeout reading 1 registers from 619; timeout reading 1 registers from 625; timeout reading 1 registers from 636; timeout reading 1 registers from 672; timeout reading 2 registers from 676 poll_need_to_read
08:08:03 WARNING use_timer: Unknown register value 255
08:08:05 WARNING use_timer: Unknown register value 255
08:08:10 WARNING use_timer: Unknown register value 255
08:08:15 WARNING use_timer: Unknown register value 255
08:08:20 WARNING use_timer: Unknown register value 255
08:08:25 WARNING use_timer: Unknown register value 255
08:08:30 WARNING use_timer: Unknown register value 255
08:08:38 WARNING use_timer: Unknown register value 255
08:08:39 WARNING Did not complete read, only read 5/1

HannesVenter, 2024-06-03:
Try adding  
- battery_1_soc
- battery_2_soc
under SENSORS  in your config file.

closed 2024-06-18 by kellerza