kellerza/sunsynk #277: grid_connected always OFF

marekorok, 2024-04-23:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py)

### You Inverter
Deye SUN-10K-SG04LP3-EU

### Describe the issue/bug and what you expect
since the begging after setup the grid_connected always reports OFF, even I switch it manually in the mqtt it immediately turns off.
[image]

Other sensors looks fine, no idea how to fix it.

Using docker edge version:

```
# Deye/sunsynk
# ------------------------------------------------
  deye:
    container_name: deye
    restart: unless-stopped
    image: ghcr.io/kellerza/hass-addon-sunsynk-multi/armv7:edge
    volumes:
      - ./mounts/deye/options.yaml:/data/options.yaml
```

DRIVER: "solarman"

grid_connected included in config:
[image]

No specific logs

rixxxx, 2024-04-25:
#274 addresses that

kellerza, 2024-04-26:
@marekorok just merged 274, let us know if this is ok now

marekorok, 2024-04-26:
yeap, working immediately:
[image]

case closed, thank you!

closed 2024-04-26 by marekorok