kellerza/sunsynk #280: RWSensor System Zero Export power MinMax Value

pr3mi3r3, 2024-05-05:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions3ph.py)

### Deye SUN-12K-SG04LP3-EU
NumberRWSensor(104, "System Zero Export power", WATT, -1)

Details about your Inverter: Deye SUN-12K-SG04LP3-EU, Firmware 1135

### Describe the issue/bug and what you expect

Please adjust the max value for this RWSensor (104).
With Firmware 1135 you can have values below 0 Watt.
By writing the Value 65485 via modbus, you will have -50 Watt.
By writing the Value 65535 via modbus, you wil have 0 Watt.

rixxxx, 2024-05-07:
What it actually does if you set it to negative value?

pr3mi3r3, 2024-05-08:
It will sell the energy more likely, if possible.
If you put the value 0 Watt into the register, it will sometimes buying power from the grid.
Negative values will avoid buying power if not necessary.

rixxxx, 2024-05-08:
Its not hard to allow negative values, but I'm a bit reluctant. Later releases only allow it down to 0 and latest allow it only down to +20(iirc). We have already had reports where users try to set it to 0 but the inverter resets it back to 20. The range is not documented and there is no way to know what the inverter actually accepts. Whats the range LCD UI allows on 1135?

rixxxx, 2024-05-08:
I have only tested this on HV inverter and there it actually works. Having it on 0 makes small 3-4W inflow. Setting it -100 caused ~100W outflow. @kellerza what do you think, should I make pull request and implement the non-documented feature which works only on some models or not?
```
diff --git a/src/sunsynk/definitions3ph.py b/src/sunsynk/definitions3ph.py
index ca9c0a5..9dd4843 100644
--- a/src/sunsynk/definitions3ph.py
+++ b/src/sunsynk/definitions3ph.py
@@ -201,7 +201,7 @@ SENSORS += (
     NumberRWSensor(
         103, "Battery low voltage", VOLT, -0.1
     ),  # interesting perhaps not available in menu. default 4500 (45V)
-    NumberRWSensor(104, "System Zero Export power", WATT, -1),
+    NumberRWSensor(104, "System Zero Export power", WATT, -1, min=-500, max=500),
     NumberRWSensor(105, "Battery Equalization Days", "days", -1),
     NumberRWSensor(106, "Battery Equalization Hours", "h", -1),  # 1 = 0.5 hours
     SwitchRWSensor(129, "Generator Charge enabled"),
diff --git a/src/sunsynk/definitions3phhv.py b/src/sunsynk/definitions3phhv.py
index e587f5a..1355ae5 100644
--- a/src/sunsynk/definitions3phhv.py
+++ b/src/sunsynk/definitions3phhv.py
@@ -116,7 +116,7 @@ SENSORS += (NumberRWSensor(191, "Grid Peak Shaving power", WATT, 10, max=100000)

 # Additional optional sensors
 SENSORS += (
-    NumberRWSensor(104, "System Zero Export power", WATT, -10),
+    NumberRWSensor(104, "System Zero Export power", WATT, -10, min=-500, max=500),
     NumberRWSensor(124, "Generator Charge Start Battery SOC", "%"),
     NumberRWSensor(125, "Generator Charge Battery current", AMPS),
     SwitchRWSensor(110, "Parallel Battery 1 and 2"),
```

pr3mi3r3, 2024-05-08:
> I have only tested this on HV inverter and there it actually works. Having it on 0 makes small 3-4W inflow. Setting it to -100 caused ~100W outflow. @kellerza what do you think, should I make pull request and implement the non-documented feature which works only on some model-firmware combinations or not?
> 
> ```
> diff --git a/src/sunsynk/definitions3ph.py b/src/sunsynk/definitions3ph.py
> index ca9c0a5..9dd4843 100644
> --- a/src/sunsynk/definitions3ph.py
> +++ b/src/sunsynk/definitions3ph.py
> @@ -201,7 +201,7 @@ SENSORS += (
>      NumberRWSensor(
>          103, "Battery low voltage", VOLT, -0.1
>      ),  # interesting perhaps not available in menu. default 4500 (45V)
> -    NumberRWSensor(104, "System Zero Export power", WATT, -1),
> +    NumberRWSensor(104, "System Zero Export power", WATT, -1, min=-500, max=500),
>      NumberRWSensor(105, "Battery Equalization Days", "days", -1),
>      NumberRWSensor(106, "Battery Equalization Hours", "h", -1),  # 1 = 0.5 hours
>      SwitchRWSensor(129, "Generator Charge enabled"),
> diff --git a/src/sunsynk/definitions3phhv.py b/src/sunsynk/definitions3phhv.py
> index e587f5a..1355ae5 100644
> --- a/src/sunsynk/definitions3phhv.py
> +++ b/src/sunsynk/definitions3phhv.py
> @@ -116,7 +116,7 @@ SENSORS += (NumberRWSensor(191, "Grid Peak Shaving power", WATT, 10, max=100000)
> 
>  # Additional optional sensors
>  SENSORS += (
> -    NumberRWSensor(104, "System Zero Export power", WATT, -10),
> +    NumberRWSensor(104, "System Zero Export power", WATT, -10, min=-500, max=500),
>      NumberRWSensor(124, "Generator Charge Start Battery SOC", "%"),
>      NumberRWSensor(125, "Generator Charge Battery current", AMPS),
>      SwitchRWSensor(110, "Parallel Battery 1 and 2"),
> ```

Thank you

jacekowski, 2024-05-18:
I've been using it with negative values for quite a while to make the inverter actually hit 0 at suppliers meter and i've not had any issues.

rixxxx, 2024-06-03:
#288

closed 2024-06-03 by kellerza