kellerza/sunsynk #286: Parameter settings for example ss_prog_power enters multiples of values instead of 1000 enters 10 000

JanKraslice, 2024-05-30:
Hello, the problem is in the value of the setting parameters!
Parameter settings for example ss_prog_power
any parameter 1 - 6 value is entered in multiples or kW but HA displays W! Or there is an error in the semicolon value.

If you enter 1000W into the inverter, 10000W is entered, see photo

Please correct
[image]
[image]

kellerza, 2024-05-30:
Hi, what inverter do you have?

You can try to use custom sensors and add a factor parameter

```python
SENSORS += (
    # The max power that can be used from the battery
    NumberRWSensor(256, "Prog1 power", WATT, max=RATED_POWER, factor=10),
    NumberRWSensor(257, "Prog2 power", WATT, max=RATED_POWER, factor=10),
```

JanKraslice, 2024-05-31:
I'll add
In the previous version it was fine. ( I change the mentioned value in automation)
Only yesterday I updated to version b092637 and shortly afterwards I found that the Prog power parameter is written to the inverter with 10 times the value.

My inverter is SUN-30K-SG01HP3-EU-BM3

I've been using your add-on for several months and until the update to version b092637 everything was fine, including the Prog power value.

Would you have to guide me how to create a custom sensor with parameters ideally a video how to with a tutorial I didn't understand.

Thank you

JanKraslice, 2024-05-31:
> Ahoj, jaký máš invertor?
> 
> Můžete zkusit použít vlastní senzory a přidat parametr faktoru
> 
> ```python
> SENSORS += (
>     # The max power that can be used from the battery
>     NumberRWSensor(256, "Prog1 power", WATT, max=RATED_POWER, factor=10),
>     NumberRWSensor(257, "Prog2 power", WATT, max=RATED_POWER, factor=10),
> ```

I'll add
In the previous version it was fine. ( I change the mentioned value in automation)
Only yesterday I updated to version b092637 and shortly afterwards I found that the Prog power parameter is written to the inverter with 10 times the value.

My inverter is SUN-30K-SG01HP3-EU-BM3

I've been using your add-on for several months and until the update to version b092637 everything was fine, including the Prog power value.

Would you have to guide me how to create a custom sensor with parameters ideally a video how to with a tutorial I didn't understand.

Thank you

JanKraslice, 2024-05-31:
I tracked down the release history.

Originally the definition of Phase 3 VN was
SENSORS += (
    NumberRWSensor(154, "Prog1 power", WATT, 10, max=RATED_POWER),
    NumberRWSensor(155, "Prog2 power", WATT, 10, max=RATED_POWER),
    NumberRWSensor(156, "Prog3 power", WATT, 10, max=RATED_POWER),
    NumberRWSensor(157, "Prog4 power", WATT, 10, max=RATED_POWER),
    NumberRWSensor(158, "Prog5 power", WATT, 10, max=RATED_POWER),
    NumberRWSensor(159, "Prog6 power", WATT, 10, max=RATED_POWER),
)

New from version b092637 you have
 NumberRWSensor(154, "Prog1 power", WATT, max=RATED_POWER),
    NumberRWSensor(155, "Prog2 power", WATT, max=RATED_POWER),
    NumberRWSensor(156, "Prog3 power", WATT, max=RATED_POWER),
    NumberRWSensor(157, "Prog4 power", WATT, max=RATED_POWER),
    NumberRWSensor(158, "Prog5 power", WATT, max=RATED_POWER),
    NumberRWSensor(159, "Prog6 power", WATT, max=RATED_POWER),

there's a missing , 10 

**Will you fix it for the next version?**

rixxxx, 2024-06-03:
#289

kellerza, 2024-06-03:
@JanKraslice please test with the fix @rixxxx submitted

closed 2024-06-03 by kellerza