kellerza/sunsynk #291: wrong values for battery voltage, PV production and battery actual power for Deye SUN-10K-SG01HP3-EU-AM2

bsafh, 2024-06-11:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py)

### You Inverter

 SUN-10K-SG01HP3-EU-AM2

accessed via RTU/Modbus adapter Waveshare  RS485 TO ETH 1
via serial-tcp mdbus driver in the Sunsyk-Multi-Addon

Sunsynk/Deye Inverter Add-on (multi)
Current version: 0.6.5 

### Describe the issue/bug and what you expect

Some of the values are totally off.

Battery voltage ist reported as around 29V, but the inverter is a high-voltage inverter which requires minimum 170V

PV and battery power is reported around a few hundred Watts, whereas the inverter display shows several kW

The documentation https://kellerza.github.io/sunsynk/reference/multi-options says there are three options: 

> The SENSOR_DEFINITION option allows you to select between `single-phase`, `three-phase` and `three-phase-hv` sensor definitions.

**`three-phase-hv` does NOT work, it is simply not selectable**

[image]

### Logs (if applicable)

```
- SERIAL_NR: "xxxxxxxxxx"
  HA_PREFIX: WR_Deye_1
  MODBUS_ID: 1
  DONGLE_SERIAL_NUMBER: "0"
  PORT: serial-tcp://192.168.80.44:8234
```

```
08:07:26 INFO    Connecting to serial-tcp://192.168.80.44:8234
08:07:26 INFO    PyModbus 3.5.2 serial-tcp: 192.168.80.44:8234
08:07:26 INFO    Reading startup sensors Prog6 Time, Prog1 Time, Battery Float voltage, Battery Shutdown Capacity, Battery Shutdown voltage, Battery Low Capacity, Rated power, Serial, Battery Low voltage, Prog5 Time, Prog2 Time, Battery Restart Capacity, Prog3 Time, Prog4 Time
08:07:26 INFO    ############################################################
08:07:26 INFO                 Inverter serial number '****xxxx'
08:07:26 INFO    ############################################################
08:07:27 INFO    Read every (inverter >1)
+----+----------------------------------------------------------------------------------+
| s  |                                     Sensors                                      |
+----+----------------------------------------------------------------------------------+
| 5  |  prog3_capacity, prog5_power, pv3_power, pv1_power, prog4_power, prog1_voltage,  |
|    |   prog3_voltage, solar_export, export_limit_power, battery_equalization_hours,   |
|    |   battery_restart_capacity, prog1_charge, generator_port_usage, gen_l2_power,    |
|    |     prog2_capacity, prog1_capacity, battery_max_charge_current, prog2_time,      |
|    |   battery_resistance, prog6_time, battery_absorption_voltage, prog6_capacity,    |
|    |   battery_power, inverter_power, prog5_voltage, load_limit, inverter_l2_power,   |
|    |     battery_max_discharge_current, gen_l3_power, prog3_power, prog5_charge,      |
|    |   grid_ct_power, grid_power, prog4_charge, prog4_capacity, inverter_l1_power,    |
|    |         prog6_charge, grid_l3_power, battery_low_voltage, load_l3_power,         |
|    |        generator_charge_enabled, prog3_time, load_l1_power, prog2_charge,        |
|    |     grid_charge_enabled, prog3_charge, gen_power, system_zero_export_power,      |
|    |      battery_shutdown_voltage, priority_load, battery_equalization_voltage,      |
|    |         prog6_power, load_l2_power, prog2_power, pv2_power, load_power,          |
|    |  battery_capacity_current, use_timer, grid_charge_battery_current, prog1_time,   |
|    |             gen_l1_power, grid_l2_power, prog1_power, prog4_voltage,             |
|    |    battery_restart_voltage, battery_float_voltage, battery_equalization_days,    |
|    |     inverter_l3_power, prog2_voltage, battery_shutdown_capacity, pv4_power,      |
|    | grid_l1_power, prog5_capacity, prog4_time, prog6_voltage, battery_low_capacity,  |
|    |                                    prog5_time                                    |
| 10 |   fault, grid_connected, pv4_current, inverter_voltage, radiator_temperature,    |
|    |        battery_current, battery_temperature, grid_voltage, grid_current,         |
|    |       battery_voltage, device_type, pv2_current, pv3_current, pv4_voltage,       |
|    |          inverter_frequency, grid_frequency, pv1_voltage, pv3_voltage,           |
|    | dc_transformer_temperature, overall_state, pv2_voltage, battery_soc, pv1_current |
| 60 |    day_load_energy, total_load_energy, day_battery_charge, day_active_energy,    |
|    |         total_battery_discharge, total_active_energy, total_grid_import,         |
|    |      day_grid_import, total_battery_charge, day_gen_energy, day_pv_energy,       |
|    |   total_pv_energy, total_grid_export, day_grid_export, day_battery_discharge,    |
|    |                                    date_time                                     |
+----+----------------------------------------------------------------------------------+
08:07:27 INFO    Report every (inverter >1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  30 |    pv3_power, load_l1_power, gen_l1_power, inverter_l2_power, grid_l2_power,     |
|     |      pv1_power, gen_l3_power, gen_power, inverter_l3_power, grid_ct_power,       |
|     |      gen_l2_power, grid_power, inverter_l1_power, pv4_power, grid_l1_power,      |
|     |       grid_l3_power, load_l2_power, load_l3_power, pv2_power, load_power,        |
|     |                          battery_power, inverter_power                           |
|  60 | use_timer, grid_charge_battery_current, prog3_capacity, prog1_time, prog5_power, |
|     |  load_limit, prog2_charge, grid_charge_enabled, battery_max_discharge_current,   |
|     |        prog3_charge, prog1_power, prog4_voltage, prog4_power, date_time,         |
|     |          battery_restart_voltage, battery_float_voltage, prog1_voltage,          |
|     | system_zero_export_power, battery_equalization_days, prog3_power, prog5_charge,  |
|     |    solar_export, prog3_voltage, export_limit_power, battery_shutdown_voltage,    |
|     |       battery_equalization_hours, battery_restart_capacity, prog1_charge,        |
|     |       generator_port_usage, priority_load, prog2_capacity, prog1_capacity,       |
|     |           prog4_charge, prog4_capacity, battery_equalization_voltage,            |
|     |   battery_shutdown_capacity, prog2_voltage, prog6_charge, battery_low_voltage,   |
|     | battery_max_charge_current, prog2_time, prog6_power, prog5_capacity, prog4_time, |
|     |    battery_resistance, prog6_time, prog6_voltage, battery_absorption_voltage,    |
|     |         battery_capacity_current, generator_charge_enabled, prog2_power,         |
|     |   battery_low_capacity, prog6_capacity, prog5_time, prog5_voltage, prog3_time    |
| 180 |  fault, total_battery_discharge, grid_connected, pv4_current, inverter_voltage,  |
|     |    day_pv_energy, radiator_temperature, day_battery_charge, battery_current,     |
|     |      day_grid_export, total_grid_export, grid_voltage, battery_temperature,      |
|     |    grid_current, battery_voltage, total_active_energy, total_battery_charge,     |
|     |        total_pv_energy, device_type, pv2_current, day_battery_discharge,         |
|     | day_load_energy, total_load_energy, pv3_current, pv4_voltage, total_grid_import, |
|     |     inverter_frequency, day_active_energy, grid_frequency, day_grid_import,      |
|     |      pv1_voltage, pv3_voltage, day_gen_energy, dc_transformer_temperature,       |
|     |               overall_state, pv2_voltage, battery_soc, pv1_current               |
+-----+----------------------------------------------------------------------------------+
```

[image]

[image]

[image]

[image]

[image]

kellerza, 2024-06-11:
Please try the dege/dev version to get access to the latest sensor definitions

bsafh, 2024-06-12:
works, but gives errors/warnings on startup

```
06:12:01 WARNING Cannot publish Use Timer: value is None
06:12:01 WARNING Cannot publish Microinverter export to grid cutoff: value is None
06:12:01 WARNING Cannot publish On Grid always on: value is None
06:12:01 WARNING Cannot publish US version grounding fault: value is None
06:12:01 WARNING Cannot publish Allow Remote: value is None
06:12:01 WARNING Cannot publish Use Timer: value is None
06:12:01 WARNING Cannot publish Microinverter export to grid cutoff: value is None
06:12:01 WARNING Cannot publish On Grid always on: value is None
06:12:01 WARNING Cannot publish US version grounding fault: value is None
06:12:01 WARNING Cannot publish Allow Remote: value is None
06:12:05 WARNING use_timer: Unknown register value 255
06:12:05 WARNING device_type: Unknown register value 6
06:12:05 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
06:12:05 WARNING on_grid_always_on: Unknown register value 0
06:12:05 WARNING us_version_grounding_fault: Unknown register value 0
06:12:05 WARNING allow_remote: Unknown register value 0
....

```

kellerza, 2024-06-18:
Closing as a duplicate

closed 2024-06-18 by kellerza