kellerza/sunsynk #292: Sunsynk/Deye Inverter Add-on (multi) Current version: 0.6.5 - model selection "three-phase-hv" not available

bsafh, 2024-06-11:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

Sunsynk/Deye Inverter Add-on (multi)
Current version: 0.6.5 Dec 9, 2023

### Describe the issue/bug

The documentation https://kellerza.github.io/sunsynk/reference/multi-options says there are three options:

> The SENSOR_DEFINITION option allows you to select between `single-phase`, `three-phase` and `three-phase-hv` sensor definitions.

**`three-phase-hv` does NOT work, it is simply not selectable** ... and throws "unknown" error when set in YAML mode

[image]

`three-phase` with a high voltage inverter results in totally wrong values for battery voltage, battery power and PV power.

**Expected behavior**
A clear and concise description of what you expected to happen.

### Your environment

- Home Assistant version: 2024.6.1
- Home Assistant Supervisor: 2024.06.0
- Addon:
  - Name: sunsynk-multi
  - Version: Sunsynk/Deye Inverter Add-on (multi) Current version: 0.6.5
- Inverter:
  - Make: Deye.
  - Model: SUN-10K-SG01HP3-EU-AM2
  - Firmware: ??
- Adaptor details:
  - RS485
  - RS485 brand: Waveshare
  - RS485 adapter model: "RS485 TO ETH 1", Firmware V4019

### You configuration

```

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2312122xxx"
    HA_PREFIX: WR_Deye_1
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.44:8234
  - SERIAL_NR: "2312126zzz"
    HA_PREFIX: WR_Deye_2
    MODBUS_ID: 2
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.45:8234
SENSOR_DEFINITIONS: three-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - settings
  - all
  - pv1_power
  - essential_power
  - battery_power
  - battery_soc
  - non_essential_power
  - grid_ct_power
  - Inverter power
  - Inverter L1 power
  - Inverter L2 power
[... 288 lines cut ...]
|     |  load_limit, prog2_charge, grid_charge_enabled, battery_max_discharge_current,   |
|     |        prog3_charge, prog1_power, prog4_voltage, prog4_power, date_time,         |
|     |          battery_restart_voltage, battery_float_voltage, prog1_voltage,          |
|     | system_zero_export_power, battery_equalization_days, prog3_power, prog5_charge,  |
|     |    solar_export, prog3_voltage, export_limit_power, battery_shutdown_voltage,    |
|     |       battery_equalization_hours, battery_restart_capacity, prog1_charge,        |
|     |       generator_port_usage, priority_load, prog2_capacity, prog1_capacity,       |
|     |           prog4_charge, prog4_capacity, battery_equalization_voltage,            |
|     |   battery_shutdown_capacity, prog2_voltage, prog6_charge, battery_low_voltage,   |
|     | battery_max_charge_current, prog2_time, prog6_power, prog5_capacity, prog4_time, |
|     |    battery_resistance, prog6_time, prog6_voltage, battery_absorption_voltage,    |
|     |         battery_capacity_current, generator_charge_enabled, prog2_power,         |
|     |   battery_low_capacity, prog6_capacity, prog5_time, prog5_voltage, prog3_time    |
| 180 |  fault, total_battery_discharge, grid_connected, pv4_current, inverter_voltage,  |
|     |    day_pv_energy, radiator_temperature, day_battery_charge, battery_current,     |
|     |      day_grid_export, total_grid_export, grid_voltage, battery_temperature,      |
|     |    grid_current, battery_voltage, total_active_energy, total_battery_charge,     |
|     |        total_pv_energy, device_type, pv2_current, day_battery_discharge,         |
|     | day_load_energy, total_load_energy, pv3_current, pv4_voltage, total_grid_import, |
|     |     inverter_frequency, day_active_energy, grid_frequency, day_grid_import,      |
|     |      pv1_voltage, pv3_voltage, day_gen_energy, dc_transformer_temperature,       |
|     |               overall_state, pv2_voltage, battery_soc, pv1_current               |
+-----+----------------------------------------------------------------------------------+

```

kellerza, 2024-06-11:
Can you try the edge/dev version of the addon?

bsafh, 2024-06-11:
> Can you try the edge/dev version of the addon?

Sure.

With dev/edge and `three-phase-hv` the l8g says:

```
12:07:18 INFO    MQTT: Removing HASS MQTT discovery info homeassistant/sensor/2312126286/grid_voltage/config
12:07:18 INFO    MQTT: Removing HASS MQTT discovery info homeassistant/sensor/2312126286/battery_temperature/config
12:07:18 INFO    MQTT: Removing HASS MQTT discovery info homeassistant/select/2312126286/generator_charge_enabled/config
12:07:18 INFO    Read every (inverter >1)
+----+----------------------------------------------------------------------------------+
| s  |                                     Sensors                                      |
+----+----------------------------------------------------------------------------------+
| 5  |     load_power, gen_l3_power, inverter_l1_power, battery_absorption_voltage,     |
|    | prog5_time, use_timer, prog4_time, prog6_power, prog3_power, inverter_l2_power,  |
|    |   load_limit, battery_low_voltage, pv4_power, prog2_capacity, prog5_capacity,    |
|    | prog5_charge, battery_low_capacity, gen_l1_power, prog1_voltage, priority_load,  |
|    |    prog4_charge, gen_l2_power, battery_max_discharge_current, prog6_voltage,     |
|    |   prog5_voltage, prog4_voltage, grid_l3_power, prog1_time, inverter_l3_power,    |
|    |        gen_power, prog1_capacity, prog2_time, prog6_capacity, prog6_time,        |
|    |       prog4_capacity, pv3_power, device_type, grid_l1_power, prog1_charge,       |
|    |       generator_charge_enabled, prog3_charge, battery_equalization_hours,        |
|    | grid_charge_battery_current, prog2_charge, prog1_power, pv1_power, prog2_power,  |
|    | load_l3_power, battery_capacity_current, grid_ct_power, prog5_power, pv2_power,  |
|    |  battery_float_voltage, system_zero_export_power, load_l1_power, solar_export,   |
|    |          grid_charge_start_battery_soc, inverter_power, grid_l2_power,           |
|    |      battery_shutdown_voltage, load_l2_power, battery_equalization_voltage,      |
|    |  battery_max_charge_current, prog3_voltage, grid_charge_enabled, prog6_charge,   |
|    |  prog3_capacity, grid_peak_shaving_power, battery_wake_up, battery_resistance,   |
|    | generator_port_usage, prog3_time, battery_equalization_days, export_limit_power, |
|    | grid_power, prog2_voltage, battery_restart_capacity, battery_shutdown_capacity,  |
|    |                       battery_restart_voltage, prog4_power                       |
| 10 |    pv1_current, pv1_voltage, undefined_load_relay_status, inverter_frequency,    |
|    |             fault, dry_contact1_status, grid_frequency, pv3_current,             |
|    |      dc_transformer_temperature, pv4_voltage, grid_l1_voltage, pv4_current,      |
|    | pv2_voltage, load_l2_voltage, grid_give_power_to_relay_status, load_l3_voltage,  |
|    |       pv3_voltage, inverter_voltage, dry_contact2_status, grid_l3_voltage,       |
|    |       load_l1_voltage, grid_connected, inv_relay_status, grid_l2_voltage,        |
|    |      overall_state, radiator_temperature, grid_relay_status, grid_current,       |
|    |                       generator_relay_status, pv2_current                        |
| 60 |     day_pv_energy, day_gen_energy, day_battery_discharge, total_grid_export,     |
|    |       day_active_energy, date_time, total_battery_charge, day_grid_import,       |
|    |    total_grid_import, day_battery_charge, total_load_energy, day_load_energy,    |
|    |  day_grid_export, total_active_energy, total_pv_energy, total_battery_discharge  |
+----+----------------------------------------------------------------------------------+
12:07:18 INFO    Report every (inverter >1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  30 |       pv1_power, load_l3_power, grid_ct_power, grid_l3_power, load_power,        |
|     |  gen_l3_power, inverter_l1_power, pv2_power, inverter_l3_power, load_l1_power,   |
|     | grid_power, gen_power, inverter_l2_power, inverter_power, pv3_power, pv4_power,  |
|     |     grid_l1_power, grid_l2_power, gen_l1_power, load_l2_power, gen_l2_power      |
|  60 |  grid_peak_shaving_power, prog2_power, prog5_voltage, battery_capacity_current,  |
|     | prog4_voltage, battery_wake_up, battery_resistance, battery_absorption_voltage,  |
|     | prog5_power, date_time, prog5_time, generator_port_usage, use_timer, prog3_time, |
|     |    battery_float_voltage, battery_equalization_days, prog4_time, prog1_time,     |
|     |     prog6_power, export_limit_power, system_zero_export_power, prog3_power,      |
|     |     prog1_capacity, solar_export, grid_charge_start_battery_soc, prog2_time,     |
|     |            prog6_capacity, prog6_time, prog2_voltage, prog4_capacity,            |
|     |            battery_restart_capacity, load_limit, battery_low_voltage,            |
|     |       battery_shutdown_capacity, battery_restart_voltage, prog2_capacity,        |
|     |        prog5_capacity, prog5_charge, battery_low_capacity, prog1_charge,         |
|     |               battery_shutdown_voltage, generator_charge_enabled,                |
|     |    battery_equalization_voltage, prog1_voltage, prog3_charge, priority_load,     |
[... cut ...]
12:07:30 WARNING use_timer: Unknown register value 255
12:07:30 WARNING device_type: Unknown register value 6
12:07:35 WARNING use_timer: Unknown register value 255
12:07:35 WARNING device_type: Unknown register value 6
12:07:35 WARNING use_timer: Unknown register value 255
12:07:35 WARNING device_type: Unknown register value 6
```

The values for PV1 Power and PV2 Power seem correct now, but I do not get any battery Power info anymore, just the daily charge/discharge.

And Lots of warnings, see above

bsafh, 2024-06-11:
just for testing, I enabled all sensors I could find in the definitionsfile for 3-phase-hv.

Log looks like this now:

```
13:00:31 WARNING use_timer: Unknown register value 255
13:00:31 WARNING device_type: Unknown register value 6
13:00:31 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:31 WARNING on_grid_always_on: Unknown register value 0
13:00:31 WARNING us_version_grounding_fault: Unknown register value 0
13:00:31 WARNING allow_remote: Unknown register value 0
13:00:31 WARNING use_timer: Unknown register value 255
13:00:31 WARNING device_type: Unknown register value 6
13:00:31 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:31 WARNING on_grid_always_on: Unknown register value 0
13:00:31 WARNING us_version_grounding_fault: Unknown register value 0
13:00:31 WARNING allow_remote: Unknown register value 0
13:00:31 WARNING Cannot publish Use Timer: value is None
13:00:31 WARNING Cannot publish Use Timer: value is None
13:00:31 WARNING Cannot publish Microinverter export to grid cutoff: value is None
13:00:31 WARNING Cannot publish On Grid always on: value is None
13:00:31 WARNING Cannot publish US version grounding fault: value is None
13:00:31 WARNING Cannot publish Allow Remote: value is None
13:00:31 WARNING Cannot publish Device type: value is None
13:00:31 WARNING Cannot publish Microinverter export to grid cutoff: value is None
13:00:31 WARNING Cannot publish On Grid always on: value is None
13:00:31 WARNING Cannot publish US version grounding fault: value is None
13:00:31 WARNING Cannot publish Allow Remote: value is None
13:00:31 WARNING Cannot publish Device type: value is None
13:00:35 WARNING use_timer: Unknown register value 255
13:00:35 WARNING device_type: Unknown register value 6
13:00:35 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:35 WARNING on_grid_always_on: Unknown register value 0
13:00:35 WARNING us_version_grounding_fault: Unknown register value 0
13:00:35 WARNING allow_remote: Unknown register value 0
13:00:35 WARNING use_timer: Unknown register value 255
13:00:35 WARNING device_type: Unknown register value 6
13:00:35 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:35 WARNING on_grid_always_on: Unknown register value 0
13:00:35 WARNING us_version_grounding_fault: Unknown register value 0
13:00:35 WARNING allow_remote: Unknown register value 0
13:00:41 WARNING use_timer: Unknown register value 255
13:00:41 WARNING device_type: Unknown register value 6
13:00:41 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:41 WARNING on_grid_always_on: Unknown register value 0
13:00:41 WARNING us_version_grounding_fault: Unknown register value 0
13:00:41 WARNING allow_remote: Unknown register value 0
13:00:41 WARNING use_timer: Unknown register value 255
13:00:41 WARNING device_type: Unknown register value 6
13:00:41 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:41 WARNING on_grid_always_on: Unknown register value 0
13:00:41 WARNING us_version_grounding_fault: Unknown register value 0
13:00:41 WARNING allow_remote: Unknown register value 0
13:00:45 WARNING use_timer: Unknown register value 255
13:00:45 WARNING device_type: Unknown register value 6
13:00:45 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:45 WARNING on_grid_always_on: Unknown register value 0
13:00:45 WARNING us_version_grounding_fault: Unknown register value 0
13:00:45 WARNING allow_remote: Unknown register value 0
13:00:45 WARNING use_timer: Unknown register value 255
13:00:45 WARNING device_type: Unknown register value 6
13:00:45 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:45 WARNING on_grid_always_on: Unknown register value 0
13:00:45 WARNING us_version_grounding_fault: Unknown register value 0
13:00:45 WARNING allow_remote: Unknown register value 0
13:00:51 WARNING use_timer: Unknown register value 255
13:00:51 WARNING device_type: Unknown register value 6
13:00:51 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:51 WARNING on_grid_always_on: Unknown register value 0
13:00:51 WARNING us_version_grounding_fault: Unknown register value 0
13:00:51 WARNING allow_remote: Unknown register value 0
13:00:51 WARNING use_timer: Unknown register value 255
13:00:51 WARNING device_type: Unknown register value 6
13:00:51 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:51 WARNING on_grid_always_on: Unknown register value 0
13:00:51 WARNING us_version_grounding_fault: Unknown register value 0
13:00:51 WARNING allow_remote: Unknown register value 0
13:00:55 WARNING use_timer: Unknown register value 255
13:00:55 WARNING device_type: Unknown register value 6
13:00:55 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:55 WARNING on_grid_always_on: Unknown register value 0
13:00:55 WARNING us_version_grounding_fault: Unknown register value 0
13:00:55 WARNING allow_remote: Unknown register value 0
13:00:55 WARNING use_timer: Unknown register value 255
13:00:55 WARNING device_type: Unknown register value 6
13:00:55 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
13:00:55 WARNING on_grid_always_on: Unknown register value 0
13:00:55 WARNING us_version_grounding_fault: Unknown register value 0
13:00:55 WARNING allow_remote: Unknown register value 0

```

bsafh, 2024-06-12:
added battery definitions from the source code.

works, but still gives errors/warnings on startup

```
06:12:01 WARNING Cannot publish Use Timer: value is None
06:12:01 WARNING Cannot publish Microinverter export to grid cutoff: value is None
06:12:01 WARNING Cannot publish On Grid always on: value is None
06:12:01 WARNING Cannot publish US version grounding fault: value is None
06:12:01 WARNING Cannot publish Allow Remote: value is None
06:12:01 WARNING Cannot publish Use Timer: value is None
06:12:01 WARNING Cannot publish Microinverter export to grid cutoff: value is None
06:12:01 WARNING Cannot publish On Grid always on: value is None
06:12:01 WARNING Cannot publish US version grounding fault: value is None
06:12:01 WARNING Cannot publish Allow Remote: value is None
06:12:05 WARNING use_timer: Unknown register value 255
06:12:05 WARNING device_type: Unknown register value 6
06:12:05 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
06:12:05 WARNING on_grid_always_on: Unknown register value 0
06:12:05 WARNING us_version_grounding_fault: Unknown register value 0
06:12:05 WARNING allow_remote: Unknown register value 0
....

```

kellerza, 2024-06-18:
Try the latest dev version, it should print less warnings (At least the None should only be during startup)

Not all registers are reported by all inverters, which very likely explains the 0's.

closed 2024-06-18 by kellerza

bsafh, 2024-06-18:
just updated to the latest dev version.

Less warnings, yes.

It looks like this now:

```
07:40:21 WARNING device_type: Unknown register value 6. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
07:40:21 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
07:40:21 WARNING on_grid_always_on: Unknown register value 0
07:40:21 WARNING us_version_grounding_fault: Unknown register value 0
07:40:21 WARNING allow_remote: Unknown register value 0
07:40:25 WARNING use_timer: Unknown register value 255
```

`use_timer` should definitely be known ;)

inverter model is:

Inverter:
Make: Deye.
Model: SUN-10K-SG01HP3-EU-AM2