kellerza/sunsynk #299: pv_total not working

apuntigam, 2024-06-21:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py)

### Your Inverter

Details about your Inverter: Sunsynk 12kW 3-phase, using pymodbus and getting sensor reading from RS485 

### Describe the issue/bug and what you expect

I see that the pv_power sensor has been added to the 3phase_hv definitions [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_hv.py), but when I add pv_power to my Sunsynk configuration, it does not show up as an entity in Home Assistant. 

In the configuration, I only have the option to select single phase or three phase, so I'm assuming the 3phase_hv sensor definitions should be pulled in by default? But then pv_power should be available, and it isn't. It doesn't appear in the log at all. And if I search for it as an entity in Home Assistant, it's not there.

I also tried adding a similar sensor to mysensors.py as explained [here](https://kellerza.github.io/sunsynk/reference/mysensors), but no luck with that either.

Note that both pv1_power and pv2_power read just fine on registers 672 and 673 respectively.

I would love to be able to get this working!

### Logs (if applicable)

I included an extract from the log, happy to provide whatever else you might need.

```logs
2024-06-21 09:42:50,688 DEBUG   sunsynk.sensors load_l3_power=379W (379,)
2024-06-21 09:42:50,688 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = 374 (old=379)
2024-06-21 09:42:50,688 DEBUG   sunsynk.sensors pv1_power=2556W (2556,)
2024-06-21 09:42:50,688 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = 2569 (old=2556)
2024-06-21 09:42:50,689 DEBUG   sunsynk.sensors pv2_power=2620W (2620,)
2024-06-21 09:42:50,689 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = 2641 (old=2620)
2024-06-21 09:42:50,689 DEBUG   sunsynk.sensors grid_charge_battery_current=40A (40,)
2024-06-21 09:42:50,689 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = 40 (old=40)
2024-06-21 09:42:50,689 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = ON (old=ON)
2024-06-21 09:42:50,689 WARNING sunsynk.rwsensors use_timer: Unknown register value 1
2024-06-21 09:42:50,689 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = None (old=None)
2024-06-21 09:42:50,689 DEBUG   sunsynk.state register {3: 12850, 4: 12345, 5: 12851, 6: 14128, 7: 13620, 20: 54464, 21: 1, 117: 20, 118: 4600, 119: 5200, 120: 4750, 101: 5520, 102: 300, 108: 150, 109: 180, 115: 10, 116: 50, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 143: 0, 144: 11, 145: 0, 146: 1, 194: 0, 500: 2, 501: 0, 502: 0, 506: 0, 507: 0, 514: 67, 515: 43, 516: 23612, 517: 0, 518: 22929, 519: 0, 520: 4, 521: 0, 522: 8144, 523: 0, 524: 45, 525: 0, 526: 73, 527: 50227, 528: 0, 529: 106, 534: 48789, 535: 0, 540: 1250, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1227, 587: 5476, 588: 84, 589: 0, 590: 61887, 600: 2236, 609: 5012, 610: 327, 611: 42, 612: 74, 619: 33, 622: 34, 623: 0, 624: 2, 625: 36, 626: 0, 627: 2200, 633: 748, 634: 89, 635: 372, 636: 1209, 637: 1209, 638: 5016, 650: 782, 651: 89, 652: 374, 653: 1245, 672: 2569, 673: 2641, 676: 3099, 677: 83, 678: 3117, 679: 84} = OFF (old=OFF)

```

rixxxx, 2024-06-21:
I think Sunsynk 12kW 3-phase is not HV but LV

rixxxx, 2024-06-21:
and there is no math sensor for LV atm

apuntigam, 2024-06-21:
After a small tweak, I have the custom sensor working! So for now it's sorted.  But please consider added this to a future update, it makes it much easier to view history graphs of PV production vs load, etc, when you have total PV.  Trying to use these graphs when you have PV1_power and PV2_power as separate plots isn't great. Thanks!

apuntigam, 2024-06-21:
FYI, for anyone who wants to set up the custom sensor, I saved this to a mysensors.py file in /share/hass-addon-sunsynk/

`from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()

SENSORS += MathSensor((672, 673), "PV Power Total", WATT, factors = (1, 1))`

rixxxx, 2024-06-21:
what happens if you do?
```
MathSensor((672, 673, 674, 675), "PV power", WATT))
```

rixxxx, 2024-06-21:
Are there any 3PH LV inverters with 4 MPPTs at all?

apuntigam, 2024-06-21:
> Are there any 3PH LV inverters with 4 MPPTs at all?

No idea unfortunately, I'm just an end-user not an installer. The user manual for my inverter says it only has 2 MPPTs though.

apuntigam, 2024-06-21:
> what happens if you do?
> 
> ```
> MathSensor((672, 673, 674, 675), "PV power", WATT)
> ```

It works, BUT it had to be as follows:
'MathSensor((672, 673, 674, 675), "PV power", WATT, factors = (1, 1, 1, 1))'

It would not detect the entity without the factors.

TheBazeman, 2024-06-24:
In Home Assistant you can just create a helper group sensor which sums the values ;-) Easy as that

rixxxx, 2024-06-24:
#303

closed 2024-07-03 by kellerza